zephyrproject-rtos/zephyr #49479: openthread/coprocessor building gives fatal error: logging/log.h: No such file or directory

lboue, 2022-08-24:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? nrf52840dongle
 - What have you tried to diagnose or workaround this issue? yes

**To Reproduce**
Steps to reproduce the behavior:
1. wget https://github.com/zephyrproject-rtos/zephyr/archive/refs/tags/v2.7.3.zip
2. unzip v2.7.3.zip
3. cd ~/zephyr-2.7.3/samples/net/openthread/coprocessor
4. west init
5. west update
6. west build -p always -b nrf52840dongle_nrf52840 -- -DOVERLAY_CONFIG="overlay-usb-nrf-br.conf"  -DCONFIG_OPENTHREAD_THREAD_VERSION_1_2=y
7. See error

**Expected behavior**
Build nrf52840dongle_nrf52840 RCP firmware

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
user@ubuntu:~/zephyr-2.7.3/samples/net/openthread/coprocessor$ west build -p always -b nrf52840dongle_nrf52840 -- -DOVERLAY_CONFIG="overlay-usb-nrf-br.conf"  -DCONFIG_OPENTHREAD_THREAD_VERSION_1_2=y
-- west build: making build dir /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.1.99 (/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: nrf52840dongle_nrf52840
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.2 (/home/user/zephyr-sdk-0.14.2)
-- Found dtc: /home/user/zephyr-sdk-0.14.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.2 (/home/user/zephyr-sdk-0.14.2)
-- Found BOARD.dts: /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840.dts
-- Found devicetree overlay: /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/app.overlay
-- Generated zephyr.dts: /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/zephyr/dts.cmake
Parsing /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/Kconfig
Loaded configuration '/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840_defconfig'
Merged configuration '/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/prj.conf'
Merged configuration '/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/overlay-usb-nrf-br.conf'
Merged configuration '/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/zephyr/.config'
Kconfig header saved to '/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/zephyr/include/generated/autoconf.h'

warning: LOG_BACKEND_RTT (defined at subsys/logging/Kconfig.backends:85) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: USE_SEGGER_RTT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_LOG_BACKEND_RTT and/or look up
LOG_BACKEND_RTT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: SHELL_PROMPT_UART (defined at subsys/shell/backends/Kconfig.backends:36) was assigned the
value 'ncp-uart:~$ ' but got the value ''. Check these unsatisfied dependencies:
SHELL_BACKEND_SERIAL (=n), SHELL_BACKENDS (=n), SHELL (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SHELL_PROMPT_UART and/or look up
SHELL_PROMPT_UART in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
[... 68 lines cut ...]
NRF_802154_PENDING_SHORT_ADDRESSES=16
NRF_802154_PENDING_EXTENDED_ADDRESSES=16
NRF_802154_RX_BUFFERS=16
NRF_802154_TX_STARTED_NOTIFY_ENABLED=1
NRF_802154_ACK_TIMEOUT_ENABLED=1
NRF_802154_ENCRYPTION_ENABLED=0
NRF_802154_SECURITY_WRITER_ENABLED=0
NRF_802154_IE_WRITER_ENABLED=0
NRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0
[460/830] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/user/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DNRF52840_XXAA -DNRF_802154_ACK_TIMEOUT_ENABLED=1 -DNRF_802154_CCA_CORR_LIMIT_DEFAULT=2 -DNRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_ED_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED -DNRF_802154_CSMA_CA_ENABLED=0 -DNRF_802154_DELAYED_TRX_ENABLED=0 -DNRF_802154_ENCRYPTION_ENABLED=0 -DNRF_802154_FRAME_TIMESTAMP_ENABLED=0 -DNRF_802154_IE_WRITER_ENABLED=0 -DNRF_802154_IFS_ENABLED=0 -DNRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0 -DNRF_802154_PENDING_EXTENDED_ADDRESSES=16 -DNRF_802154_PENDING_SHORT_ADDRESSES=16 -DNRF_802154_RX_BUFFERS=16 -DNRF_802154_SECURITY_WRITER_ENABLED=0 -DNRF_802154_TX_STARTED_NOTIFY_ENABLED=1 -DNRF_802154_USE_RAW_API=1 -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/zephyr/include/generated -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/soc/arm/nordic_nrf/common/. -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/subsys/usb/device -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/subsys/settings/include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/hal/cmsis/CMSIS/Core/Include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/hal/nordic/nrfx -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/hal/nordic/nrfx/drivers/include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/hal/nordic/nrfx/mdk -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/modules/hal_nordic/nrfx/. -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/modules/hal_nxp/usb/. -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/modules/openthread/platform/. -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/hal/nordic/drivers/nrf_802154/driver/include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/hal/nordic/drivers/nrf_802154/sl/include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/hal/nordic/drivers/nrf_802154/sl/sl_opensource/include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build/modules/openthread/build/etc/cmake -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/lib/openthread/etc/cmake -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/crypto/mbedtls/include -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/crypto/mbedtls/library -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/modules/mbedtls/configs -I/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/modules/mbedtls/include -isystem /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/lib/libc/minimal/include -isystem /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr/subsys/cpp/include -isystem /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/lib/openthread/include -isystem /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/modules/lib/openthread/examples/platforms -isystem /home/user/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr [... cut ...]
/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/src/main.c:9:10: fatal error: logging/log.h: No such file or directory
    9 | #include <logging/log.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[464/830] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/build
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.3 LTS
 - Toolchain: zephyr-2.7.3
 - Commit zephyr-2.7.3

thedjnK, 2022-08-24:
You are mixing zephyr versions up: `-- Zephyr version: 3.1.99 (/home/user/zephyr-2.7.3/samples/net/openthread/coprocessor/zephyr)`

The way to get v2.7 is to use west directly, i.e. `west init --mr v2.7.3` not download a zip and use west

lboue, 2022-08-25:
Thanks @thedjnK.
So for fixing _fatal error: logging/log.h: No such file or directory_ do you have any advice?

Regards
Ludovic

thedjnK, 2022-08-26:
If you want to develop using zephyr 3.1/main, get `zephyr-v3.1.0` or `main`. If you want to develop using zephyr 2.7 (LTS), get `v2.7.3`, as per:
```
west init
cd zephyr
git checkout <version>
west update
cd samples/net/openthread/coprocessor
mkdir build
cd build
cmake -GNinja -DBOARD=nrf52840dongle_nrf52840 -DOVERLAY_CONFIG="overlay-usb-nrf-br.conf" -DCONFIG_OPENTHREAD_THREAD_VERSION_1_2=y ..
ninja
```

carlescufi, 2022-08-30:
> If you want to develop using zephyr 3.1/main, get `zephyr-v3.1.0` or `main`. If you want to develop using zephyr 2.7 (LTS), get `v2.7.3`, as per:
> 
> ```
> west init
> cd zephyr
> git checkout <version>
> west update
> west zephyr-export
> source zephyr-env.sh
> cd samples/net/openthread/coprocessor
> mkdir build
> cd build
> cmake -GNinja -DBOARD=nrf52840dongle_nrf52840 -DOVERLAY_CONFIG="overlay-usb-nrf-br.conf" -DCONFIG_OPENTHREAD_THREAD_VERSION_1_2=y ..
> ninja
> ```

Please try out these instructions and reopen the issue if it still fails for you.

closed 2022-08-30 by carlescufi