zephyrproject-rtos/zephyr #49480: arduino_nano_33_ble: samples/hello_world: USB/uart is not enabled

rzr, 2022-08-24:
**Describe the bug**

Default images have console properly configured (at least nucleo_l476rg have it enabled)
I expected same on this board too.

I suspect some regression since zephyr 2.7.0 (need to verify and bissec if working)

**To Reproduce**
Steps to reproduce the behavior:

```
west build -p auto -b "$boardname" "$example"
screen /dev/ttyACM*
```

**Expected behavior**

Should connect and display:

```
*** Booting Zephyr OS build zephyr-v3.1.0-4045-g3dccd319b6bf  ***
Hello World! qemu_x86
```

**Impact**
annoyance

**Logs and console output**

```
sudo dmesg
[171973.319289] usb 1-3.3.1: Product: Arduino Nano 33 BLE
[171973.319291] usb 1-3.3.1: Manufacturer: Arduino
[171973.319293] usb 1-3.3.1: SerialNumber: *
[171973.327999] cdc_acm 1-3.3.1:1.0: ttyACM0: USB ACM device
[171973.578839] usb 1-3.3.1: USB disconnect, device number 83
[171983.990147] usb 1-3.3.1: new full-speed USB device number 84 using xhci_hcd
[171984.072644] usb 1-3.3.1: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
[171984.072654] usb 1-3.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[171984.081031] usb 1-3.3.1: can't set config #1, error -32
[171984.081811] usb 1-3.3.1: USB disconnect, device number 84
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA or Version used: zephyr-v3.1.0-4132-gf033040812

**Additional context**
To be updated if asked

Relate-to: https://github.com/zephyrproject-rtos/zephyr/issues/36084#issuecomment-1224300787

jfischer-no, 2022-09-02:
> Default images have console properly configured (at least nucleo_l476rg have it enabled)
> I expected same on this board too.

nucleo_l476rg board has a debug adapter on the board with CDC ACM support, arduino_nano_33_ble does not. As described elsewhere, it makes no sense to enable CDC ACM support for this type of boards by default, it leads to other problems and exact same repeated configuration for many boards. It is also not a bug, it is just the way it is. Either use overlays or wait for https://github.com/zephyrproject-rtos/zephyr/pull/40645.

rzr, 2022-09-06:
well it used to work differently, so I thought it was a regression

jfischer-no, 2024-02-09:
Resolved by https://docs.zephyrproject.org/latest/snippets/cdc-acm-console/README.html

closed 2024-02-09 by jfischer-no