zephyrproject-rtos/zephyr #49484: CONFIG_BOOTLOADER_SRAM_SIZE should not be defined by default

quic-egmc, 2022-08-24:
The Kconfig option BOOTLOADER_SRAM_SIZE defaults to reserving 16KB if XIP is not defined and, strangely, only on ARM or Xtensa architectures.

The description states:
` This option specifies the amount of SRAM (measure in kB) reserved for
      a bootloader image, when either:
      - the Zephyr image itself is to act as the bootloader, or
      - Zephyr is a !XIP image, which implicitly assumes existence of a
      bootloader that loads the Zephyr !XIP image onto SRAM.`

but the implicit assumption that we want to reserve space for a bootloader is not valid for many platforms.  Please make this flag explicit or dependent on something obvious.  This default behavior results in wasted space and is non-obvious!

microbuilder, 2023-01-01:
@d3zd3z Mind taking a look at this?

quic-egmc, 2023-03-03:
@microbuilder  @d3zd3z  reminder please

d3zd3z, 2023-03-03:
I'm not sure why it is defined on xtensa, either, since only arm even makes use of the symbol. 21628d41d added this, which allows a particular configuration with a bootloader that uses this reserved memory for its sram so that the image can use the rest of the sram. @benwrs do you have any insight on when this gets used, and if maybe we could make the configuration clearer?

mmahadevan108, 2023-07-13:
@benwrs @peter-mitsis, can you help look at this issue.

mmahadevan108, 2023-07-13:
> @quic-egmc but the implicit assumption that we want to reserve space for a bootloader is not valid for many platforms. 
This SRAM space is reserved for the case only when Zephyr is configured to act as a bootloader and not for all bootloaders. I thought I will mention this incase it helps.

microbuilder, 2023-07-13:
> This SRAM space is reserved for the case only when Zephyr is configured to act as a bootloader and not for all bootloaders. I thought I will mention this incase it helps.

Thanks for the context. I didn't understand the use case here.

microbuilder, 2023-07-13:
This does feel very arbitrary:

```
config BOOTLOADER_SRAM_SIZE
	int "SRAM reserved for bootloader"
	default 16
	depends on !XIP || IS_BOOTLOADER
	depends on ARM || XTENSA
	help
	  This option specifies the amount of SRAM (measure in kB) reserved for
	  a bootloader image, when either:
	  - the Zephyr image itself is to act as the bootloader, or
	  - Zephyr is a !XIP image, which implicitly assumes existence of a
	  bootloader that loads the Zephyr !XIP image onto SRAM.
```

And given that using Zephyr AS the bootloader is a pretty specific edge case, this shouldn't be reserved by default in every case for Arm/Xtensa. I'm not the right person to sort this out since I don't know the various use cases, but @d3zd3z or @peter-mitsis or @bendiscz If you have suggestions, this shouldn't updated to reflect that it's an exception not the rule with Arm/Xtensa, etc.

microbuilder, 2023-07-13:
@nordicjm @carlescufi  Do you know of use cases for using Zephyr itself as the bootloader with Nordic customers? Struggling to understand the context for this, since I've normally only seen MCUBoot or a HW bootloader used with Zephyr, not Zephyr as bootloader.

microbuilder, 2023-07-13:
I'd suggest this should default to `0` and in the edge case where it's needed, it can be set higher. I already see some boards in tree doing this:

```
# Override the setting in misc/Kconfig to allow full use of SRAM:
config BOOTLOADER_SRAM_SIZE
	default 0 if !XIP
```

Or: 

```
CONFIG_BOOTLOADER_SRAM_SIZE=0
```

microbuilder, 2023-07-13:
Sorry for all the messages, but ... the mystery that is `CONFIG_IS_BOOTLOADER` apparently goes pretty far back: https://github.com/zephyrproject-rtos/zephyr/issues/9566#issuecomment-463973329 ... and this feature was initially introduced in 2016: https://github.com/zephyrproject-rtos/zephyr/commit/21628d41d02115090208b42739ff5448c801900a

I'm going to flag this for review in the arch meeting since I  don't know the context here, but we should sort it out.

@quic-egmc I'm not sure this is still important to you, but can you give me some context on how you're hitting this issue? It seems like this should only be enabled if `CONFIG_IS_BOOTLOADER=y` for your board. 

Doing a quick test with a random Arm board, I'm not seeing the 16KB being assigned:

```bash
$ west build -p auto -b qemu_cortex_m0 samples/hello_world
...
Memory region         Used Size  Region Size  %age Used
           FLASH:       15518 B       256 KB      5.92%
             RAM:        3936 B        16 KB     24.02%
        IDT_LIST:          0 GB         2 KB      0.00%
```

If you can give me a build target where `BOOTLOADER_SRAM_SIZE` is being set, it would help.

microbuilder, 2023-07-13:
@quic-egmc Thanks for the reply, Eugene. I'll raise a PR tomorrow setting the default to 0, but I'd like to discuss the !XIP bit in the arch call to make sure I understand where this came from and why, and see how we can better document this. And also why it's only enabled for Arm and Xtensa.

quic-egmc, 2023-07-13:
I'm struggling to get CONFIG_XIP not set upstream versus on our internal projects....

quic-egmc, 2023-07-13:
Ok I found a real example in-tree.  Look at the board bcm958402m2_m7_defconfig. 

It disables XIP but does not override BOOTLOADER_SRAM_SIZE and in its .config it gets:

`CONFIG_BOOTLOADER_SRAM_SIZE=16`

and the resulting RAM usage map shows the *total* RAM reduced by 16K, at 496KB:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:          0 GB         0 GB
             RAM:       24596 B       496 KB      4.84%
        IDT_LIST:          0 GB         2 KB      0.00%
```

now force CONFIG_BOOTLOADER_SRAM_SIZE to zero in boards/arm/bcm958402m2_m7/bcm958402m2_m7_defconfig and you instead see the total RAM size increased to 512KB:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:          0 GB         0 GB
             RAM:       24596 B       512 KB      4.69%
        IDT_LIST:          0 GB         2 KB      0.00%
```

this stems from boards/arm/bcm958402m2_m7/bcm958402m2_m7_defconfig:
`#define RAM_SIZE (CONFIG_SRAM_SIZE * 1K - CONFIG_BOOTLOADER_SRAM_SIZE * 1K)`

microbuilder, 2023-07-13:
@quic-egmc I raised a PR setting the default to `0` (you're tagged for review), but if you're available and if this topic gets scheduled in next week's arch call, I'd be interested to discuss `CONFIG_XIP` and what you'd like to see changed upstream.

PR #60371 (2023-07-17): kconfig: Set `BOOTLOADER_SRAM_SIZE` default to 0
The current default value for `BOOTLOADER_SRAM_SIZE` is set to an arbitrary-seeming value of `16`, which feels like a random magic number.

Given that adding an offset in SRAM for a bootloader should always be a conscious choice for a specific platform, this PR sets the default to `0`. An appropriate value can be set at the SoC or board level, overriding this default.

Fixes: #49484
closed 2023-07-17 by fabiobaltieri