zephyrproject-rtos/zephyr #49491: arch.interrupt test fails on ARM64 QEMU targets when compiled with GCC 12

stephanosio, 2022-08-24:
**Describe the bug**

The `arch.interrupt` test fails on the ARM64 QEMU targets (`qemu_cortex_a53` and `qemu_cortex_a53_smp`) when compiled with GCC 12, which is the toolchain included in the Zephyr SDK 0.15.0.

**To Reproduce**

```
scripts/twister -v -N -M -p qemu_cortex_a53 -s tests/kernel/interrupt/arch.interrupt
```

**Expected behavior**

Test passes.

**Logs and console output**

```
Running TESTSUITE interrupt_feature
===================================================================
START - test_isr_dynamic
installing dynamic ISR for IRQ 0
 PASS - test_isr_dynamic in 0.001 seconds
===================================================================
START - test_isr_offload_job
 SKIP - test_isr_offload_job in 0.001 seconds
===================================================================
START - test_isr_offload_job_identi
 PASS - test_isr_offload_job_identi in 0.015 seconds
===================================================================
START - test_isr_offload_job_multiple
 PASS - test_isr_offload_job_multiple in 0.021 seconds
===================================================================
START - test_nested_isr
Triggering irq : 14
isr0: Enter
Triggering irq : 15

    Assertion failed at WEST_TOPDIR/zephyr-testing/tests/kernel/interrupt/src/nested_irq.c:106: isr0: (isr1_result not equal to ISR1_TOKEN)
isr1 did not execute
isr0: Leave
isr1: Enter
isr1: Leave
 FAIL - test_nested_isr in 0.006 seconds
===================================================================
START - test_prevent_interruption
locking interrupts
unlocking interrupts
timer fired
 PASS - test_prevent_interruption in 2.540 seconds
===================================================================
TESTSUITE interrupt_feature failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  80.00% [interrupt_feature]: pass = 4, fail = 1, skip = 1, total = 6 duration = 2.584 seconds
 - PASS - [interrupt_feature.test_isr_dynamic] duration = 0.001 seconds
 - SKIP - [interrupt_feature.test_isr_offload_job] duration = 0.001 seconds
 - PASS - [interrupt_feature.test_isr_offload_job_identi] duration = 0.015 seconds
 - PASS - [interrupt_feature.test_isr_offload_job_multiple] duration = 0.021 seconds
 - FAIL - [interrupt_feature.test_nested_isr] duration = 0.006 seconds
 - PASS - [interrupt_feature.test_prevent_interruption] duration = 2.540 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 548cf47f515fc665f4459ceaf48fa404
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: 822c37745e189c3505c7dc45c405ee16135be1e7

**Additional context**

Adding a `printk` call in the [`arm_gic_get_active` function](https://github.com/zephyrproject-rtos/zephyr/blob/a1a2107b429def632dfa277459d078d3a64b83c6/drivers/interrupt_controller/intc_gicv3.c#L213-L221) makes this test pass even when compiling with GCC 12.

stephanosio, 2022-08-24:
cc @carlocaione @npitre @povergoing @SgrrZhf

povergoing, 2022-08-25:
Weird, I need to take a further look. `fvp_base_revc_2xaemv8a` also fails but `fvp_baser_aemv8r` has no issue

povergoing, 2022-08-25:
I found it. Any changes related to `printk`, more specifically, `va_list`, `va_start` and `va_end`? Compared with the previous version, I found GCC12 will insert a `ldp     q0, q1, [sp, #xx]` and `ldp     q0, q1, [sp, #48],` for example,
Old version
```
    88001f70:   3d8017e0        str     q0, [sp, #80]
    88001f74:   3d801be1        str     q1, [sp, #96]
    88001f78:   3d801fe2        str     q2, [sp, #112]
    88001f7c:   3d8023e3        str     q3, [sp, #128]
    88001f80:   3d8027e4        str     q4, [sp, #144]
    88001f84:   3d802be5        str     q5, [sp, #160]
    88001f88:   3d802fe6        str     q6, [sp, #176]
    88001f8c:   3d8033e7        str     q7, [sp, #192]
    88001f90:   a90f9be5        stp     x5, x6, [sp, #248]
    88001f94:   f90087e7        str     x7, [sp, #264]
        vprintk(fmt, ap);
    88001f98:   97ffffd6        bl      88001ef0 <vprintk>
```
GCC12:
```
    88001ef4:   3d8017e0        str     q0, [sp, #80]
    88001ef8:   3d801be1        str     q1, [sp, #96]
        vprintk(fmt, ap);
    88001efc:   ad4187e0        ldp     q0, q1, [sp, #48]
{
    88001f00:   3d801fe2        str     q2, [sp, #112]
    88001f04:   3d8023e3        str     q3, [sp, #128]
        vprintk(fmt, ap);
    88001f08:   ad0087e0        stp     q0, q1, [sp, #16]
{
    88001f0c:   3d8027e4        str     q4, [sp, #144]
    88001f10:   3d802be5        str     q5, [sp, #160]
    88001f14:   3d802fe6        str     q6, [sp, #176]
    88001f18:   3d8033e7        str     q7, [sp, #192]
    88001f1c:   a90e93e3        stp     x3, x4, [sp, #232]
    88001f20:   a90f9be5        stp     x5, x6, [sp, #248]
    88001f24:   f90087e7        str     x7, [sp, #264]
        vprintk(fmt, ap);
    88001f28:   97ffffd9        bl      88001e8c <vprintk>
```
That will cause an issue with `CONFIG_FPU_SHARING` because this feature simulates the `str  qx, qy [sp #xx]` for performance I think? https://github.com/zephyrproject-rtos/zephyr/blob/180f50feda2514b792d2588f58907856e0de1461/arch/arm64/core/fpu.c#L147-L157
But doesn't simulate the `ldp     q0, q1, [sp, #48]` instruction which will case a fpu trap, enabling the fpu and disabling the IRQ and thus forbid a nested irq. Thus test fails.
https://github.com/zephyrproject-rtos/zephyr/blob/180f50feda2514b792d2588f58907856e0de1461/arch/arm64/core/fpu.c#L242-L251

Is there any config for GCC12 to go back to the original `va_list` compiling behavior? Or does anyone think it will be better to simulate the `ldr` instruction (also plus a `stp`)?

npitre, 2022-08-25:
Yep, more context in commit b8d24ffb459f ("arm64: mitigate 
FPU-in-exception usage side effects").

> Is there any config for GCC12 to go back to the original `va_list` 
> compiling behavior? 

I wonder why it has to move 32 bytes between two locations on the stack
like that in the first place.

> Or does anyone think it will be better to simulate the `ldr`
> instruction (also plus a `stp`)?

If gcc has a legitimate reason for doing this then yes.

povergoing, 2022-08-29:
> I wonder why it has to move 32 bytes between two locations on the stack
like that in the first place.

Yes, good question. I am trying to find out why :)

povergoing, 2022-08-31:
>I wonder why it has to move 32 bytes between two locations on the stack
like that in the first place.

@npitre Seems the GCC12 uses `qn` registers to copy content to substitute the original `xn` ones, for performance improvement maybe? 
For example,
```
ldp     x4, x5, [x1]
ldp     x2, x3, [x1, #16]
stp     x4, x5, [sp]
stp     x2, x3, [sp, #16]
```
=>
```
ldp     q0, q1, [x1]
stp     q0, q1, [sp]
```
So, we may need to simulate the `ldp` and `stp` instructions. But due to the optimization, this pair (`ldp` and `stp`) of instructions can appear randomly, which is probably located separately from those `str qx, [sp, #xx]`s, which means the simulation would be done with several traps instead of the previous one (trap one time, simulate all and skip by fixing the pc before eret)
Something like:
```
str     q0, [sp, #80]
str     q1, [sp, #96]
...
str     q7, [sp, #192]
...
normal instructions
...
ldp     q0, q1, [x1]
...
normal instructions
...
stp     q0, q1, [sp]
```
or 
```
str     q0, [sp, #80]
str     q1, [sp, #96]

ldp     q0, q1, [sp, #48]

str     q2, [sp, #112]
str     q3, [sp, #128]

stp     q0, q1, [sp, #16]

str     q4, [sp, #144]
str     q5, [sp, #160]
str     q6, [sp, #176]
str     q7, [sp, #192]
```

npitre, 2022-08-31:
Can this copy optimization be disabled independently of FP support?

Otherwise I'd be tempted to simply say that the combination of gcc v12 + va_arg usage within IRQ handlers + recursive IRQs is just not supported. 
This shouldn't happen much in practice with real applications anyway, so it might not be worth emulating more of those instructions just for this test sake.

Alternately, the test could be reworked to not print anything from the IRQ handlers by simply adjusting a status variable and let the main thread print the results.

npitre, 2022-08-31:
| Alternately, the test could be reworked to not print anything from the 
| IRQ handlers by simply adjusting a status variable and let the main 
| thread print the results.

Something like this (I don't have gcc 12 installed to test it):

```
diff --git a/tests/kernel/interrupt/src/nested_irq.c b/tests/kernel/interrupt/src/nested_irq.c
index ee48a0366d..1d0a279304 100644
--- a/tests/kernel/interrupt/src/nested_irq.c
+++ b/tests/kernel/interrupt/src/nested_irq.c
@@ -79,20 +79,14 @@ void isr1(const void *param)
 {
 	ARG_UNUSED(param);
 
-	printk("isr1: Enter\n");
-
 	/* Set verification token */
 	isr1_result = ISR1_TOKEN;
-
-	printk("isr1: Leave\n");
 }
 
 void isr0(const void *param)
 {
 	ARG_UNUSED(param);
 
-	printk("isr0: Enter\n");
-
 	/* Set verification token */
 	isr0_result = ISR0_TOKEN;
 
@@ -104,8 +98,6 @@ void isr0(const void *param)
 
 	/* Validate nested ISR result token */
 	zassert_equal(isr1_result, ISR1_TOKEN, "isr1 did not execute");
-
-	printk("isr0: Leave\n");
 }
 
 /**
```

povergoing, 2022-09-01:
> Can this copy optimization be disabled independently of FP support? 

Seems we can't (but not 100% sure)

> Otherwise I'd be tempted to simply say that the combination of gcc v12 + va_arg usage within IRQ handlers + recursive IRQs is just not supported. This shouldn't happen much in practice with real [...]

Agreed, I have tried to emulate them but things are kind of ugly :(

> Alternately, the test could be reworked to not print anything from the IRQ handlers by simply adjusting a status variable and let the main thread print the results.

I have created a PR, which can pass the nested irq test.

stephanosio, 2022-09-02:
An important update -- this problem is no longer specific to `va_list` usage, but all 256-bit copies:

https://github.com/zephyrproject-rtos/zephyr/pull/49806#issuecomment-1235256667
https://github.com/zephyrproject-rtos/zephyr/pull/49806#issuecomment-1235264899

npitre, 2022-09-02:
Agreed. I elaborated in #49806 why I think this should be considered a gcc bug.

stephanosio, 2022-09-23:
UPDATE: https://github.com/zephyrproject-rtos/zephyr/pull/49806#issuecomment-1255830930

I will be sending a patch to fix this shortly.

PR #50573 (2022-09-23): arch: arm64: Disable ldp/stp Qn for consecutive 32-byte loads/stores
GCC may generate ldp/stp instructions with the Advanced SIMD Qn
registers for consecutive 32-byte loads and stores.

This commit disables this GCC behaviour because saving and restoring
the Advanced SIMD context is very expensive, and it is preferable to
keep it turned off by not emitting these instructions for better
context switching performance.

Fixes #49491
closed 2022-09-23 by carlescufi