zephyrproject-rtos/zephyr #49494: testing.ztest.ztress test fails on qemu_cortex_r5 when compiled with GCC 12

stephanosio, 2022-08-24:
**Describe the bug**

The `testing.ztest.ztress` test fails on `qemu_cortex_r5` when compiled with GCC 12, which is the toolchain included in the Zephyr SDK 0.15.0.

**To Reproduce**

With Zephyr SDK 0.15.0,

```
scripts/twister -v -N -M -p qemu_cortex_r5 -s tests/ztest/ztress/testing.ztest.ztress 
```

**Expected behavior**

Test passes.

**Logs and console output**

```
Running TESTSUITE ztress_tests
===================================================================
START - test_timeout

0% remaining:0 ms
Ztress execution report:
	 context 0:
		 - executed:16, preempted:0
		 - ticks initial:20, optimized:4
	 context 1:
		 - executed:17, preempted:0
		 - ticks initial:20, optimized:4
	Average CPU load:1%, measurements:10

0% remaining:0 ms
Ztress execution report:
	 context 0:
		 - executed:20, preempted:0
		 - ticks initial:20, optimized:4
	 context 1:
		 - executed:16, preempted:0
		 - ticks initial:20, optimized:4
	 context 2:
		 - executed:16, preempted:0
		 - ticks initial:20, optimized:4
	Average CPU load:1%, measurements:10
 PASS - test_timeout in 2.226 seconds
===================================================================
START - test_abort

Ztress execution report:
	 context 0:
		 - executed:4, preempted:0
		 - ticks initial:1, optimized:4
	 context 1:
		 - executed:4, preempted:0
		 - ticks initial:1, optimized:4
	Average CPU load:5%, measurements:1
 PASS - test_abort in 0.127 seconds
===================================================================
START - test_repeat_completion

Ztress execution report:
	 context 0:
		 - executed:10, preempted:0
		 - ticks initial:20, optimized:4
	 context 1:
		 - executed:10, preempted:0
		 - ticks initial:20, optimized:4
	Average CPU load:1%, measurements:8

Ztress execution report:
	 context 0:
		 - executed:11, preempted:0
		 - ticks initial:30, optimized:4
	 context 1:
		 - executed:10, preempted:0
		 - ticks initial:30, optimized:4
	 context 2:
		 - executed:10, preempted:0
		 - ticks initial:30, optimized:4
	Average CPU load:1%, measurements:8
 PASS - test_repeat_completion in 1.700 seconds
===================================================================
START - test_no_context_requirements

Ztress execution report:
	 context 0:
		 - executed:1, preempted:0
		 - ticks initial:20, optimized:4
	 context 1:
		 - executed:1, preempted:0
		 - ticks initial:20, optimized:4
	Average CPU load:1%, measurements:3

    Assertion failed at WEST_TOPDIR/zephyr-testing/tests/ztest/ztress/src/main.c:119: test_no_context_requirements: (exec_cnt >= repeat && exec_cnt < repeat + 10 is false)
exec_cnt: 1
 FAIL - test_no_context_requirements in 0.358 seconds
===================================================================
TESTSUITE ztress_tests failed.
===================================================================
RunID: 101c2e450e6e8438799b7a0b391195f8
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: 822c37745e189c3505c7dc45c405ee16135be1e7

nashif, 2023-05-30:
@bbolen  can you please take a look?

bbolen, 2023-05-31:
Yes.  I'll try to give it some time this week.

bbolen, 2023-08-15:
never could reproduce.  going to revert the change that disabled the test on qemu_cortex_r5 and submit it.

closed 2023-10-30 by github-actions[bot]