zephyrproject-rtos/zephyr #49542: sdk: it8xxx2_evb cannot build the hello_world sample after zephyr SDK upgrade to 0.15.0

hunterhu3000, 2022-08-25:
**Describe the bug**
it8xxx2_evb cannot build the any test case including "hellow_world" after zephyr SDK upgrade to 0.15.0.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? -> it8xxx2_evb
 - What have you tried to diagnose or workaround this issue? -> It looks like a riscv issue. Roll back to 0.14.2, everything is ok. Boards of other architecture look good.
 - Is this a regression? -> No, it is caused by the SDK upgrade

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b it8xxx2_evb samples/hello_world/

**Expected behavior**
The test can pass.

**Impact**
None of the test case can be built for it8xxx2_evb on latest SDK 0.15.0

**Logs and console output**
```
ITE@00:00:41:~/zephyrproject/zephyr$ west build -p always -b it8xxx2_evb samples/hello_world/
-- west build: making build dir /home/ztest/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/ztest/zephyrproject/zephyr/samples/hello_world
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter
-- Cache files will be written to: /home/ztest/.cache/zephyr
-- Zephyr version: 3.1.99 (/home/ztest/zephyrproject/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: it8xxx2_evb
-- Found host-tools: zephyr 0.15.0 (/home/ztest/zephyr-sdk-0.15.0)
-- Found dtc: /home/ztest/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.15.0 (/home/ztest/zephyr-sdk-0.15.0)
-- Found BOARD.dts: /home/ztest/zephyrproject/zephyr/boards/riscv/it8xxx2_evb/it8xxx2_evb.dts
-- Generated zephyr.dts: /home/ztest/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/ztest/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/ztest/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/ztest/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ztest/zephyrproject/zephyr/boards/riscv/it8xxx2_evb/it8xxx2_evb_defconfig'
Merged configuration '/home/ztest/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/ztest/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/ztest/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ztest/zephyrproject/zephyr/build
-- west build: building application
[1/150] Preparing syscall dependency handling

[2/150] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/home/ztest/zephyrproject/zephyr), build: zephyr-v3.1.0-4151-g180f50feda25
[33/150] Building ASM object zephyr/CMakeFiles/zephyr.dir/soc/riscv/riscv-ite/common/soc_irq.S.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/soc/riscv/riscv-ite/common/soc_irq.S.obj
ccache /home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/ztest/zephyrproject/zephyr/kernel/include -I/home/ztest/zephyrproject/zephyr/arch/riscv/include -I/home/ztest/zephyrproject/zephyr/include -I/home/ztest/zephyrproject/zephyr/build/zephyr/include/generated -I/home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/it8xxx2 -I/home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/common/. -isystem /home/ztest/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/include -isystem /home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/include-fixed -Os -imacros /home/ztest/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mabi=ilp32 -march=rv32iac --sysroot=/home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/riscv64-zephyr-elf -xassembler-with-cpp -imacros /home/ztest/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -D_ASMLANGUAGE -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -mcmodel=medlow -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ztest/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/soc/riscv/riscv-ite/common/soc_irq.S.obj -MF zephyr/CMakeFiles/zephyr.dir/soc/riscv/riscv-ite/common/soc_irq.S.obj.d -o zephyr/CMakeFiles/zephyr.dir/soc/riscv/riscv-ite/common/soc_irq.S.obj -c /home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/common/soc_irq.S
/home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/common/soc_irq.S: Assembler messages:
/home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/common/soc_irq.S:43: Error: unrecognized opcode `csrr a0,mcause'
[36/150] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
ccache /home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/ztest/zephyrproject/zephyr/kernel/include -I/home/ztest/zephyrproject/zephyr/arch/riscv/include -I/home/ztest/zephyrproject/zephyr/include -I/home/ztest/zephyrproject/zephyr/build/zephyr/include/generated -I/home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/it8xxx2 -I/home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/common/. -isystem /home/ztest/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/include -isystem /home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/include-fixed -fno-strict-aliasing -Os -imacros /home/ztest/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mabi=ilp32 -march=rv32iac --sysroot=/home/ztest/zephyr-sdk-0.15.0/riscv64-zephyr-elf/riscv64-zephyr-elf -imacros /home/ztest/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -mcmodel=medlow -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ztest/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj -MF zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj -c /home/ztest/zephyrproject/zephyr/lib/os/onoff.c
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h: Assembler messages:
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:239: Error: unrecognized opcode `csrs mstatus,s6'
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:239: Error: unrecognized opcode `csrs mstatus,a4'
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:239: Error: unrecognized opcode `csrs mstatus,a4'
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:239: Error: unrecognized opcode `csrs mstatus,a4'
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:225: Error: unrecognized opcode `csrrc s6,mstatus,8'
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:225: Error: unrecognized opcode `csrrc a2,mstatus,8'
[... 30 lines cut ...]
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:239: Error: unrecognized opcode `csrs mstatus,a5'
/home/ztest/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:225: Error: unrecognized opcode `csrrc a3,mstatus,8'
[... cut ...]
/home/ztest/zephyrproject/zephyr/soc/riscv/riscv-ite/common/soc_common_irq.c:35: Error: unrecognized opcode `csrr a5,mie'
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/ztest/zephyrproject/zephyr/build
ITE@00:00:46:~/zephyrproject/zephyr$
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...) 0.15.0
 - Commit SHA or Version used: 180f50feda2514b792d2588f58907856e0de1461

stephanosio, 2022-08-25:
The main branch is not fully compatible with Zephyr SDK 0.15.0 yet.

This issue is addressed in https://github.com/zephyrproject-rtos/zephyr/pull/49496.

closed 2022-08-25 by stephanosio

vindicatorr, 2023-02-26:
@stephanosio, from the links I walked through, it looks like this is already supposed to be fixed/released/merged, however I'm also encountering this issue as well...
```
$ echo $BOARD
BOARD=esp32c3_devkitm
$ echo $CROSS_COMPILE
<pathTo>/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-
//toolchain_linux-x86_64_riscv64-zephyr-elf.15.2.tar.gz
$ git log
55802e5e8637949b113a9f225c9f9a11f7f9b49d
//zephyr main
2ede05fe756f9985db7a34f909b701c6d316744a
//hal_espressif zephyr
$ rm -rf *; cmake -S ../zephyr/samples/sensor/esp32_temp_sensor -B . -G Ninja
$ ninja zephyr_final
```

```
<pathTo>/zephyr/include/zephyr/arch/riscv/arch.h:252: Error: unrecognized opcode `csrs mstatus,a5'
...
<pathTo>/zephyr/soc/riscv/esp32c3/soc_irq.S:15: Error: unrecognized opcode `csrr a0,mcause'
...
<pathTo>/zephyr/include/zephyr/arch/riscv/arch.h: Assembler messages:
<pathTo>/zephyr/include/zephyr/arch/riscv/arch.h:234: Error: unrecognized opcode `csrrc s0,mstatus,8'
<pathTo>/zephyr/include/zephyr/arch/riscv/arch.h:252: Error: unrecognized opcode `csrs mstatus,s0'
...
...
```
(I don't use `west`)