zephyrproject-rtos/zephyr #49567: Files System mounting with custom offset

nicokorn, 2022-08-26:
**Is your feature request related to a problem? Please describe.**
I am using signed images holding **only** a file system. (I wrote a tool to sign generic binaries). I write the image into a partition called "file-system" and want to mount it:

``
#define LITTLEFS_NODE   DT_NODE_BY_FIXED_PARTITION_LABEL(file_system)

static struct fs_mount_t lfs_storage_mnt = {
    .type = FS_LITTLEFS,
    .fs_data = &file_system,
    .storage_dev = (void *)FLASH_AREA_ID(file_system),
    .mnt_point = "/lfs",
};

...
fs_mount(&lfs_storage_mnt )
...
``

Of course, that does not work. The beginning of the mounting point is not the file system, it is the image header.

**Describe the solution you'd like**
I want to be able to add an offset to a mounting point. As described above the mounting point is the image header thus I need an offset to mount the beginning of the files system starting after the image header.

I invested some time to add an mounting point offset but without success. And in the discord channel the answer to my issue was, that there is no use case, so it is not possible.

**Describe alternatives you've considered**
My workaround is to use nested partitions but that looks nasty. One partition starting with the image header, so I can update it and one nested partition starting with the filesystem for mounting it.

Laczen, 2022-08-27:
@nicokorn, IMO you should aim at dropping the header while writing. The header will become meaningless as soon as you start using the fs anyhow.

nicokorn, 2022-08-27:
> @nicokorn, IMO you should aim at dropping the header while writing. The header will become meaningless as soon as you start using the fs anyhow.

In general that's a good point! But I need the header:
- For image validation (The tail holds SHA256 & RSA Signature fields calculated from header and image).
- My file system holds a web-ui that never changes, once written.
- I need to check regularly for authenticity of the web-ui / file system

Laczen, 2022-08-27:
> * For image validation (The image tail holds SHA256 & RSA Signature fields calculated from header and image).
> * My file system holds a web-ui that never changes, once written.
> * I need to check regularly for authenticity of the web-ui / file system

As said, what is the meaning of the image validation as soon as your filesystem is in use? What is the image tail ? Seems like there are two parts that are not part of the filesystem. Adding these two parts to a partition will be very error prone because they have to respect certain conditions (e.g. The filesystem itself will need to be aligned to the flash sector size, and the same is valid for the header and the trailer). Creating a good solution to cover this would be difficult.

In general it is not a good idea to add a header and a trailer to an image. It would be better to create only a header containing all required validation info and keeping the header separate from the filesystem. Then you first write the header, then the filesystem and only if they are both written a "upgrade" is started.

nicokorn, 2022-08-27:
What is the meaning of the image validation as soon as your filesystem is in use?
Same story as with the mcu firmware which is validated every reboot in the MCUboot bootloader. The file system (Web-UI) must never be changed except on an update.

What is the image tail?
I need to clarify a bit:
The file system is packed into a MCUboot signed image. Details about header and tail here: https://docs.mcuboot.com/design.html. I have written a tool to sign generic binaries. According to the MCUboot specification. 

In general:
With that approach I can use the MCUboot bootloader for validating and updating my Web-UI. Which is very comfortable. Now I can use the MCUboot bootloader to manage any partition, for example a partition holding an FPGA image, or an image for any other chip which is hooked up to my mcu and needs to be updated.

Laczen, 2022-08-27:
@nicokorn, I understand your needs, but from your description you were talking about a filesystem. A filesystem is not a const area in flash like other images. In a filesystem the content changes and as a result the calculated hashes are no longer valid. Also because flash is involved  the filesystem will erase the partition in blocks of erase block size and the filesystem will need to take this into account.

If your filesystem is not a filesystem but a const area (no changes at all to the filesystem), then this solution might work but it is a very limited use case. If the filesystem only consists of const items probably you do not even need a filesystem and you'd better not use littlefs.

nicokorn, 2022-08-27:
> Also because flash is involved the filesystem will erase the partition in blocks of erase block size and the filesystem will need to take this into account.

I need to add:
The Web-UI is a littlefs filesystem. I am using a python lib to convert  a Web-UI build folder into a littlefs binary with specified block-size and block-count.
I thought littlefs takes care of such topics you described, it has even a dynamic wear leveling feature and detects bad blocks.

> If the filesystem only consists of const items probably you do not even need a filesystem and you'd better not use littlefs.

The big advantage of the filesystem is to call the Web-UI files/items in use with civetweb directly over links. That makes it very comfortable. Do you see any other approach?

Laczen, 2022-08-27:
> I need to add: The Web-UI is a littlefs filesystem. I am using a python lib to convert a Web-UI build folder into a littlefs binary with specified block-size and block-count. 
> I think littlefs takes care of such topics you described, it has even a dynamic wear leveling feature and detects bad blocks.

No littlefs does not take care of that. Zephyr configures littlefs to use a specific erase-block-size and specific alignment to this erase-block-size. As you already have a solution you can test if it all keeps working when you are adding some data to the filesystem. Verify that you can still get the correct hashes (and that they match) after a write.

It might be that it works in your specific case (board/flash/...), but in general it will not.

> The big advantage of the filesystem is to call the Web-UI files/items in use with civetweb directly over links. That makes it very comfortable. Do you see any other approach?

I have no experience with civetweb so I can't really propose a different approach.

nicokorn, 2022-08-27:
So far it works like a charm.  Even before my whole update story, my coworker used a partition as littlefs file system together with civetweb and uploading data into the filesystem via curl implementation and deleting files afterwards was never an issue.

Laczen, 2022-08-27:
For sure this works without the update procedure, this uses the partition just as a filesystem.

It is the update that will not work. Test the update procedure, make a change in the filesystem and check the validity of the hash.

nicokorn, 2022-08-27:
It is not the idea to change the Web-UI build in the filesystem. Of course the secure hash algorithm would fail in that case. It is only about to have easy access to the Web-UI items with civetweb.

Laczen, 2022-08-28:
OK, but if the secure hash fails, should mcuboot remove the filesystem ?

And if mcuboot does not remove it, what will happen after there have been so many changes to the filesystem that the first sector of the partition needs to be removed ?

nicokorn, 2022-08-28:
For my case:
The filesystem must be read only thus only be used in read only use cases -> The Web-UI is read only.

In the write case:
The image would be invalid. The MCUboot bootloader checks every boot for validity of multiboot partitions. And if an image is not valid, the bootloader handles according to its specification: **https://docs.mcuboot.com/design.html**
I already tested it last month, the result was, that the bootloader refuses to jump into the application. I did not do any further research on that because it is out of scope at the moment.

IMO:
I would not use a writeable filesystem as an signed image. In the current development that is not feasible. Additional steps would be needed to get that cleanly working.

Laczen, 2022-08-28:
OK, let us summarize:
a. The proposed solution is only tested for read-only filesystems (and even only for littlefs),
b. It does not work for write filesystems as mcuboot refuses to boot,

The use case of ro filesystems is to limited to add such a RFC. Adding a offset to a partition to mount a fs is not without risk of errors and therefore should not be done.

BTW it is possible to get it working with a rw filesystem, but it is so limited by boundary conditions that even this should not be added:
a. The "image" header should be a multiple of the erase-block-size,
b. The "image" size should be a multiple of erase-block-size,
c. The "image" header and trailer should be "subtracted" from the partition instead of only the header.
But even in this case the secure hash fails as soon as the filesystem is written, so even this is not a good solution.

nicokorn, 2022-08-28:
> OK, let us summarize: a. The proposed solution is only tested for read-only filesystems (and even only for littlefs), b. It does not work for write filesystems as mcuboot refuses to boot,
>
> The use [...]

Of course there need to be checks when mounting. When you are defining partition start address in the device tree it gets also checked when you start to build the application and throws an error if alignment is not correct.

> BTW it is possible to get it working with a rw filesystem, but it is so limited by boundary conditions that even this should not be added: a. The "image" header should be a multiple of the [...]

Correct, point A to C are common sense and my signed filesystem of course follows these rules tough I use it as ro. Also here everything can be checked for proper operation when generating the image. That's not rocket sience :-) It is our response to check the code for proper functionality before releasing it.
You forgot point D:
On every filesystem change the TLVs in the signed image need to be updated but that slows performance depending on mcu significantly down. Calculating SHA256 and RSA-2048 signature costs Time.

Laczen, 2022-08-28:
You are making it look more simple and straight forward than it is: the build system has no idea how the filesystem image was generated, so it can't check if there is an agreement in erase-block-size.

Regarding point D: this is not possible as the app has no knowledge of keys so it cannot update the secure TLVs, regardless of what time is needed.

nicokorn, 2022-08-28:
> You are making it look more simple and straight forward than it is: the build system has no idea how the filesystem image was generated, so it can't check if there is an agreement in erase-block-size.
The pre-condition is that the image is properly generated by the tool used for signing.

> Regarding point D: this is not possible as the app has no knowledge of keys so it cannot update the secure TLVs, regardless of what time is needed.
Depends on the application and where the keys are stored. But that's not my intention to have a rw signed filesystem.