zephyrproject-rtos/zephyr #49572: Reproducable buils with MCUboot signing

caco3, 2022-08-26:
We are trying to switch to [reproducable builds](https://reproducible-builds.org/).

As for the zephyr project, this seems to be fine, how ever we found out that the MCUboot signing is not-deterministic:
At least with the suggested steps (see [Generating a new keypair](https://www.mcuboot.com/documentation/readme-zephyr/), they are using a [RSA-PSS](https://en.wikipedia.org/wiki/RSA-PSS) padding scheme which is non-deterministic by design.

We found an [explanation ](https://crypto.stackexchange.com/questions/34554/can-rsa-pss-signing-be-made-deterministic-without-loss-of-security/34573#34573) how to change it to be deterministic but don't like to play around with the signing mechianism by ourself.

Has there ever been a discussion about this?

Laczen, 2022-08-28:
@caco3, maybe you shouldn't try to make the MCUboot signing deterministic. IMO it is needed to look differently at what is the output of a build system, see https://reproducible-builds.org/docs/embedded-signatures/.

henrikbrixandersen, 2022-08-30:
I am not sure this qualifies as a bug.

mmahadevan108, 2022-08-30:
Not a bug

closed 2022-08-30 by mmahadevan108