zephyrproject-rtos/zephyr #49576: tests: kernel: timer: timer_behavior: kernel.timer.timer fails

nordic-piks, 2022-08-26:
**Describe the bug**
The tests/kernel/timer/timer_behavior/kernel.timer.timer fails

Observed for
- nrf9160dk_nrf9160
- nrf9160dk_nrf9160_ns
- nrf52840dk_nrf52840
- nrf5340dk_nrf5340_cpuapp
- nrf5340dk_nrf5340_cpuapp_ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/kernel/timer/timer_behavior -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-08-26T07:49:46.753Z] ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-4233-g8a5d568a7979  ***
[2022-08-26T07:49:46.753Z] Running TESTSUITE timer_behavior
[2022-08-26T07:49:46.753Z] ===================================================================
[2022-08-26T07:49:46.753Z] START - test_periodic_behavior
[2022-08-26T07:49:46.753Z] periodic timer behavior test
[2022-08-26T07:49:46.753Z] periodic timer samples gathered, calculating statistics
[2022-08-26T07:49:46.753Z] timer clock rate 32768, kernel tick rate 32768, expected period 1000 us, samples 10000, period 32.768000 cycles
[2022-08-26T07:49:46.753Z] statistics samples 10000, rollovers 0, mean 1007.083130 us, variance 0.093123 us, stddev 0.305161 us, min 1007.080078 us, max 1037.597656 us
[2022-08-26T07:49:46.753Z] statistics samples 10000, rollovers 0, mean 33.000100 cycles, variance 0.000100 cycles, stddev 0.010000 cycles, min 33 cycles, max 34 cycles
[2022-08-26T07:49:46.753Z] timer start cycle 15030, end cycle 345031, total time 10070831.298828, expected time 10000000.000000, difference 70831.298828
[2022-08-26T07:49:46.753Z] Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/kernel/timer/timer_behavior/src/main.c:176: timer_behavior_test_periodic_behavior: (abs(time_diff_us) < CONFIG_TIMER_TEST_MAX_DRIFT is false)
[2022-08-26T07:49:46.753Z] Drift (in microseconds) outside expected bound
[2022-08-26T07:49:46.753Z] FAIL - test_periodic_behavior in 10.281 seconds
[2022-08-26T07:49:46.753Z] ===================================================================
[2022-08-26T07:49:46.753Z] TESTSUITE timer_behavior failed.
[2022-08-26T07:49:46.753Z] ------ TESTSUITE SUMMARY START ------
[2022-08-26T07:49:46.753Z] SUITE FAIL -   0.00% [timer_behavior]: pass = 0, fail = 1, skip = 0, total = 1 duration = 10.281 seconds
[2022-08-26T07:49:46.753Z] - FAIL - [timer_behavior.test_periodic_behavior] duration = 10.281 seconds
[2022-08-26T07:49:46.753Z] ------ TESTSUITE SUMMARY END ------
[2022-08-26T07:49:46.753Z] ===================================================================
[2022-08-26T07:49:46.753Z] RunID: 6823904787c6f417e7a78c889bbc8520
[2022-08-26T07:49:46.753Z] PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: v3.1.0-4233-g8a5d568a7979

nashif, 2022-08-26:
@teburd  you might want to give some hints here.

teburd, 2022-08-26:
There's a few options that could resolve this issue

1. Fix the timer driver accuracy, so add some tick +/- 1 to get the mean timer time to be more accurate every N timer fires.
2. Add some board/platform specifics such that the test period is a nice divisor for 32768 KHz clock (so update testcase.yaml for nordic platforms), so say 1024 us I believe would work out.
3. Add some board/platform specifics such that the expected accuracy of a 1ms timer is understood to be imperfect, off by 60-70ms per 10s.
4. We change the test such that accuracy is defined in ticks rather than microseconds, but for humans that's less useful IMO. As someone using Zephyr I care about wall clock time accuracy not kernel tick time accuracy and this is what the test validates.

hakehuang, 2022-09-07:
nxp paltfrom also has such issue, e.g. mimxrt1060_evk
```
*** Booting Zephyr OS build zephyr-v3.1.0-4786-ge9e7ad388da7  ***
Running TESTSUITE timer_behavior
===================================================================
START - test_periodic_behavior
periodic timer behavior test
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 4096, expected period 1000 us, samples 10000, period 32.768000 cycles
statistics samples 10000, rollovers 0, mean 1220.697021 us, variance 0.186227 us, stddev 0.431541 us, min 1190.185547 us, max 1220.703125 us
statistics samples 10000, rollovers 0, mean 39.999800 cycles, variance 0.000200 cycles, stddev 0.014141 cycles, min 39 cycles, max 40 cycles
timer start cycle 658, end cycle 400656, total time 12206970.214844, expected time 10000000.000000, difference 2206970.214844
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/main.c:166: timer_behavior_test_periodic_behavior: (max_us <= max_us_bound is false)
Longest timer period too long (off by more than 10%)
FAIL - test_periodic_behavior in 12.291 seconds
===================================================================
TESTSUITE timer_behavior failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [timer_behavior]: pass = 0, fail = 1, skip = 0, total = 1 duration = 12.291 seconds
- FAIL - [timer_behavior.test_periodic_behavior] duration = 12.291 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: e1600f406ed38b97eb0ec0302b8d2b3d
PROJECT EXECUTION FAILED

```

closed 2022-09-21 by fabiobaltieri