zephyrproject-rtos/zephyr #49584: STM32WB55 Failed read remote feature, remote version and LE set PHY

Aljohcomb, 2022-08-26:
**Describe the bug**
Running Bluetooth application with SMP in peripheral mode.
When connected to from remote device errors codes are received indicating that connection procedure failed.
However the device still connects.
Following pairing attempts succeeds.
 
When using the device for communication it is unreliable and will connect/disconnect immediately.

Please also mention any information which could help others to understand
the problem you're facing:
 Platform: STM32WB55CG

I have tried to enable configuration for ctlr phy in prj.conf:
\`\`\`
CONFIG_BT_CTLR=y
CONFIG_BT_LL_SW_SPLIT=y
CONFIG_BT_PHY_UPDATE=y
CONFIG_BT_HCI=y
CONFIG_BT_DRIVERS=y 
\`\`\`

The prj.conf for the test application used to replicate this:
\`\`\`
CONFIG_LOG=y
CONFIG_APP_LOG_LEVEL_DBG=n

CONFIG_BT=y
CONFIG_BT_BAS=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_DEBUG_SMP=y
CONFIG_BT_DEBUG_HCI_CORE=n
CONFIG_BT_DEVICE_NAME="LE PHY TEST"
CONFIG_BT_DEVICE_NAME_DYNAMIC=y
CONFIG_BT_DIS=y
CONFIG_BT_LOG_LEVEL_INF=y
CONFIG_BT_LOG_LEVEL_WRN=n
CONFIG_BT_LOG_LEVEL_ERR=n
CONFIG_BT_REMOTE_VERSION=y
CONFIG_BT_SMP=y

CONFIG_CONSOLE=y
CONFIG_DEBUG_OPTIMIZATIONS=y
CONFIG_LOG=y
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_SERIAL=y
CONFIG_UART_CONSOLE=y
\`\`\`

I have also tested different stack from ST, HCI, HCI-extended, FULL, etc. However, from another issue [46400](https://github.com/zephyrproject-rtos/zephyr/issues/46400) and zephyr documentation _stm32wb5x_BLE_HCILayer_extended_fw.bin_ 13.3 seems to be the correct one.

I have also tried with using extended advertising, improvement.

I have also build and run the peripheral_gatt_write sample and have the same issue there with the exception of Failed to read remote version.

**To Reproduce**
Steps to reproduce the behavior:
1. build application
2. cmake -DBOARD=board\_xyz
3. make
4. Connect to device from remote.
5. Pair to device.
6. See error

**Expected behavior**
Connection successful without errors.

**Impact**
Breaks connectability to especially with certain chipsets on remote device, e.g. intel AX201.
Crashes upon pairing.
[... 340 lines cut ...]
mtu_exchange: Current MTU = 23
mtu_exchange: Exchange MTU...
Updated MTU: TX: 247 RX: 247 bytes
mtu_exchange_cb: MTU exchange successful (247)
le_param_updated: int 0x0006 lat 0 to 500
le_param_updated: int 0x0024 lat 0 to 500
write_cmd_cb: count= 0, len= 0, rate= 0 bps.
[00:00:12.223,000] <wrn> hci_ipm: Invalid peer addr FF:FF:FF:FF:FF:FF (random)
[00:00:12.236,000] <wrn> bt_hci_core: opcode 0x2016 status 0x3a
[00:00:12.236,000] <err> bt_hci_core: Failed read remote features (-5)
[00:00:12.237,000] <wrn> bt_hci_core: opcode 0x2032 status 0x0c
[00:00:12.237,000] <err> bt_hci_core: Failed LE Set PHY (-5)
\`\`\`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: [88aa873](https://github.com/zephyrproject-rtos/zephyr/commit/88aa873fbc544b99c1e48e37d5c74662f90b3de0)

**Additional Context**
When looking up the line from the crash with: 
addr2line -e zephyr.elf 0x08002d48

It leads here:
modules/crypto/tinycrypt/lib/source/aes_encrypt.c:71

erwango, 2022-08-26:
@Aljohcomb Thanks for raising this point.
We'll try to investigate based on the trace provided.

Note, following Kconfig flags are not compatible with STM32WB. I think they are discarded in the build, but it would be cleaner not to disable them:
```
CONFIG_BT_CTLR=y
CONFIG_BT_LL_SW_SPLIT=y
```

Aljohcomb, 2022-08-29:
> @Aljohcomb Thanks for raising this point. We'll try to investigate based on the trace provided.
> 
> Note, following Kconfig flags are not compatible with STM32WB. I think they are discarded in the build, but it would be cleaner not to disable them:
> 
> ```
> CONFIG_BT_CTLR=y
> CONFIG_BT_LL_SW_SPLIT=y
> ```

Thank you for the response! 
Those configuration options are not in the actual project, only given as examples of somethings I've tried. But you are correct of course.

The issue is very problematic for our use of the STM32 since the bluetooth does not work at all. I would be very happy to provide more logs or testing if there are any ideas.

erwango, 2022-09-01:
> I have also tested different stack from ST, HCI, HCI-extended, FULL, etc. However, from another issue https://github.com/zephyrproject-rtos/zephyr/issues/46400 and zephyr documentation stm32wb5x_BLE_HCILayer_extended_fw.bin 13.3 seems to be the correct one.

Note that for general usage, `stm32wb5x_BLE_HCILayer_fw.bin` is the one to be used.

Aljohcomb, 2022-09-01:
> > I have also tested different stack from ST, HCI, HCI-extended, FULL, etc. However, from another issue #46400 and zephyr documentation stm32wb5x_BLE_HCILayer_extended_fw.bin 13.3 seems to be the correct one.
> 
> Note that for general usage, `stm32wb5x_BLE_HCILayer_fw.bin` is the one to be used.

I have also tried older versions of that stack with no major difference.

erwango, 2022-09-01:
@Aljohcomb I'm not able to reproduce this issue on my side (I'm testing current main with V13.2.0, stm32wb5x_BLE_HCILayer_fw.bin). But maybe I'm lacking a remote device that allows to reproduce the issue.
In order to help on investigation, would  you be able to try to reproduce the issue using full STM32Cube application ?

Aljohcomb, 2022-09-01:
> @Aljohcomb I'm not able to reproduce this issue on my side (I'm testing current main with V13.2.0, stm32wb5x_BLE_HCILayer_fw.bin). But maybe I'm lacking a remote device that allows to reproduce the [...]

The issue is mainly prevalent when remote is using Intels AX201 chipset. However I get the faults no matter remote device.
I did manage just now to get the set phy working with:
`
CONFIG_BT_AUTO_PHY_UPDATE=n
`
I this makes the remote decide on PHY instead of the STM32 in this case.
Also fixed the crash.
The error of failed to read remote features still remains.

Connection towards the AX201 chipset is still not possible.

I could run a cube application for testing for sure, might take some time to setup unless there is something ready to go.

erwango, 2022-09-01:
> I did manage just now to get the set phy working with:
CONFIG_BT_AUTO_PHY_UPDATE=n
I this makes the remote decide on PHY instead of the STM32 in this case. Also fixed the crash.

Agreed. I'm checking with our BLE experts on the consequence on deactivating this option.
Note that I don't get crash even when this is on.

> The error of failed to read remote features still remains.

Ok, I can see this error as well. We need to investigate that also, but I don't think this is related with your issue.

> I could run a cube application for testing for sure, might take some time to setup unless there is something ready to go.

Here is the equivalent Cube Application https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_HeartRate

> @erwango Another theory I have is regarding clocks. Is your STM32 board running BT with LSE clock?
We don't have one and are using HSE currently. I'm speculating that this could cause synchronization issues.

You might be interested by https://github.com/zephyrproject-rtos/zephyr/issues/31959.
This is on my TODO list for long time If this is helping in your case, I'll try to speed things up.

Aljohcomb, 2022-09-01:
> Here is the equivalent Cube Application https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_HeartRate
> 
I will take a look at this and get it running on our HW.

> You might be interested by #31959. This is on my TODO list for long time If this is helping in your case, I'll try to speed things up.
>
Thanks, will read through this.

Will try to get back on this asap.

Aljohcomb, 2022-09-02:
Regarding [#31959](https://github.com/zephyrproject-rtos/zephyr/issues/31959) we have made following patch to _ipm_stm32wb.c_.

`Subject: [PATCH] drivers: hci: stm32_ipm: Add HSE as BLE RF wake up source

LSI is not a valid RF wake up clock source anymore.
For more info see, STM32WB55 errata chap. 2.2.1.
---
 drivers/bluetooth/hci/ipm_stm32wb.c | 14 ++++++++++----
 1 file changed, 10 insertions(+), 4 deletions(-)

diff --git a/drivers/bluetooth/hci/ipm_stm32wb.c b/drivers/bluetooth/hci/ipm_stm32wb.c
index a487b8b019..13bb8a6e58 100644
--- a/drivers/bluetooth/hci/ipm_stm32wb.c
+++ b/drivers/bluetooth/hci/ipm_stm32wb.c
@@ -69,6 +69,8 @@ struct aci_set_ble_addr {
 #define HCI_CONFIG_DATA_PUBADDR_OFFSET		0
 #define HCI_CONFIG_DATA_RANDOM_ADDRESS_OFFSET	0x2E
 
+#define CFG_BLE_HSE_SOURCE 1
+
 static bt_addr_t bd_addr_udn;
 
 /* Rx thread definitions */
@@ -92,7 +94,11 @@ static void stm32wb_start_ble(void)
 	    CFG_BLE_MAX_ATT_MTU,
 	    CFG_BLE_SLAVE_SCA,
 	    CFG_BLE_MASTER_SCA,
+	    #if STM32_LSE_CLOCK
 	    CFG_BLE_LSE_SOURCE,
+	    #else
+	    CFG_BLE_HSE_SOURCE,
+	    #endif
 	    CFG_BLE_MAX_CONN_EVENT_LENGTH,
 	    CFG_BLE_HSE_STARTUP_TIME,
 	    CFG_BLE_VITERBI_MODE,
@@ -406,13 +412,13 @@ static void start_ble_rf(void)
 	/* Switch OFF LSI */
 	LL_RCC_LSI2_Disable();
 #else
-	LL_RCC_LSI2_Enable();
-	while (!LL_RCC_LSI2_IsReady()) {
+	LL_RCC_HSE_Enable();
+	while (!LL_RCC_HSE_IsReady()) {
 	}
 
 	/* Select wakeup source of BLE RF */
-	LL_RCC_SetRFWKPClockSource(LL_RCC_RFWKP_CLKSOURCE_LSI);
-	LL_RCC_SetRTCClockSource(LL_RCC_RTC_CLKSOURCE_LSI);
+	LL_RCC_SetRFWKPClockSource(LL_RCC_RFWKP_CLKSOURCE_HSE_DIV1024);
+	LL_RCC_SetRTCClockSource(LL_RCC_RTC_CLKSOURCE_HSE_DIV32);
 #endif
 
 	/* Set RNG on HSI48 */
-- `
To get HSE working as clock. Due to an errata from ST we found out that the LSI clock does not work.

erwango, 2022-09-02:
> Regarding https://github.com/zephyrproject-rtos/zephyr/issues/31959 we have made following patch to ipm_stm32wb.c.

Did it helped in fixing your issue ?

Aljohcomb, 2022-09-02:
> > Regarding #31959 we have made following patch to ipm_stm32wb.c.
> 
> Did it helped in fixing your issue ?

It did fix the issue, we have used this patch for a while.
Might be important to point out that this issue is mostly towards a specific inter chip, AX201.

I'm attaching a log file from a wireshark session were the issue is observer. Can provide longer recordings if necessary.
[STM32_read_remote_features_fail.json.txt](https://github.com/zephyrproject-rtos/zephyr/files/9476720/STM32_read_remote_features_fail.json.txt)

Have not done the cube project yet.

erwango, 2022-09-02:
@Aljohcomb Sorry but I'm lost. Which issue did it fixed ? As I understand this is not current issue.

Aljohcomb, 2022-09-02:
> @Aljohcomb Sorry but I'm lost. Which issue did it fixed ? As I understand this is not current issue.

Sorry that was unclear from my side. It is not the current issue. It was only in regards to the issue you linked yesterday.
I thought that maybe our use of HSE instead of LSE could have an effect on the syncronisation which was a longshot theory for the read remote features fail.

KenthJohan, 2022-09-26:
I have the same problem when connecting then disconnecting then connecting then disconnecting then connecting:

```
CONFIG_BT_AUTO_PHY_UPDATE=y
```
```ansi
*** Booting Zephyr OS build v3.2.0-rc3-24-g8364715998c9  ***
Bluetooth initialized        
Advertising successfully started
[00:00:00.346,000] [0m<inf> bt_hci_core: Identity: 02:80:E1:00:00:00 (public)[0m
[00:00:00.346,000] [0m<inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x00
30[0m
[00:00:00.346,000] [0m<inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126[0m
Connected
[00:00:04.258,000] [1;33m<wrn> bt_hci_core: opcode 0x2016 status 0x3a[0m
[00:00:04.258,000] [1;31m<err> bt_hci_core: Failed read remote features (-5)[0m
[00:00:04.258,000] [1;33m<wrn> bt_hci_core: opcode 0x2032 status 0x0c[0m
[00:00:04.258,000] [1;31m<err> bt_hci_core: Failed LE Set PHY (-5)[0m
Disconnected (reason 0x13)
Connected
[00:00:09.668,000] [1;33m<wrn> bt_hci_core: opcode 0x2016 status 0x3a[0m
[00:00:09.668,000] [1;31m<err> bt_hci_core: Failed read remote features (-5)[0m
[00:00:09.669,000] [1;33m<wrn> bt_hci_core: opcode 0x2032 status 0x0c[0m
[00:00:09.669,000] [1;31m<err> bt_hci_core: Failed LE Set PHY (-5)[0m
Disconnected (reason 0x13)              
Connected                                                                              
[00:00:14.150,000] [1;33m<wrn> bt_hci_core: opcode 0x2016 status 0x3a[0m
[00:00:14.150,000] [1;31m<err> bt_hci_core: Failed read remote features (-5)[0m
[00:00:14.151,000] [1;33m<wrn> bt_hci_core: opcode 0x2032 status 0x0c[0m
[00:00:14.151,000] [1;31m<err> bt_hci_core: Failed LE Set PHY (-5)[0m
```
After the last `Connected` using `nRF Connect for Mobile` it fails to connect and there is no error printing from zephyr log.

<hr>

```
CONFIG_BT_AUTO_PHY_UPDATE=n
```
```ansi
*** Booting Zephyr OS build v3.2.0-rc3-24-g8364715998c9  ***
Bluetooth initialized
Advertising successfully started
[00:00:00.346,000] [0m<inf> bt_hci_core: Identity: 02:80:E1:00:00:00 (public)[0m
[00:00:00.346,000] [0m<inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x00
30[0m
[00:00:00.346,000] [0m<inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126[0m
Connected
[00:00:07.251,000] [1;33m<wrn> bt_hci_core: opcode 0x2016 status 0x3a[0m
[00:00:07.251,000] [1;31m<err> bt_hci_core: Failed read remote features (-5)[0m
Disconnected (reason 0x13)
Connected                        
[00:00:15.788,000] [1;33m<wrn> bt_hci_core: opcode 0x2016 status 0x3a[0m
[00:00:15.788,000] [1;31m<err> bt_hci_core: Failed read remote features (-5)[0m
Disconnected (reason 0x13)                    
Connected
[00:00:19.577,000] [1;33m<wrn> bt_hci_core: opcode 0x2016 status 0x3a[0m
[00:00:19.577,000] [1;31m<err> bt_hci_core: Failed read remote features (-5)[0m
[00:00:20.166,000] [1;31m<err> os: ***** BUS FAULT *****[0m
[00:00:20.166,000] [1;31m<err> os:   Imprecise data bus error[0m
[00:00:20.166,000] [1;31m<err> os: r0/a1:  0x200018e0  r1/a2:  0x200300d8  r2/a3:  0x2003d238[0m
[00:00:20.166,000] [1;31m<err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x08001ce7[0m
[00:00:20.166,000] [1;31m<err> os:  xpsr:  0x61000000[0m
[00:00:20.166,000] [1;31m<err> os: Faulting instruction address (r15/pc): 0x08001ce6[0m
[00:00:20.166,000] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
[00:00:20.166,000] [1;31m<err> os: Current thread: 0x200012b8 (unknown)[0m
[00:00:20.237,000] [1;31m<err> os: Halting system[0m
```
After the last `Connected` using `nRF Connect for Mobile` it fails to connect but now zephyr log prints some error at least.

erwango, 2022-09-26:
> I have the same problem when connecting then disconnecting then connecting then disconnecting then connecting:

This is likely a different error. Would you mind raising a new issue and provide all information on the SW you're using ?
Also note that AUTO_PHY_UPDATE is not supported by STM32WB BLE Controller. Config should be `CONFIG_BT_AUTO_PHY_UPDATE=n`.

KenthJohan, 2022-09-26:
> This is likely a different error. Would you mind raising a new issue and provide all information on the SW you're using ? Also note that AUTO_PHY_UPDATE is not supported by STM32WB BLE Controller. [...]

https://github.com/zephyrproject-rtos/zephyr/issues/50655

erwango, 2022-10-12:
btmon extract:
```
> ACL Data RX: Handle 2049 flags 0x02 dlen 11                                      #157 12.112500
      ATT: Read By Type Request (0x08) len 6
        Handle range: 0x0027-0x0029
        Attribute type: Include (0x2802)
< ACL Data TX: Handle 2049 flags 0x00 dlen 9                                       #158 12.114900
      ATT: Error Response (0x01) len 4
        Read By Type Request (0x08)
        Handle: 0x0027
        Error: Attribute Not Found (0x0a)

```

erwango, 2022-10-18:
@jhedberg question for you on 2 specific errors we're facing here (see btmon extract above):

First issue: "LE Read Remote Used Features": Controller Busy
I wonder what is the expected behavior in that case. Is host supposed to renew the request some time later ?
I haven't found anything in the specification.

In any case, there is nothing controller can do for that ...

Second issue:  "LE Set PHY" command disallowed:
Actually controller perform the PHY update on his own and doesn't expect host to trigger the procedure.
I can perform the following configuration for STM32WB (either in_deconfig or .deconfig file):
```
CONFIG_BT_PHY_UPDATE=y
CONFIG_BT_USER_PHY_UPDATE=n
CONFIG_BT_AUTO_PHY_UPDATE=n
```
But this won't prevent any wrong configuration happening later by the user. 
Is it the best option I have?

jhedberg, 2022-10-18:
@cvinayak any thoughts on the questions @erwango raises in the comment above?

cvinayak, 2022-10-20:
@erwango The device is a peripheral and the controller does not support both the peripheral initiated feature request and PHY update control procedures.

Zephyr Host will not initiate a remote feature request if the controller does not support it, see the code:

https://github.com/zephyrproject-rtos/zephyr/blob/07de2ffbb01a4633cdec7e48190ae369ecc37c9b/subsys/bluetooth/host/hci_core.c#L1010-L1017

The same with PHY update procedure:

https://github.com/zephyrproject-rtos/zephyr/blob/07de2ffbb01a4633cdec7e48190ae369ecc37c9b/subsys/bluetooth/host/hci_core.c#L1027-L1036

It appear the controller is at fault as it returns the features as supported when host asks the local supported features:

https://github.com/zephyrproject-rtos/zephyr/blob/07de2ffbb01a4633cdec7e48190ae369ecc37c9b/subsys/bluetooth/host/hci_core.c#L2922-L2930

Users using this controller has to use `CONFIG_BT_PHY_UPDATE=n` and live with error with the remote feature request failure as there is no Kconfig to disable (as Host trusts a Conformant  Controller to return correct local supported features).

jhedberg, 2022-10-20:
@cvinayak @erwango btw, we have the concept of "quirks" (special flags) that HCI drivers can set, in case it's known that a controller misbehaves in a specific way and it can't be properly fixed worked around in any other reasonable way.

Just as I wrote the above, I realized that the quirks are not implemented in an ideal way for Zephyr. The design is inherited from Linux, where controllers can come and go dynamically. With Zephyr a more optimal solution would be promptless Kconfig symbols that the HCI driver would `select`, and the host would then have `if (IS_ENABLED(...)) {` branches for them. The only case where Zephyr might need runtime quirks is if discovering the need for one would require doing e.g. some HCI communication first.

Anyway, considering the "add a new quirk" option should really be the last resort. If you can get the controller firmware fixed, or work around the issue in some other clean way, then that would be better.

erwango, 2022-10-20:
@jhedberg, @cvinayak Thanks for your inputs.

About the `remote feature request` point, the fact is that the controller actually supports it. Problem is that the request is received at a point where it is not able to process it and hence returns "Controller Busy".
"Controller Busy" is an allowed response distinct from "Command Disallowed", so one can expect a specific treatment on host side. This being said, I haven't found anything in the specification about the behavior expected in this case, but I would have expected the host to have a new try later. Are we in a grey area here ?

About the  `phy update request`, indeed `CONFIG_BT_PHY_UPDATE=n` should be used, but there is no way to prevent user to set it otherwise. As mentioned, it would be nice to have promptless Kconfig symbols which would prevent this possibility. Anyway, I'm not in position to implement this atm.

hlukasz, 2022-10-21:
I am facing the same issue in my STM32WB board - reading remote features returns "controller busy" error. I did a quick workaround to retry this command with 100ms delay, after around 5 tries it finally passes. The only concern is if adding about 0,5s delay in `conn_auto_initiate()` does not affect something else in negative way.

Here is the patch
```diff
diff --git a/subsys/bluetooth/host/hci_core.c b/subsys/bluetooth/host/hci_core.c
index b53f091099..4f3fda80fd 100644
--- a/subsys/bluetooth/host/hci_core.c
+++ b/subsys/bluetooth/host/hci_core.c
@@ -334,6 +334,8 @@ int bt_hci_cmd_send_sync(uint16_t opcode, struct net_buf *buf,
                switch (status) {
                case BT_HCI_ERR_CONN_LIMIT_EXCEEDED:
                        return -ECONNREFUSED;
+               case BT_HCI_ERR_CONTROLLER_BUSY:
+                       return -EBUSY;
                default:
                        return -EIO;
                }
@@ -997,7 +999,7 @@ static bool skip_auto_phy_update_on_conn_establishment(struct bt_conn *conn)
 
 static void conn_auto_initiate(struct bt_conn *conn)
 {
-       int err;
+       int err, retries;
 
        if (conn->state != BT_CONN_CONNECTED) {
                /* It is possible that connection was disconnected directly from
@@ -1010,7 +1012,14 @@ static void conn_auto_initiate(struct bt_conn *conn)
        if (!atomic_test_bit(conn->flags, BT_CONN_AUTO_FEATURE_EXCH) &&
            ((conn->role == BT_HCI_ROLE_CENTRAL) ||
             BT_FEAT_LE_PER_INIT_FEAT_XCHG(bt_dev.le.features))) {
-               err = hci_le_read_remote_features(conn);
+               retries = 10;
+               do {
+                       err = hci_le_read_remote_features(conn);
+                       if (err != -EBUSY) {
+                               break;
+                       }
+                       k_sleep(K_MSEC(100));
+               } while (--retries > 0);
                if (err) {
                        BT_ERR("Failed read remote features (%d)", err);
                }
```

closed 2022-11-02 by jhedberg

erwango, 2025-07-15:
> is there any way we can get this in tree? thank you.

Please open a new issue.
@asm5878 FYI

gadifrish, 2025-07-17:
I’m trying to run the Bluetooth shell test on a Nucleo-WB55RG with **HSE** (no LSE crystal) under Zephyr, but I’m stuck at connection time.
I followed the steps and disabled low-power modes. `CONFIG_PM=n` 

My DTS overlay:

```dts
&ble_rf {
    status = "okay";
    clocks = <&rcc STM32_CLOCK(AHB3, 20)>,
             <&rcc STM32_SRC_HSE  RFWKP_SEL(3)>;
};

&rtc {
    status = "okay";
    clocks = <&rcc STM32_CLOCK_BUS_APB1 0x100000>,
             <&rcc STM32_SRC_HSE      RTC_SEL(3)>;
};```

**With LSE:** init → advertising → connect → feature exchange → works.

**With HSE:** init succeeds, advertising starts, you see “Connected / Remote features” but then get Disconnected, reason 0x08 immediately.

1. Is there a known working DTS overlay for WB55 + HSE in Zephyr?
2. Are there any extra Kconfig flags or clock/divider settings I’ve missed?
3. Any pointers to sample code or tests that successfully use the HSE path?

erwango, 2025-07-17:
@gadifrish Please open a dedicated issue or discussion.

PR #51519 (2022-11-02): soc: stm32wb: Set BT_USER_PHY_UPDATE as supported
STM32WB Controller supports application to initiate the "PHY Update Procedure" (BT_USER_PHY_UPDATE) while it doesn't support it to be automatically triggered on connection establishment (BT_AUTO_PHY_UPDATE).

Default BT_USER_PHY_UPDATE to true, which automatically defaults BT_AUTO_PHY_UPDATE to false.

Partially fixes #49584