zephyrproject-rtos/zephyr #49587: cross-compile toolchain variant doesn't working with multilib toolchain

evgeniy-paltsev, 2022-08-26:
**Describe the bug**
In case of `ZEPHYR_TOOLCHAIN_VARIANT=cross-compile` we don't set `SYSROOT_DIR` cmake variable, so we don't execute
block of code which sets `LIBC_LIBRARY_DIR` variable based on the toolchain C flags (including mcpu flag):

https://github.com/zephyrproject-rtos/zephyr/blob/88aa873fbc544b99c1e48e37d5c74662f90b3de0/cmake/compiler/gcc/target.cmake#L74-L88

So, in case of multilib cross-compile toolchain we always use default libraries set no mater which mcpu value we've provided. Obviously it's incorrect.

**Impact**
Implicit & unexpected usage of libraries which are build for different mcpu than was specified. 
If we are lucky we may get compile error - for example if we got something not linkable, like 64bit libraries from toolchain and 32 bit object files from rest of zephyr. However we may get successful linkage and use some not optimal libraries (i.e buuild for processor which use smaller instruction set) or get runtime issues.

**possible solutions**

* add additional environment variable for cross-compile toolchain variant, which will be pointing to correct SYSROOT_DIR path.
* try to extract SYSROOT_DIR path from the `CROSS_COMPILE` variable. I've checked the value of `CROSS_COMPILE` I've previously used - it doesn't seems to be feasible for these toolchains I.e:
```
CROSS_COMPILE=/home/user/dev/arc64-unknown-elf-1205/arc64-unknown-elf/bin/arc64-snps-elf- 
SYSROOT_DIR should be /home/user/dev/arc64-unknown-elf-1205/arc64-unknown-elf/arc64-unknown-elf/

CROSS_COMPILE=/global/tools/Linux/arc_gnu_2021.09_prebuilt_elf32_le/bin/arc-elf32-
SYSROOT_DIR should be /global/tools/Linux/arc_gnu_2021.09_prebuilt_elf32_le/arc-snps-elf/
```

evgeniy-paltsev, 2022-08-26:
BTW, gcc has option with promising name - `-print-sysroot` - `Display the target libraries directory.`. However it always doesn't print anything for all toolchains I've tried (no matter ARC ones or host toolchains, like X86_64)

tejlmand, 2022-09-23:
> add additional environment variable for cross-compile toolchain variant, which will be pointing to correct SYSROOT_DIR path.

@evgeniy-paltsev it's already possible today to use `-DSYSROOT_DIR=<sysroot-dir>`

stephanosio, 2022-09-23:
@evgeniy-paltsev FYI, https://github.com/zephyrproject-rtos/zephyr/pull/50584#pullrequestreview-1118344285. There likely is a configuration issue with your toolchain build.

PR #50584 (2022-09-23): cmake: detect sysroot for cross-compile toolchains
Fixes: #49587

Try to detect sysroot for cross-compile toolchain if not specified by the user with `-DSYSROOT_DIR=<path>`.

First, the C compiler is asked if it knows its sysroot, some C compilers are able to return the sysroot, as an example:
```
$ arm-none-eabi-gcc -print-sysroot
<path>/bin/../arm-none-eabi
$
```

but majority of gcc-based C compilers seems to return empty string:
```
$ arm-zephyr-eabi-gcc --print-sysroot
$
```

in such cases, the cross-compiler target CMake file will try to discover the sysroot by searching for libc.a, star [... cut]
commit fd9e5429 (2022-09-23): cmake: detect sysroot for cross-compile toolchains

Fixes: #49587

Try to detect sysroot for cross-compile toolchain if not specified by
the user with `-DSYSROOT_DIR=<path>`.

First, the C compiler is asked if it knows its sysroot, some C compilers
are able to return the sysroot, as an example:
> $ arm-none-eabi-gcc -print-sysroot
> <path>/bin/../arm-none-eabi
> $

but majority of gcc-based C compilers seems to return empty string:
> $ arm-zephyr-eabi-gcc --print-sysroot
> $

in such cases, the cross-compiler target CMake file will try to discover
the sysroot by searching for libc.a, starting  [... cut]
closed 2022-09-23 by carlescufi