zephyrproject-rtos/zephyr #49588: Json parser is incorrect with undefined parameter

ashegheparyshun, 2022-08-26:
**Describe the bug**
The JSON parser can not ignore undefined fields in the JSON.  

I have a json named config to parse structured like this:
{
	"configVersion": "2.9.0",
	"updateTime": "2022-07-23T16:00:00",
	"network": {
		"ssid": "Sarveen_wifi",
		"psk": "$arveenTech",
		"country": "CN"
	},
	"systemSettings": {
		"system": "",
		"beaconFunctionality": {
			"enabled": true,
			"advertisingInterval": 100
		}
	},
	"control": [
		{ 
		"n2":0, 
		"n1":0,
		"n0":1,
		"d2":0,
		"d1":0,
		"d0":0
		}
	]
}
when I define the descriptor (and all structures included) like this:
static const struct json_obj_descr sysConfigDescr[] = {

	JSON_OBJ_DESCR_PRIM(struct SystemConfig, configVersion,
			    JSON_TOK_STRING),
    JSON_OBJ_DESCR_PRIM(struct SystemConfig, updateTime, JSON_TOK_STRING),
	JSON_OBJ_DESCR_OBJECT(struct SystemConfig, network, networkDescr),
	JSON_OBJ_DESCR_OBJECT(struct SystemConfig, systemSettings, systemSettingsDescr),
    JSON_OBJ_DESCR_OBJ_ARRAY(struct SystemConfig, control, MAX_NUMBER_OF_SENSORS, sensorsCount, sensorDescr, ARRAY_SIZE(sensorDescr)),
};
Everything perfect but in fact I don't need the systemSettings field. So I removed this field so the descriptor changed to 
static const struct json_obj_descr sysConfigDescr[] = {

	JSON_OBJ_DESCR_PRIM(struct SystemConfig, configVersion,
			    JSON_TOK_STRING),
    JSON_OBJ_DESCR_PRIM(struct SystemConfig, updateTime, JSON_TOK_STRING),
	JSON_OBJ_DESCR_OBJECT(struct SystemConfig, network, networkDescr),
    JSON_OBJ_DESCR_OBJ_ARRAY(struct SystemConfig, control, MAX_NUMBER_OF_SENSORS, sensorsCount, sensorDescr, ARRAY_SIZE(sensorDescr)),
};
In this situation the field before "systemSettings" are parsed correctly but the "control" field is incorrect. The result for the mentioned json is:
ver:2.9.3
 uptime: 2022-07-23T16:00:00
country:CN
psk:$arveenTech
, ssid:Sarveen_wifi, d0: 928336364, n0: -1582706018
As you can see d0 should be 0 but the parser considered it: 928336364 and n0 should be 1 but the parser considered it: -1582706018
Any idea?

**To Reproduce**
Steps to reproduce the behavior:
0. define the mentioned JSON above.
1. define the structures like below.
`struct Network
{
    const char *ssid;
    const char *psk;
    const char *country;
};`
`
struct SensorsCoeff
{
    int n2, n1, n0, d2, d1, d0;
};
`
`struct SystemConfig
{
    const char *configVersion;
    const char *updateTime;
    struct Network network;
    struct SensorsCoeff control[MAX_NUMBER_OF_SENSORS];
    size_t sensorsCount;
};`
2. Define the descriptors like below
`static const struct json_obj_descr networkDescr[] = {

	JSON_OBJ_DESCR_PRIM(struct Network, ssid, JSON_TOK_STRING),
	JSON_OBJ_DESCR_PRIM(struct Network, psk, JSON_TOK_STRING),
	JSON_OBJ_DESCR_PRIM(struct Network, country, JSON_TOK_STRING),
};
static const struct json_obj_descr sensorDescr[] = {
	JSON_OBJ_DESCR_PRIM(struct SensorsCoeff, n2, JSON_TOK_NUMBER),
	JSON_OBJ_DESCR_PRIM(struct SensorsCoeff, n1, JSON_TOK_NUMBER),
	JSON_OBJ_DESCR_PRIM(struct SensorsCoeff, n0, JSON_TOK_NUMBER),
	JSON_OBJ_DESCR_PRIM(struct SensorsCoeff, d2, JSON_TOK_NUMBER),
	JSON_OBJ_DESCR_PRIM(struct SensorsCoeff, d1, JSON_TOK_NUMBER),
	JSON_OBJ_DESCR_PRIM(struct SensorsCoeff, d0, JSON_TOK_NUMBER),
};
static const struct json_obj_descr sysConfigDescr[] = {

	JSON_OBJ_DESCR_PRIM(struct SystemConfig, configVersion,
			    JSON_TOK_STRING),
    JSON_OBJ_DESCR_PRIM(struct SystemConfig, updateTime, JSON_TOK_STRING),
	JSON_OBJ_DESCR_OBJECT(struct SystemConfig, network, networkDescr),
    JSON_OBJ_DESCR_OBJ_ARRAY(struct SystemConfig, control, MAX_NUMBER_OF_SENSORS, sensorsCount, sensorDescr, ARRAY_SIZE(sensorDescr)),
};
`
3. Parse the JSON using 
`
	struct SystemConfig *config =
		(struct SystemConfig *)k_malloc(sizeof(struct SystemConfig));
	int ret =
		json_obj_parse(json, strlen(json), sysConfigDescr,
			       ARRAY_SIZE(sysConfigDescr), config);
`
5. Print the result:
`	printk("ver:%s\n uptime: %s\ncountry:%s\npsk:%s\n, ssid:%s, d0: %d, n0: %d\n\n\n",
	config->configVersion, config->updateTime, config->network.country, config->network.psk,
	config->network.ssid, config->control[0].d0, config->control[0].n0);
`
**Expected behavior**
d0 should be 0, and n0 should be 1.

**Logs and console output**
`‍ver:2.9.3
 uptime: 2022-07-23T16:00:00
country:CN
psk:$arveenTech
, ssid:Sarveen_wifi, d0: 928336364, n0: -1582706018`

**Environment (please complete the following information):**
 - OS: Ubuntu 20
 - Toolchain Zephyr SDK 0.14.2

mmahadevan108, 2022-08-30:
cc @mrfuchs

mrfuchs, 2022-10-01:
@ashegheparyshun, would you give #47989 a try?

closed 2022-12-15 by github-actions[bot]

erickcinco, 2023-02-13:
Will this be revisited?