zephyrproject-rtos/zephyr #49611: ehl_crb: Failed to run timer testcases

Kunli123, 2022-08-29:
**Describe the bug**
The testcase tests/kernel/timer/timer_behavior failed to run on the EHL CRB.

**To Reproduce**
west build -b ehl_crb tests/kernel/timer/timer_behavior/ -p always 
west flash /home/ztest/EHL_X86_PXE.sh

**Logs and console output**
```
Press [Enter]  to directly boot.
Press [F2]     to enter setup and select boot options.
Press [F7]     to show boot menu options.
Press [F12]    to PXE boot.
Press [Pause]  to pause and <any key> to continue booting.
**  WARNING: Test Key is used.  **
>>Start PXE over IPv4.
Station IP address is 192.165.1.93
Server IP address is 192.165.1.1
NBP filename is ehl_1/zephyr.efi
NBP filesize is 2206529 Bytes
Downloading NBP file...
NBP file downloaded successfully.
*** Zephyr EFI Loader ***
RSDP found at 0x76c4e014
Zeroing 4931584 bytes of memory at 0x10d000
Copying 32768 data bytes to 0x1000 from image offset 
Copying 53248 data bytes to 0x100000 from image offset 32768
Copying 2113536 data bytes to 0x5c1000 from image offset 86016
Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
*** Booting Zephyr OS build zephyr-v3.1.0-4286-gccfe9513ecee  ***
Running TESTSUITE timer_behavior
===================================================================
START - test_periodic_behavior
periodic timer behavior test
periodic timer samples gathered, calculating statistics
timer clock rate 19200000, kernel tick rate 10000, expected period 1000 us, samples 10000, period 19200.000000 cycles
statistics samples 10000, rollovers 0, mean 1000.003333 us, variance 4556.551368 us, stddev 67.502228 us, min 891.458333 us, max 1131.614583 us
statistics samples 10000, rollovers 0, mean 19200.064000 cycles, variance 1679727.096304 cycles, stddev 1296.042847 cycles, min 17116 cycles, max 21727 cycles
timer start cycle 321316359, end cycle 513316999, total time 10000033.333333, expected time 10000000.000000, difference 33.333333
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/main.c:164: timer_behavior_test_periodic_behavior: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than 10%)
FAIL - test_periodic_behavior in 10.086 seconds
===================================================================
TESTSUITE timer_behavior failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [timer_behavior]: pass = 0, fail = 1, skip = 0, total = 1 duration = 10.086 seconds
- FAIL - [timer_behavior.test_periodic_behavior] duration = 10.086 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: ddf58a16fc16a41ecf74d325dd9dd8bb
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.0
 - Commit: ccfe9513ecee

yinfangw, 2022-09-01:
When I use APIC_TSC_DEADLINE_TIMER, the testing can pass.

twister -W -p ehl_crb --device-testing --west-flash=$HOME/RPL_X86_USB_v2.sh --device-serial /dev/ttyUSB0 -v -T tests/kernel/timer/timer_behavior/  -vv -x=CONFIG_APIC_TSC_DEADLINE_TIMER=y -x=CONFIG_HPET_TIMER=n
The tests log is as follow:
DEBUG   - DEVICE: *** Zephyr EFI Loader ***
DEBUG   - DEVICE: RSDP found at 0x76c4e014
DEBUG   - DEVICE: Zeroing 4931584 bytes of memory at 0x10d000
DEBUG   - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
DEBUG   - DEVICE: Copying 53248 data bytes to 0x100000 from image offset 32768
DEBUG   - DEVICE: Copying 2113536 data bytes to 0x5c1000 from image offset 86016
DEBUG   - DEVICE: Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-4500-g202b2ff91fbc  ***
DEBUG   - DEVICE: Running TESTSUITE timer_behavior
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_periodic_behavior
DEBUG   - DEVICE: periodic timer behavior test
DEBUG   - DEVICE: periodic timer samples gathered, calculating statistics
DEBUG   - DEVICE: timer clock rate 1900000000, kernel tick rate 10000, expected period 1000 us, samples 10000, period 1900000.000000 cycles
DEBUG   - DEVICE: statistics samples 10000, rollovers 0, mean 999.999398 us, variance 0.002399 us, stddev 0.048980 us, min 995.648947 us, max 1000.082105 us
DEBUG   - DEVICE: statistics samples 10000, rollovers 0, mean 1899998.855300 cycles, variance 8660.429362 cycles, stddev 93.061432 cycles, min 1891733 cycles, max 1900156 cycles
DEBUG   - DEVICE: timer start cycle 31860128001, end cycle 50860116554, total time 9999993.975263, expected time 10000000.000000, difference -6.024737
DEBUG   - DEVICE: PASS - test_periodic_behavior in 1.022 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: TESTSUITE timer_behavior succeeded
DEBUG   - DEVICE:
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY START ------
DEBUG   - DEVICE:
DEBUG   - DEVICE: SUITE PASS - 100.00% [timer_behavior]: pass = 1, fail = 0, skip = 0, total = 1 duration = 1.022 seconds
DEBUG   - DEVICE: - PASS - [timer_behavior.test_periodic_behavior] duration = 1.022 seconds
DEBUG   - DEVICE:
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
DEBUG   - DEVICE:
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: RunID: 6d19a94dbd8755edbf8fffe9caf7fc24
DEBUG   - DEVICE: PROJECT EXECUTION SUCCESSFUL
DEBUG   - run status: ehl_crb/tests/kernel/timer/timer_behavior/kernel.timer.timer passed
INFO    - 1/1 ehl_crb                   tests/kernel/timer/timer_behavior/kernel.timer.timer PASSED (device 33.147s)

fabiobaltieri, 2022-09-21:
@Kunli123 does the comment above help? Lowering priority seems it appears that a config option could fix this, feel free to send a PR to change the default for the platform if that's appropriate for this case.

Kunli123, 2022-09-22:
@fabiobaltieri OK ,I will.

enjiamai, 2022-09-23:
Hi @fabiobaltieri , I am still working on finding the root cause...

enjiamai, 2022-09-30:
We find this is related to "HPET timer on SMP". We observed that in timer_behavior test, the expect time (10 ticks) is about to have  one tick of error. This won't happens when not in SMP mode or using the TSC deadline timer. See following  log:
```
Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
===== hz(19200000) cyc_per_tick (1920) HPET_COUNTER_CLK_PERIOD(1000000000000000) cp(0x52083333)
*** Booting Zephyr OS build v3.2.0-rc3-98-g1c953834d671  ***
Running TESTSUITE timer_behavior
===================================================================
START - test_periodic_behavior
periodic timer behavior test
periodic timer samples gathered, calculating statistics
sample 1 spent cycles (21126) error [1926] cpu{0}
sample 131 spent cycles (17245) error [-1955] cpu{0}
sample 132 spent cycles (21131) error [1931] cpu{0}
sample 145 spent cycles (17250) error [-1950] cpu{0}
sample 146 spent cycles (21123) error [1923] cpu{0}
sample 195 spent cycles (17251) error [-1949] cpu{0}
sample 520 spent cycles (17249) error [-1951] cpu{0}
sample 712 spent cycles (17246) error [-1954] cpu{0}
sample 713 spent cycles (21122) error [1922] cpu{0}
sample 842 spent cycles (17097) error [-2103] cpu{0}
sample 843 spent cycles (21142) error [1942] cpu{0}
sample 857 spent cycles (17278) error [-1922] cpu{0}
sample 950 spent cycles (17243) error [-1957] cpu{0}
sample 951 spent cycles (21123) error [1923] cpu{0}
sample 1114 spent cycles (17247) error [-1953] cpu{0}
sample 1115 spent cycles (21120) error [1920] cpu{0}
sample 1231 spent cycles (17247) error [-1953] cpu{0}
sample 1256 spent cycles (17248) error [-1952] cpu{0}
sample 1352 spent cycles (17245) error [-1955] cpu{0}
sample 1353 spent cycles (21122) error [1922] cpu{0}
sample 1710 spent cycles (21120) error [1920] cpu{0}
sample 2106 spent cycles (17244) error [-1956] cpu{0}
sample 2107 spent cycles (21126) error [1926] cpu{0}
sample 2541 spent cycles (17118) error [-2082] cpu{0}
sample 2593 spent cycles (17248) error [-1952] cpu{0}
sample 2705 spent cycles (17244) error [-1956] cpu{0}
sample 2817 spent cycles (17245) error [-1955] cpu{0}
sample 2967 spent cycles (17247) error [-1953] cpu{0}
....
sample 8377 spent cycles (21122) error [1922] cpu{0}
sample 8931 spent cycles (17226) error [-1974] cpu{0}
sample 8984 spent cycles (17245) error [-1955] cpu{0}
sample 8985 spent cycles (21124) error [1924] cpu{0}
sample 9016 spent cycles (17247) error [-1953] cpu{0}
sample 9290 spent cycles (17249) error [-1951] cpu{0}
sample 9291 spent cycles (21124) error [1924] cpu{0}
sample 9362 spent cycles (17251) error [-1949] cpu{0}
sample 9367 spent cycles (17222) error [-1978] cpu{0}
sample 9368 spent cycles (21121) error [1921] cpu{0}
sample 9384 spent cycles (17244) error [-1956] cpu{0}
sample 9385 spent cycles (21121) error [1921] cpu{0}
sample 9450 spent cycles (17250) error [-1950] cpu{0}
sample 9451 spent cycles (21120) error [1920] cpu{0}
sample 9489 spent cycles (17245) error [-1955] cpu{0}
sample 9490 spent cycles (21127) error [1927] cpu{0}
sample 9505 spent cycles (17244) error [-1956] cpu{0}
sample 9506 spent cycles (21121) error [1921] cpu{0}
sample 9640 spent cycles (17246) error [-1954] cpu{0}
sample 9641 spent cycles (21122) error [1922] cpu{0}
sample 9788 spent cycles (17121) error [-2079] cpu{0}
sample 9798 spent cycles (17226) error [-1974] cpu{0}
===== cyc_per_tick(1920), failed counts(109)timer clock rate 19200000, kernel tick rate 10000, expected period 1000 us, samples 10000, period 19200.000000 cycles, expected period drift 0.000000 us
statistics samples 10000, rollovers 0, mean 1000.013292 us, variance 144.386149 us, stddev 12.016079 us, min 890.468750 us, max 1101.822917 us
statistics samples 10000, rollovers 0, mean 19200.255200 cycles, variance 53226.510073 cycles, stddev 230.708710 cycles, min 17097 cycles, max 21155 cycles
timer start cycle 314862853, end cycle 506865405, total time 10000132.916667, expected time 10000000.000000, expected time drift 0.000000, difference 132.916667
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/main.c:206: timer_behavior_test_periodic_behavior: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
FAIL - test_periodic_behavior in 10.676 seconds
===================================================================
TESTSUITE timer_behavior failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [timer_behavior]: pass = 0, fail = 1, skip = 0, total = 1 duration = 10.676 seconds
- FAIL - [timer_behavior.test_periodic_behavior] duration = 10.676 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 566d0124bfa5e17309ddc7ddb4e6b66a
PROJECT EXECUTION FAILED
```

By the way, a tick align before start testing is also needed to reduce the error.

PR #51099 (2022-10-12): drivers: timer: use sys_read64 to read HPET counter on 64 bits cpu
drivers: timer: use sys_read64 to read HPET counter on 64 bits cpu

For 32 bit processor to read the 64 bits hpet counter, the HPET spec
2.4.7 suggest to read HPET counter high and low then checking the
high bits to decide if it rollovers or not.

But this logic seems to cause problem for 64 bits processor under SMP,
there is a possible one tick earlier under tickless mode. It is likely
to be the cache coherence issue, because a mfence instruction before
reading the timer works.

So we change to read the 64 bits counter by sys_ [... cut]
commit dcda15d1 (2022-10-12): drivers: timer: use sys_read64 to read HPET counter on 64 bits cpu

For 32 bit processor to read the 64 bits hpet counter, the HPET spec
2.4.7 suggest to read HPET counter high and low then checking the
high bits to decide if it rollovers or not.

But this logic seems to cause problem for 64 bits processor under SMP,
there is a possible one tick earier under tickless mode. It is likely
to be the cache coherence issue, because a mfence instruction before
reading the timer works.

So we change to read the 64 bits counter by sys_read64 on 64bit
processor to prevent this issue.

Fixes #49611