zephyrproject-rtos/zephyr #49614: acrn_ehl_crb: The testcase tests/kernel/sched/schedule_api failed to run.

Kunli123, 2022-08-29:
**Describe the bug**
The testcase tests/kernel/sched/schedule_api failed to run on the acrn_ehl_crb.

**To Reproduce**
twister -p acrn_ehl_crb --device-testing --device-serial-pty=$HOME/acrn-env/acrn-test-pty.exp --west-flash=$HOME/acrn-env/acrn_efi.sh -v -T tests/kernel/sched/schedule_api/ -vv

**Logs and console output**
```
ACRN:\>
ACRN:\>vm_console 0
----- Entering VM 0 Shell -----
*** Booting Zephyr OS build zephyr-v3.1.0-4286-gccfe9513ecee  ***
Running TESTSUITE threads_scheduling
===================================================================
START - test_bad_priorities
PASS - test_bad_priorities in 0.001 seconds
...
===================================================================
START - test_sleep_cooperative
PASS - test_sleep_cooperative in 0.561 seconds
===================================================================
START - test_sleep_wakeup_preemptible
PASS - test_sleep_wakeup_preemptible in 0.001 seconds
===================================================================
START - test_slice_perthread
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:159: slice_expired: ((dt - PERTHREAD_SLICE_TICKS) <= TICK_SLOP is false)
slice expired >4 ticks late (dt=357)
FAIL - test_slice_perthread in 0.044 seconds
===================================================================
START - test_slice_reset
PASS - test_slice_reset in 1.801 seconds
===================================================================
```

**Environment (please complete the following information):**

OS: Linux
Toolchain: Zephyr SDK 0.15.0
Commit: https://github.com/zephyrproject-rtos/zephyr/commit/ccfe9513ecee6dda0525735c96cf0efa1f30d824

Zhaoningx, 2022-09-01:
This is due to the commit 672d0962c004a5a790d91bad9f4b84a85f43855b
```
672d0962c004a5a790d91bad9f4b84a85f43855b is the first bad commit
commit 672d0962c004a5a790d91bad9f4b84a85f43855b
Author: Andy Ross <andrew.j.ross@intel.com>
Date:   Wed Mar 2 11:36:14 2022 -0800

    tests/kernel/schedule_api: Add TIMESLICE_PER_THREAD case
    
    Simple coverage exerciser for the per-thread timeslice feature.  Added
    as a(nother) new variant of the schedule_api test.
    
    Signed-off-by: Andy Ross <andrew.j.ross@intel.com>

 tests/kernel/sched/schedule_api/src/main.c         |  3 +-
 tests/kernel/sched/schedule_api/src/test_sched.h   |  1 +
 .../sched/schedule_api/src/test_slice_scheduling.c | 71 +++++++++++++++++++++-
 tests/kernel/sched/schedule_api/testcase.yaml      |  6 ++
 4 files changed, 79 insertions(+), 2 deletions(-)
```

fabiobaltieri, 2022-09-21:
@andyross does this one ring any bell? (see the comment above)

mengxianglinx, 2022-09-22:
acrn_ehl_crb use apic_tsc timer, for acrn, I suspect that CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000 is too high, I set it to 1000, 800, 500, the failed test cases passed.

enjiamai, 2022-09-27:
After bisect, the 
- tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing
- tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing
started failed from ec8a3ba7a8ca9de53d9171aaef5dcd856d0b3b66.

And this is a newly added test cases, the failure is order dependent.
- tests/kernel/sched/schedule_api/kernel.scheduler.slice_perthread

enjiamai, 2022-10-10:
Got the problem. The test_slice_perthread() test should be skipped under CONFIG_TIMESLICING is not set.

enjiamai, 2022-10-13:
The debugging results：
(1) For test_slice_perthread() falied:
After debugging, I found issue happens because at certain point of timing, the hypervisor will have a long delay.
(At about uptime at 106 ms, 632 ms, we will get many cycles late when we call k_cycle_get_32/64().
These points affectes some timing senstive testcases (such as tests/timer/timer_behavior, tests/kernel/smp etc) or even some scheduling tests those measure the time.
We are still not totally clear why the hypervisor will have such a delay at these point. It is most likely not an issue of Zephyr, because the clock cycle read from rdtsc provided by hypervisor.
A suggested workarond solution is to add a 700ms boot delay to avoid these point. We can add CONFIG_BOOT_DELAY=700 for it.

(2) For test_priority_preemptible_wait_prio() failed:
There are threads t0, t1, t2, t3, t4 in this test, and the expecting order should be 0,1,2,3. The t1 has higher priority and waiting for long time, and t1 should be executed prior to t2. In the falied test we got an order 0,2,1,3, because the t2 already completed but t1 has not start yet, due to the thread spend less time to start. (it runs at 1.9G), in this case, the delay interval of each thread starting should be scale.

enjiamai, 2022-10-13:
For issue2, we try to add debugging msg for it:
```
diff --git a/tests/kernel/sched/schedule_api/src/test_sched_priority.c b/tests/kernel/sched/schedule_api/src/test_sched_priority.c
index 2f2de9be2c..12d0873b34 100644
--- a/tests/kernel/sched/schedule_api/src/test_sched_priority.c
+++ b/tests/kernel/sched/schedule_api/src/test_sched_priority.c
@@ -15,6 +15,11 @@ static int last_prio;
 static uint8_t tid_num[4];
 static struct k_sem sync_sema;

+uint64_t thread_start_cyc[THREAD_NUM];
+uint64_t cyc_start[4];
+uint64_t cyc_enter[4];
+uint64_t cyc_end[4];
+

 static void thread_entry(void *p1, void *p2, void *p3)
 {
@@ -25,8 +30,9 @@ static void thread_entry_prio(void *p1, void *p2, void *p3)
 {
        static int i;

+       cyc_enter[POINTER_TO_INT(p1)] = k_cycle_get_64();
        k_sem_take(&sync_sema, K_MSEC(100));
-
+       cyc_end[POINTER_TO_INT(p1)] = thread_start_cyc[POINTER_TO_INT(p1)] = k_cycle_get_64();
        tid_num[i++] = POINTER_TO_INT(p1);
 }

@@ -139,18 +145,22 @@ ZTEST(threads_scheduling_1cpu, test_priority_preemptible_wait_prio)
        tid[0] = k_thread_create(&tdata_prio[0], tstacks[0], STACK_SIZE,
                        thread_entry_prio, INT_TO_POINTER(0), NULL, NULL,
                        K_PRIO_PREEMPT(0), 0, K_MSEC(10));
+       cyc_start[0] = k_cycle_get_64();
        /* the highest-priority thread that has waited the shorter */
        tid[1] = k_thread_create(&tdata_prio[1], tstacks[1], STACK_SIZE,
                        thread_entry_prio, INT_TO_POINTER(1), NULL, NULL,
                        K_PRIO_PREEMPT(0), 0, K_MSEC(20));
+       cyc_start[1] = k_cycle_get_64();
        /* the lowest-priority thread that has waited longest */
        tid[2] = k_thread_create(&tdata_prio[2], tstacks[2], STACK_SIZE,
                        thread_entry_prio, INT_TO_POINTER(2), NULL, NULL,
                        K_PRIO_PREEMPT(1), 0, K_MSEC(10));
+       cyc_start[2] = k_cycle_get_64();
        /* the lowest-priority thread that has waited shorter */
        tid[3] = k_thread_create(&tdata_prio[3], tstacks[3], STACK_SIZE,
                        thread_entry_prio, INT_TO_POINTER(3), NULL, NULL,
                        K_PRIO_PREEMPT(1), 0, K_MSEC(20));
+       cyc_start[3] = k_cycle_get_64();

        /* relinquish CPU for above threads to start */
        k_sleep(K_MSEC(30));
@@ -159,6 +169,15 @@ ZTEST(threads_scheduling_1cpu, test_priority_preemptible_wait_prio)
                k_sem_give(&sync_sema);
        }

+       for (int i = 0; i < THREAD_NUM; i++) {
+               printk("tid_num:%d tid_chk: %d start cyc 0x%llx\n",
+                       (uint8_t)tid_num[i], (uint8_t)tid_chk[i], thread_start_cyc[i]);
+       }
+       for (int i = 0; i < THREAD_NUM; i++) {
+               printk("== t%d cyc start: %lld enter: %lld end: %lld diff:[%lld]\n",
+                       i, cyc_start[i], cyc_enter[i], cyc_end[i], cyc_end[i] - cyc_enter[i]);
+       }
+
        zassert_true((memcmp(tid_num, tid_chk, 4) == 0),
                     "scheduling priority failed");

```
And we can get following info:

test failed log:
DEBUG   - DEVICE: == t0 cyc start: 53118940124 enter: 53138117934 end: 53337767446 diff:[199649512]
DEBUG   - DEVICE: == t1 cyc start: 53118946455 enter: **53157117596** end: **53337846801** diff:[180729205]
DEBUG   - DEVICE: == t2 cyc start: 53118952668 enter: **53138121384** end: **53337804749** diff:[199683365]
DEBUG   - DEVICE: == t3 cyc start: 53118958948 enter: 53157120936 end: 53337887749 diff:[180766813]

test pass log
DEBUG   - DEVICE: == t0 cyc start: 53396653173 enter: 53406330759 end: 53615484319 diff:[209153560]
DEBUG   - DEVICE: == t1 cyc start: 53396659455 enter: 53415830424 end: 53615521665 diff:[199691241]
DEBUG   - DEVICE: == t2 cyc start: 53396665648 enter: 53406334148 end: 53615558915 diff:[209224767]
DEBUG   - DEVICE: == t3 cyc start: 53396671906 enter: 53415833815 end: 53615596135 diff:[199762320]

enjiamai, 2022-12-02:
I think this issue is likely not the problem of zephyr. Lower its priority.

closed 2023-02-16 by github-actions[bot]