zephyrproject-rtos/zephyr #49616: acrn_ehl_crb: The testcase tests/kernel/common failed to run.

Kunli123, 2022-08-29:
**Describe the bug**
The testcase tests/kernel/common failed to run on the acrn_ehl_crb.

**To Reproduce**
twister -p acrn_ehl_crb --device-testing --device-serial-pty=$HOME/acrn-env/acrn-test-pty.exp --west-flash=$HOME/acrn-env/acrn_efi.sh -v -T tests/kernel/common/ -vv

**Logs and console output**

```

Running TESTSUITE irq_offload
===================================================================
START - test_irq_offload
PASS - test_irq_offload in 0.001 seconds
===================================================================
START - test_nop
time k_cycle_get_32() takes 41 cycles
arch_nop() takes 0 cycles
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/common/src/irq_offload.c:195: nop: (diff > 0 is false)
arch_nop() takes 0 cpu cycles
FAIL - test_nop in 0.001 seconds
===================================================================
TESTSUITE irq_offload failed.
```
**Environment (please complete the following information):**

OS: Linux
Toolchain: Zephyr SDK 0.15.0
Commit: https://github.com/zephyrproject-rtos/zephyr/commit/ccfe9513ecee6dda0525735c96cf0efa1f30d824

Zhaoningx, 2022-08-30:
This issue also failed to run in **v2.7-branch** on the **acrn_ehl_crb** board.

enjiamai, 2022-08-31:
There is a design problem in this test_nop testcase. It sometimes files a false alarm. Hi, @mmahadevan108 , I would like to set it's to priority low, thanks!

PR #49949 (2022-09-21): tests: kernel: common: remove the nop test case
The test case was originally designed for the coverage of
nop()/arch_nop() function. It is not very meaningful for
testing something but causing lots of false alarms on the
kernel/common test so far. Suggest removing this test case.

Fix #49616
closed 2022-09-21 by mmahadevan108