zephyrproject-rtos/zephyr #49621: STM32WB55 BLE Extended Advertising support

piotrParf, 2022-08-29:
**Is your feature request related to a problem? Please describe.**

The new HAL(v1.1.14) for stm32wb is released and it should support BLE extended advertising in Zephyr while using firmware stm32wb5x_BLE_HCILayer_extended_fw. Now in hal_stm32 the extended advertising is turned off. It can be turned on by modifying CFG_BLE_OPTIONS in app_conf.h. Unfortunately, the ipm_stm32wb.c in zephyr is still unmodified and does not support new options like:
	CFG_BLE_MAX_COC_INITIATOR_NBR,
     	CFG_BLE_MIN_TX_POWER,
    	CFG_BLE_MAX_TX_POWER,
    	CFG_BLE_RX_MODEL_CONFIG,
     	CFG_BLE_MAX_ADV_SET_NBR,
     	CFG_BLE_MAX_ADV_DATA_LEN,
     	CFG_BLE_TX_PATH_COMPENS,
     	CFG_BLE_RX_PATH_COMPENS

I modified the mentioned ipm_stm32wb.c and extended advertising work. Extended advertising seems to have an issue. In hal_stm32 the default value of advertising sets is set to a maximum of 3. The problem happens when I try to start more than one advertising set - I receive an error that zephyr was unable to start the second advertising set.

**Describe the solution you'd like**
Modify Zephyr sources to be able to turn on extended advertising sets and/or help with working simultaneously with multiple advertising sets.

**Describe alternatives you've considered**
---

**Additional context**
I prepared a small demo of the issue with starting multiple advertising sets here with needed patches to Zephyr and hal_stm32 to enable extended advertising with stm32wb5x_BLE_HCILayer_extended_fw radio firmware: https://github.com/piotrParf/stm32_multiadvertising_demo

piotrParf, 2022-08-30:
ping @erwango

erwango, 2022-08-30:
@piotrParf In order to use `stm32wb5x_BLE_Stack_full_extended_fw.bin`, release notes mentions:
```
Warning: To use this binary, it is necessary to adapt the scatter file in the BLE applications as:
The RAM_A shared range shall be reduced to memory range [0x20030000:0x200307FF]
The Mail-box buffers(MB_MEM1, MB_MEM2) shall be located in RAM_B shared defined in memory range [0x20038000:0x2003A7FF]
The RAM_B shared shall be added to Total_RAM_region
```
I don't see traces of this in the code you shared. Did you do something around this ?

piotrParf, 2022-08-30:
The same applies to stm32wb5x_BLE_HCILayer_extended_fw ?

erwango, 2022-08-30:
Ok, no it doesn't. My bad. Naming confusion.

lkoudelka, 2022-09-12:
Hello,
I was trying on my side with code you linked above. Problem preventing two advertising sets on my side was adv_param used in bt_le_ext_adv_create. Define BT_LE_ADV_OPT_EXT_ADV was missing there.

With following setting I can see two advertising
`static struct bt_le_adv_param adv_param1=BT_LE_ADV_PARAM_INIT(
		BT_LE_ADV_OPT_EXT_ADV | BT_LE_ADV_OPT_USE_IDENTITY, BT_GAP_ADV_FAST_INT_MIN_2, BT_GAP_ADV_FAST_INT_MAX_2, NULL);
static struct bt_le_adv_param adv_param2=BT_LE_ADV_PARAM_INIT(
		BT_LE_ADV_OPT_EXT_ADV | BT_LE_ADV_OPT_USE_IDENTITY, BT_GAP_ADV_FAST_INT_MIN_2, BT_GAP_ADV_FAST_INT_MAX_2, NULL);`

lkoudelka, 2022-09-13:
On STM32 in general is possible to have multiple legacy advertising (if just 1 advertising is connectable). Could you please share some airtrace/config detail to give us better understanding what exactly you need?

piotrParf, 2022-09-13:
Ok, you mean "Additional" advertising support via STM32WB specific ACI commands? I can't make multiple connectable advertising sets then? Is that possible with extended advertising operates then? I mean can I have two(or more) connectable extended advertising sets but not two connectable legacy advertising sets?

lkoudelka, 2022-09-14:
Unfortunately I need to agree with you - Additional advertising is not available with zephyr. Extended advertising is targeting your  question, just there is the question of compatibility as extended advertising is still not so common among BLE devices/phones on market

piotrParf, 2022-09-20:
Ok. We should manage utilizing callbacks with this. But for future we would like to ask ST team to implement "legacy multiadvertising" and also add support in Zephyr for extended advertising with STM32WB55 platform. As it works as is ext advertising should be switchable with CONFIG_BT_EXT_ADV without manipulating app_conf.h in hal_stm32 and patching ipm_stm32wb.c file. The same with CFG_BLE_MAX_ADV_SET_NBR and Zephye BT stack CONFIG_BT_EXT_ADV_MAX_ADV_SET.

erwango, 2022-10-18:
@piotrParf 

>  But for future we would like to ask ST team to implement "legacy multiadvertising" and also add support in Zephyr for extended advertising with STM32WB55 platform.

Please open a ticket directly through ST online support tool for this specific request

> add support in Zephyr for extended advertising with STM32WB55 platform.

I'll open a dedicated zephyr issue for this specific point.

And then we can close current point. Is that fine from  your point of view ?

piotrParf, 2022-10-19:
For me it is OK to open a separate zephyr issue to add extended advertising with stm32wb55. I am opening a feature request with ST platform for "multiadvertising by extended advertising api".

closed 2022-10-19 by erwango