zephyrproject-rtos/zephyr #49624: Bluetooth: Controller: Recent RAM usage increase for hci_rpmsg build

Thalley, 2022-08-29:
**Is your enhancement proposal related to a problem? Please describe.**
Recently the file `ull_llcp.c` file was added, which increased the RAM usage of the `hci_rpmsg` sample by roughly 4500 bytes. The result of this is that `CONFIG_BT_CTLR_SYNC_ISO` and `CONFIG_BT_CTLR_ADV_ISO` can no longer fit in RAM for the sample anymore. 

**Describe the solution you'd like**
Since the `hci_rpmsg` is a sample that makes it possible to use the nRF5340 across multiple cores (app and net core), it is imperative that it is possible to build a general-use controller build with that sample. It should be possible to use the `hci_rpmsg` with extended advertising and ISO support (both connected and broadcast). 

**Describe alternatives you've considered**
N/A

**Additional context**
I've attached logs of a recent commit (`53f2ccf2a2f587882503e4eece00fa871a9492d5`) and an older (pre `ull_llcp.c`) commit (`c3302d5f41c9f17775f4c69a5baa9e8c72f9cf3d`) that shows the increase in memory. The logs files are from building `hci_rpmsg` with `CONFIG_BT_EXT_ADV=y`, `CONFIG_BT_PER_ADV_SYNC=y` and `CONFIG_BT_CTLR_SYNC_ISO=y`. 

[ram_53f2ccf2a2f587882503e4eece00fa871a9492d5.txt](https://github.com/zephyrproject-rtos/zephyr/files/9444253/ram_53f2ccf2a2f587882503e4eece00fa871a9492d5.txt)
[ram_c3302d5f41c9f17775f4c69a5baa9e8c72f9cf3d.txt](https://github.com/zephyrproject-rtos/zephyr/files/9444254/ram_c3302d5f41c9f17775f4c69a5baa9e8c72f9cf3d.txt)

Thalley, 2022-08-29:
ping @cvinayak @kruithofa @wopu-ot

Thalley, 2022-08-29:
I also created https://github.com/zephyrproject-rtos/zephyr/pull/49623 so that we can track the (vanilla) edition of the sample on the nRF5340 netcore.

wopu-ot, 2022-08-29:
We are seeing an increase in memory consumption for our platform as well. ping @erbr-ot

ppryga-nordic, 2022-08-30:
@Thalley does it mean that with enabled ISO (and other required features)  the sample is about edge of free memory for nRF5340 net core?

Thalley, 2022-08-30:
> @Thalley does it mean that with enabled ISO (and other required features) the sample is about edge of free memory for nRF5340 net core?

It actually goes over the edge if I enable both `CONFIG_BT_CTLR_SYNC_ISO` and `CONFIG_BT_CTLR_ADV_ISO` (so that is excluding support for connected ISO).

kruithofa, 2022-12-13:
RAM usage for vanilla hci-rpmsg sample was reduced, but there is still a significant memory usage for the hci-rpmsg sample with   both sync receiver and broadcast source enabled

kruithofa, 2023-01-25:
Increased RAM usage is pretty much a result of added functionality.
Adjusting parameters (like reducing max. nr. of connections) will help.

PR #50480 (2022-10-06): Bluetooth: controller: conditionally compile legacy ctrl tx buffers
Conditionally compile the legacy control procedure defined
control Tx buffers that is not required in the new control
procedure implementation

fixes #50461
fixes #49624
closed 2023-01-25 by kruithofa