zephyrproject-rtos/zephyr #49628: Compilation fails when ASAN is used with gcc

chrta, 2022-08-29:
**Describe the bug**

I try to execute tests using twister with enabled address sanitizer using gcc. This fails since commit https://github.com/zephyrproject-rtos/zephyr/commit/74cc5347587624ded3ad9c2cc5266909e864cad9 (see https://github.com/zephyrproject-rtos/zephyr/pull/49127)

The newly added macro `#define __noasan __attribute__((no_sanitize("address")))` does not seem to work for gcc and produces the error `error: ‘no_sanitize’ attribute ignored [-Werror=attributes]`

The no_sanitize attribute for gcc does not seems to work in the same way as it does for clang.

Changing `#define __noasan __attribute__((no_sanitize("address")))` to `#define __noasan /**/` "fixes" the issue for gcc.

**To Reproduce**
Steps to reproduce the behavior:
1. twister --enable-asan -p native_posix -T tests/kernel/common/
2. See error

**Expected behavior**
All tests compile

**Impact**
annoyance

**Logs and console output**
Compiler output from /work/zephyr/twister-out/native_posix/tests/kernel/common/kernel.common/build.log (Ubuntu 22.04, gcc 11.2.0)
```
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /usr/bin/gcc -DKERNEL -DTC_RUNID=f9bafbf94988d20c494f8ab4edae3c66 -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I/work/zephyr/include -I/work/zephyr/twister-out/native_posix/tests/kernel/common/kernel.common/zephyr/include/generated -I/work/zephyr/soc/posix/inf_clock -I/work/zephyr/boards/posix/native_posix -I/work/zephyr/subsys/testsuite/include -I/work/zephyr/subsys/testsuite/ztest/include -fno-strict-aliasing -Os -imacros /work/zephyr/twister-out/native_posix/tests/kernel/common/kernel.common/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -Werror -fno-asynchronous-unwind-tables -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/work/zephyr/tests/kernel/common=CMAKE_SOURCE_DIR -fmacro-prefix-map=/work/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/work=WEST_TOPDIR -ffunction-sections -fdata-sections -m32 -include /work/zephyr/arch/posix/include/posix_cheats.h -fno-freestanding -fsanitize-recover=all -fsanitize=address -std=c11 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /work/zephyr/tests/kernel/common/src/main.c
In file included from /work/zephyr/include/zephyr/toolchain/gcc.h:88,
                 from /work/zephyr/include/zephyr/toolchain.h:50,
                 from /work/zephyr/include/zephyr/sys/printk.h:11,
                 from /work/zephyr/subsys/testsuite/ztest/include/zephyr/ztest.h:44,
                 from /work/zephyr/tests/kernel/common/src/main.c:8:
/work/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test_new.h:339:40: error: ‘no_sanitize’ attribute ignored [-Werror=attributes]
  339 |         static STRUCT_SECTION_ITERABLE(ztest_unit_test, z_ztest_unit_test_##suite##_##fn) = {      \
      |                                        ^~~~~~~~~~~~~~~
/work/zephyr/include/zephyr/toolchain/common.h:188:55: note: in definition of macro ‘Z_DECL_ALIGN’
  188 | #define Z_DECL_ALIGN(type) __aligned(__alignof(type)) type
      |                                                       ^~~~
/work/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test_new.h:339:16: note: in expansion of macro ‘STRUCT_SECTION_ITERABLE’
  339 |         static STRUCT_SECTION_ITERABLE(ztest_unit_test, z_ztest_unit_test_##suite##_##fn) = {      \
      |                ^~~~~~~~~~~~~~~~~~~~~~~
/work/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test_new.h:354:39: note: in expansion of macro ‘Z_TEST’
  354 | #define Z_ZTEST(suite, fn, t_options) Z_TEST(suite, fn, t_options, 0)
      |                                       ^~~~~~
/work/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test_new.h:387:26: note: in expansion of macro ‘Z_ZTEST’
  387 | #define ZTEST(suite, fn) Z_ZTEST(suite, fn, 0)
      |                          ^~~~~~~
/work/zephyr/tests/kernel/common/src/main.c:42:1: note: in expansion of macro ‘ZTEST’
   42 | ZTEST(common, test_version)
      | ^~~~~

```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 (gcc 9.4.0) and Ubuntu 22.04 (gcc 11.2.0)
 - default gcc versions from Ubuntu, gcc 9.4.0 and 11.2.0
 - current zephyr main branch, commit 52fe06495dc2d9bfa394353778c3f211441a65fb

andyross, 2022-08-31:
Hrm.  This seems to be working for me.   It's true, though, that gcc (I'm on 12.2.0) doesn't actually need this.  With gcc, it's enough to flag a bit of data "const" to prevent it from instrumenting it.  So it's surely OK to limit that attribute to clang only, something like this seems to work on my end, can you try it?

```diff
diff --git a/include/zephyr/toolchain/gcc.h b/include/zephyr/toolchain/gcc.h
index f3cef4d12a..b783928e9d 100644
--- a/include/zephyr/toolchain/gcc.h
+++ b/include/zephyr/toolchain/gcc.h
@@ -618,7 +618,7 @@ do {                                                                    \
  */
 #define Z_IS_POW2(x) (((x) != 0) && (((x) & ((x)-1)) == 0))
 
-#ifdef CONFIG_ASAN
+#if defined(CONFIG_ASAN) && defined(__clang__)
 #define __noasan __attribute__((no_sanitize("address")))
 #else
 #define __noasan /**/
```

Please ping me if I forget this.

chrta, 2022-09-01:
@andyross Thanks for the fast reply. Your suggested diff works for me. If `__noasan` is never set for gcc and only for clang, maybe a better option would be to move the complete `__noasan`-block from `include/zephyr/toolchain/gcc.h` to `include/zephyr/toolchain/llvm.h`.

closed 2022-09-21 by andyross