zephyrproject-rtos/zephyr #49629: ISO Broadcast sample fails to send data on nRF5340

Frodevan, 2022-08-29:
When the ISO broadcast/receive samples are run with the Packetcraft controller on the nRF5340, the samples are able to establish BIG sync but does not attempt to actually broadcast/receive any ISO-data.

Boards:
- arm/nrf5340dk_nrf5340
- arm/nrf5340dk_nrf5340

Samples:
- bluetooth/iso_broadcast
- bluetooth/iso_receive

Controller:
- https://github.com/nrfconnect/sdk-nrf/blob/main/applications/nrf5340_audio/bin/ble5-ctr-rpmsg_3301.hex

**To Reproduce**

Append to prj.conf of both samples:
```
CONFIG_BT_HCI_VS_EXT=n
CONFIG_BT_BUF_ACL_RX_SIZE=256
CONFIG_BT_BUF_ACL_TX_SIZE=256
CONFIG_BT_BUF_CMD_TX_SIZE=255
CONFIG_BT_BUF_EVT_DISCARDABLE_SIZE=255
CONFIG_UART_0_NRF_TX_BUFFER_SIZE=1024
CONFIG_BT_ISO_TX_BUF_COUNT=255
CONFIG_BT_ISO_TX_MTU=496
CONFIG_BT_ISO_RX_MTU=496
```

1. west init zephyr
2. west update
3. west build -p auto -b nrf5340dk_nrf5340_cpuapp -d ./iso_broadcast_build -s ./zephyr/samples/bluetooth/iso_broadcast
4. west build -p auto -b nrf5340dk_nrf5340_cpuapp -d ./iso_receive_build -s ./zephyr/samples/bluetooth/iso_receive
5. Flash controller to NET-core and app.hex of sample to APP-core, broadcast on one devkit, receive on other.
6. Reset both devkits and observe debug logs from UART

**Expected behaviour**
Both samples establishing BIG sync, and broadcaster starts sending ISO packets to the received every 10ms. Every 60 seconds, the broadcast sample is expected to end the BIG sync, and re-establish it before continuing sending more data.

**Observed behaviour**
Both samples establishing sync, but broadcast sample fails with error -122 when trying to send the first packet with ISO data.

**Impact**
The sample is broken, and does not exercise the broadcast feature it is expected to demonstrate.

**Logs and console output**
Broadcast sample
```
*** Booting Zephyr OS build zephyr-v3.1.0-4326-g8195ea7ea0bf  ***
Starting ISO Broadcast Demo
Waiting for BIG complete...ISO Channel 0x20000120 connected
done.
Unable to broadcast data: -122[00:00:00.455,505] <inf> bt_hci_core: Identity: EA:28:08:2D:F7:24 (random)
[00:00:00.455,535] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0ce5, manufacturer 0x07e8
[00:00:00.455,566] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0x0ce5
```
Receive sample
```
*** Booting Zephyr OS build zephyr-v3.1.0-4326-g8195ea7ea0bf  ***
Starting Synchronized Receiver Demo
Get reference to LED device...done.
Configure GPIO pin...done.
Scan callbacks register...success.
Periodic Advertising callbacks register...Success.
Start scanning...success.
Start blinking LED...
Waiting for periodic advertising...
[DEVICE]: 3E:C2:53:A6:D6:7A (random), AD evt type 5, Tx Pwr: 127, RSSI -23 Test ISO Broadcaster C:0 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 2M, Interval: 0x0000 (0 ms), SID: 0
[DEVICE]: 3E:C2:53:A6:D6:7A (random), AD evt type 5, Tx Pwr: 127, RSSI -22 Test ISO Broadcaster C:0 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 2M, Interval: 0x03c0 (1200 ms), SID: 0
Found periodic advertising.
Stop scanning...success.
Creating Periodic Advertising Sync...success.
Waiting for periodic sync...
[DEVICE]: 3E:C2:53:A6:D6:7A (random), AD evt type 5, Tx Pwr: 127, RSSI -23 Test ISO Broadcaster C:0 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 2M, Interval: 0x03c0 (1200 ms), SID: 0
[DEVICE]: 3E:C2:53:A6:D6:7A (random), AD evt type 5, Tx Pwr: 127, RSSI -23 Test ISO Broadcaster C:0 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 2M, Interval: 0x03c0 (1200 ms), SID: 0
[00:00:00.456,207] <inf> bt_hci_core: Identity: D5:87:4E:A5:7D:6B (random)
[00:00:00.456,237] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0ce5, manufacturer 0x07e8
[00:00:00.456,268] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0x0ce5
[DEVICE]: 3E:C2:53:A6:D6:7A (random), AD evt type 5, Tx Pwr: 127, RSSI -23 Test ISO Broadcaster C:0 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 2M, Interval: 0x03c0 (1200 ms), SID: 0
[DEVICE]: 3E:C2:53:A6:D6:7A (random), AD evt type 5, Tx Pwr: 127, RSSI -23 Test ISO Broadcaster C:0 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 2M, Interval: 0x03c0 (1200 ms), SID: 0
PER_ADV_SYNC[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random) synced, Interval 0x03c0 (1200 ms), PHY LE 2M
Periodic sync established.
Waiting for BIG info...
PER_ADV_SYNC[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), tx_power 0, RSSI -23, CTE 0, data length 0, data: 
BIG INFO[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), sid 0x00, num_bis 1, nse 3, interval 0x0008 (10 ms), bn 1, pto 0, irc 3, max_pdu 4, sdu_interval 10000 us, max_sdu 4, phy LE 2M, without framing, not encrypted
Periodic sync established.
Create BIG Sync...
success.
Waiting for BIG sync...
ISO Channel 0x20000120 connected
PER_ADV_SYNC[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), tx_power 0, RSSI -23, CTE 0, data length 0, data: 
BIG INFO[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), sid 0x00, num_bis 1, nse 3, interval 0x0008 (10 ms), bn 1, pto 0, irc 3, max_pdu 4, sdu_interval 10000 us, max_sdu 4, phy LE 2M, without framing, not encrypted
BIG sync established.
Stop blinking LED.
Waiting for BIG sync lost...
PER_ADV_SYNC[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), tx_power 0, RSSI -23, CTE 0, data length 0, data: 
BIG INFO[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), sid 0x00, num_bis 1, nse 3, interval 0x0008 (10 ms), bn 1, pto 0, irc 3, max_pdu 4, sdu_interval 10000 us, max_sdu 4, phy LE 2M, without framing, not encrypted
PER_ADV_SYNC[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), tx_power 0, RSSI -23, CTE 0, data length 0, data: 
BIG INFO[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), sid 0x00, num_bis 1, nse 3, interval 0x0008 (10 ms), bn 1, pto 0, irc 3, max_pdu 4, sdu_interval 10000 us, max_sdu 4, phy LE 2M, without framing, not encrypted
PER_ADV_SYNC[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), tx_power 0, RSSI -23, CTE 0, data length 0, data: 
BIG INFO[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), sid 0x00, num_bis 1, nse 3, interval 0x0008 (10 ms), bn 1, pto 0, irc 3, max_pdu 4, sdu_interval 10000 us, max_sdu 4, phy LE 2M, without framing, not encrypted
PER_ADV_SYNC[0]: [DEVICE]: 3E:C2:53:A6:D6:7A (random), tx_power 0, RSSI -23, CTE 0, data length 0, data: 
...
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: GNU 10.3.0
 - Zephyr version: 3.1.99 (main @ 2022-08-29)

cvinayak, 2022-08-29:
> @cvinayak The ISO broadcast does not yet have data path, right? So we can't send SDUs from the host, but isn't it still sending some dummy data?

The issue is reported on use of Packetcraft Controller.

Preliminary data path is supported in Zephyr Controller. iso_broadcast and iso_receive samples should be working with datapath using Zephyr Controller, this is based on the fact that bsim_test_iso continues to pass in PR CI.

Thalley, 2022-08-29:
Right. This is likely related to the SDU size checks implemented recently.

closed 2022-09-08 by fabiobaltieri