zephyrproject-rtos/zephyr #49631: arch: arm: FP stack warning with GCC 12 and `CONFIG_FPU=y`

microbuilder, 2022-08-29:
**Describe the bug**

When compiling code that uses `__attribute__ ((interrupt ("IRQ")))` (etc.), such as the `ARCH_ISR_DIRECT_DECLARE` macro, GCC 12 will produce the following warning:

```
zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:87:1: warning: FP registers might be clobbered despite 'interrupt' attribute: compile with '-mgeneral-regs-only' [-Wattributes]
   87 | ISR_DIRECT_DECLARE(radio_nrf5_isr)
      | ^~~~~~~~~~~~~~~~~~
```

**To Reproduce**

Using GCC 12.x:

```bash
$ west build -p -b nrf52dk_nrf52832 samples/bluetooth/beacon -D CONFIG_FPU=y
```

**Expected behavior**

No warning should be present.

**Environment (please complete the following information):**

- GCC 12.0

**Additional context**

This warning was introduce as part of this commit:

- https://github.com/gcc-mirror/gcc/commit/3c3b4224875d7b8bfd4126b9dd1d9cb028997285

And is related to this (still open) bug in GCC:

- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=94743

It's unclear to me if this warning is also relevant to Cortex-M, or only applies to Cortex-A/R.

Currently reading through the v6/7/8 reference manuals to properly understand the situation here. Filing this issue to track progress since it may take some digging to determine how to address this properly in GCC.

stephanosio, 2022-08-29:
cc @povergoing @SgrrZhf

stephanosio, 2022-08-29:
> Currently reading through the v7/8 reference manuals to properly understand the situation here.

See the "Exception entry behavior" section in the Architecture Reference Manual.

For the A and R profiles (or the ARM archs prior to the Cortex days), the ISR is responsible for stacking all registers, including the FP registers.

For the M profile, the NVIC takes care of saving all the caller saved registers upon exception entry and the ISR routine is only responsible for saving the callee saved registers.

https://gist.github.com/stephanosio/afef05ee52fa96a12a32646832da7ef5

stephanosio, 2022-08-29:
Re: the commit message in https://github.com/gcc-mirror/gcc/commit/3c3b4224875d7b8bfd4126b9dd1d9cb028997285

> If the handler calls other functions, we'd likely need to save all of
them, for lack of knowledge of which registers they actually clobber.

As per the AAPCS, we would only need to save callee saved registers (i.e. not ALL registers).

> The interrupt attribute does not guarantee that the FP registers are
saved

For the M profile, it does not need to because they (caller saved FP registers) are automatically saved upon exception entry. And the callee saved registers, by definition, need not be saved because the ISR function and any functions it calls, as per the AAPCS, will save what they use.

microbuilder, 2022-08-29:
Reading through the v8-M ARM, I agree this doesn't seem to apply to M targets.

Looking through the GCC code I don't see a single 'Cortex M' flag we could use as an exception to this warning, but still poking around to see the most concise way to only check on A/R in GCC.

stephanosio, 2022-08-29:
> Looking through the GCC code I don't see a single 'Cortex M' flag we could use as an exception to this warning, but still poking around to see the most concise way to only check on A/R in GCC.

might want to try `arm_arch_notm`

SgrrZhf, 2022-08-30:
Is there any issue that needs to be fixed on A/R?

microbuilder, 2022-08-30:
> Is there any issue that needs to be fixed on A/R?

This only applies to Cortex M, so no.

PR #49704 (2022-08-31): arch: arm: Ignore Wattributes warning in direct ISR
GCC 11 and above may report the following warning for the functions
declared with the `interrupt` attribute when compiling FPU-enabled
ARM targets:

  warning: FP registers might be clobbered despite 'interrupt'
  attribute: compile with '-mgeneral-regs-only' [-Wattributes]

This commit disables the above warning because:

* For M-profile architectures (Cortex-M), this warning does not apply
  because the caller saved registers, including the FPU registers, are
  automatically saved upon exception entry and the callee saved
  registers are sa [... cut]
closed 2022-08-31 by stephanosio