zephyrproject-rtos/zephyr #49648: tests/subsys/logging/log_switch_format, log_syst build failures on CAVS

laurenmurphyx64, 2022-08-30:
**Describe the bug**

Failure of switch format and sys-T tests to build:

```
FAILED: zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_output_syst.c.obj 
ccache /home/herbert/zephyr-sdk-0.15.0/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-gcc -DKERNEL -DTC_RUNID=fde3958e4387a906e33624bab6eda5b6 -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/herbert/zephyr-external/zephyr/kernel/include -I/home/herbert/zephyr-external/zephyr/arch/xtensa/include -I/home/herbert/zephyr-external/zephyr/soc/xtensa/intel_adsp/cavs/include -I/home/herbert/zephyr-external/zephyr/drivers -I/home/herbert/zephyr-external/zephyr/include -I/home/herbert/zephyr-external/zephyr/twister-out/intel_adsp_cavs18/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/include/generated -I/home/herbert/zephyr-external/zephyr/soc/xtensa/intel_adsp/cavs -I/home/herbert/zephyr-external/zephyr/soc/xtensa/intel_adsp/cavs/include/intel_cnl_adsp -I/home/herbert/zephyr-external/zephyr/soc/xtensa/intel_adsp/common/include -I/home/herbert/zephyr-external/zephyr/lib/libc/newlib/include -I/home/herbert/zephyr-external/zephyr/subsys/testsuite/include -I/home/herbert/zephyr-external/zephyr/subsys/testsuite/ztest/include -I/home/herbert/zephyr-external/modules/hal/xtensa/include -I/home/herbert/zephyr-external/modules/hal/xtensa/zephyr/soc/intel_cnl_adsp -I/home/herbert/zephyr-external/modules/debug/mipi-sys-t/mipi_syst_library -I/home/herbert/zephyr-external/zephyr/twister-out/intel_adsp_cavs18/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/modules/mipi-sys-t/library/include -I/home/herbert/zephyr-external/zephyr/subsys/logging -I/home/herbert/zephyr-external/modules/debug/mipi-sys-t/library/include -I/home/herbert/zephyr-external/modules/debug/mipi-sys-t/include -I/home/herbert/zephyr-external/zephyr/tests/subsys/logging/log_syst/src -fno-strict-aliasing -Os -imacros /home/herbert/zephyr-external/zephyr/twister-out/intel_adsp_cavs18/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always --sysroot=/home/herbert/zephyr-sdk-0.15.0/xtensa-intel_s1000_zephyr-elf/xtensa-intel_s1000_zephyr-elf -imacros /home/herbert/zephyr-external/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -Werror -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/herbert/zephyr-external/zephyr/tests/subsys/logging/log_syst=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/herbert/zephyr-external/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/herbert/zephyr-external=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_output_syst.c.obj -MF zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_output_syst.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_output_syst.c.obj -c /home/herbert/zephyr-external/zephyr/subsys/logging/log_output_syst.c
In file included from /home/herbert/zephyr-external/zephyr/twister-out/intel_adsp_cavs18/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/modules/mipi-sys-t/library/include/mipi_syst.h:600,
                 from /home/herbert/zephyr-external/zephyr/subsys/logging/log_output_syst.c:11:
/home/herbert/zephyr-external/zephyr/subsys/logging/mipi_syst/platform.h:16: error: "MIPI_SYST_PCFG_PRINTF_ARGBUF_SIZE" redefined [-Werror]
   16 | #define MIPI_SYST_PCFG_PRINTF_ARGBUF_SIZE CONFIG_LOG_MIPI_SYST_ARGS_BUFFER_SIZE
      | 
/home/herbert/zephyr-external/zephyr/twister-out/intel_adsp_cavs18/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/modules/mipi-sys-t/library/include/mipi_syst.h:532: note: this is the location of the previous definition
  532 | #define MIPI_SYST_PCFG_PRINTF_ARGBUF_SIZE 1024
      | 
cc1: all warnings being treated as errors
```

Observed on latest main (at time of writing, 4f81568286747f7aa74c7dda5da3c536226616c7).

**To Reproduce**

```
$ twister -b -p intel_adsp_cavs15 -p intel_adsp_cavs18 -p intel_adsp_cavs25 -p intel_adsp_ace15_mtpm -T tests/subsys/logging/log_syst -T tests/subsys/logging/log_switch_format
```

**Expected behavior**
Build to succeed

**Impact**
Not sure, will leave it up to @aasthagr 

**Logs and console output**

```
intel_adsp_cavs15    tests/subsys/logging/log_switch_format/logging.log_switch_format.immediate FAILED: Build failure
intel_adsp_cavs15    tests/subsys/logging/log_switch_format/logging.log_switch_format.deferred FAILED: Build failure
intel_adsp_cavs15    tests/subsys/logging/log_syst/logging.log_syst.mipi_syst FAILED: Build failure

intel_adsp_cavs18    tests/subsys/logging/log_switch_format/logging.log_switch_format.immediate FAILED: Build failure
intel_adsp_cavs18    tests/subsys/logging/log_switch_format/logging.log_switch_format.deferred FAILED: Build failure
intel_adsp_cavs18    tests/subsys/logging/log_syst/logging.log_syst.mipi_syst FAILED: Build failure

intel_adsp_cavs25    tests/subsys/logging/log_switch_format/logging.log_switch_format.immediate FAILED: Build failure
intel_adsp_cavs25    tests/subsys/logging/log_switch_format/logging.log_switch_format.deferred FAILED: Build failure
intel_adsp_cavs25    tests/subsys/logging/log_syst/logging.log_syst.mipi_syst FAILED: Build failure
```

nashif, 2022-08-30:
did you do we `west update`? This sample builds just fine!

laurenmurphyx64, 2022-08-30:
Damnit... just did that and it passed. Sorry @aasthagr, that should've been the first thing I did trying to file a bug for a module

closed 2022-08-30 by laurenmurphyx64