zephyrproject-rtos/zephyr #49656: acrn_ehl_crb: testcases tests/kernel/smp failed to run on v2.7-branch

Zhaoningx, 2022-08-30:
**Describe the bug**
The test case tests/kernel/smp failed on v2.7-branch on the ACRN board. I tested the case on the main branch, and it works.

**To Reproduce**
twister -p acrn_ehl_crb --device-testing --device-serial-pty=$HOME/acrn-env/acrn-test-pty.exp --west-flash=$HOME/acrn-env/acrn_efi.sh -v -T tests/kernel/smp/ -v

**Logs and console output**
```
ACRN:\>
DEBUG   - DEVICE: ACRN:\>vm_console 0
DEBUG   - DEVICE: 
DEBUG   - DEVICE: ----- Entering VM 0 Shell -----
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.7.3  ***
DEBUG   - DEVICE: Running test suite smp
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_coop_threads
DEBUG   - DEVICE: PASS - test_smp_coop_threads in 0.563 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cpu_id_threads
DEBUG   - DEVICE: PASS - test_cpu_id_threads in 1.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_coop_resched_threads
DEBUG   - DEVICE: PASS - test_coop_resched_threads in 0.301 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_preempt_resched_threads
DEBUG   - DEVICE: PASS - test_preempt_resched_threads in 0.512 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_yield_threads
DEBUG   - DEVICE: PASS - test_yield_threads in 0.301 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sleep_threads
DEBUG   - DEVICE: PASS - test_sleep_threads in 1.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_wakeup_threads
DEBUG   - DEVICE: PASS - test_wakeup_threads in 0.201 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_ipi
DEBUG   - DEVICE: cpu num=2 PASS - test_smp_ipi in 0.301 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_get_cpu
DEBUG   - DEVICE: PASS - test_get_cpu in 0.51 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_fatal_on_smp
DEBUG   - DEVICE: E: RAX: 0x0000000000000003 RBX: 0x0000000000135168 RCX: 0x00000000000f4240 RDX: 0x0000000d00000000
DEBUG   - DEVICE: E: RSI: 0x0000000000000200 RDI: 0x00000000000007d0 RBP: 0x00000000001203c0 RSP: 0x00000000001203c0
DEBUG   - DEVICE: E:  R8: 0x0000000000000001  R9: 0x00000000001388d8 R10: 0x0000000000000000 R11: 0x0000000000000000
DEBUG   - DEVICE: E: R12: 0x000000000010f360 R13: 0x0000000000000200 R14: 0x000000000010f360 R15: 0x0000000000000000
DEBUG   - DEVICE: E: RSP: 0x00000000001203c0 RFLAGS: 0x0000000000000206 CS: 0x0018 CR3: 0x0000000000140000
DEBUG   - DEVICE: E: RIP: 0x00000000001009bb
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 1
DEBUG   - DEVICE: E: Current thread: 0x10e940 (test_fatal_on_smp)
DEBUG   - DEVICE: PASS - test_fatal_on_smp in 0.5 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_workq_on_smp
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/smp/src/main.c:712: test_workq_on_smp: k_work_busy_get(&work) not equal to 0
DEBUG   - DEVICE: 
DEBUG   - DEVICE: FAIL - test_workq_on_smp in 0.51 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_release_global_lock
DEBUG   - DEVICE: E: RAX: 0x0000000000000003 RBX: 0x0000000005a995c0 RCX: 0x0000000000000000 RDX: 0x0000000000000000
DEBUG   - DEVICE: E: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x000000000011d7f0 RSP: 0x000000000011d7e8
DEBUG   - DEVICE: E:  R8: 0x0000000000100149  R9: 0x000000000010a862 R10: 0x0000000000000000 R11: 0x0000000000000000
DEBUG   - DEVICE: E: R12: 0x000000008b93a39c R13: 0x0000000000000200 R14: 0x000000000010e940 R15: 0x0000000000000000
DEBUG   - DEVICE: E: RSP: 0x000000000011d7e8 RFLAGS: 0x0000000000000202 CS: 0x0018 CR3: 0x0000000000140000
DEBUG   - DEVICE: E: RIP: 0x0000000000100152
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 1
DEBUG   - DEVICE: E: Current thread: 0x10d760 (unknown)
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/smp/src/main.c:649: k_sys_fatal_error_handler: (main_thread_id != child_thread_id is false)
DEBUG   - DEVICE: fatal on the same core
DEBUG   - DEVICE: FAIL - test_smp_release_global_lock in 0.23 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_inc_concurrency
DEBUG   - DEVICE: type 0: cnt 60000, spend 8 ms
DEBUG   - DEVICE: type 1: cnt 60000, spend 73 ms
DEBUG   - DEVICE: type 2: cnt 60000, spend 123 ms
DEBUG   - DEVICE: PASS - test_inc_concurrency in 0.203 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite smp failed.
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 14.1
 - Commit: b9fc341

mmahadevan108, 2022-08-30:
cc @cfriedt

Zhaoningx, 2022-09-01:
This issue is not stable to reproduce, it's difficult to bisect. Still working on looking for what's commit caused this bug.

fabiobaltieri, 2022-09-21:
@Zhaoningx any update?

cfriedt, 2022-09-22:
Tests with a randomized value and tests with SMP enabled can sometimes fail due to race conditions and timeouts. Sometimes we need to increase timeouts as a result. In CI, we explicitly retry tests with `--retry-failed $N` for some N (I usually choose 3, and most times the test pass, but sometimes they can fail once).

I would be much more concerned if tests fail in a majority of 10 trials.

@Zhaoningx - can you retry with `--retry-failed 10`?

mengxianglinx, 2022-09-23:
Thank you @cfriedt , sure, run this test case with "--retry-failed 10" it definitely pass, but "--retry-failed 3" it doesn't always pass.
So if "--retry-failed 10" is acceptable for daily testing, it could be a solution.

cfriedt, 2022-09-23:
CI tries 3 times, for référence. It could be timing related as well, so perhaps increasing some timeout would make the test more reliable

https://github.com/zephyrproject-rtos/zephyr/blob/fad899d2ad5e3c5b857ee0b5a331185addcc8deb/.github/workflows/twister.yaml#L136

mengxianglinx, 2022-09-24:
acrn_ehl_crb use apic_tsc timer, for acrn, I suspect that CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000 is too high, I set it to 1000, 800, 500, the failed test cases passed.

cfriedt, 2022-09-24:
> acrn_ehl_crb use apic_tsc timer, for acrn, I suspect that CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000 is too high, I set it to 1000, 800, 500, the failed test cases passed.

Nice catch @Zhaoningx - do you know if this bug is present or has already been fixed in `main`? In either case, are you able to make a PR?

mengxianglinx, 2022-09-25:
This bug is not present in main, not sure it is fixed by some patches or by side effect of some patches.

enjiamai, 2022-10-13:
After debugging, I found issue happens because at certain point of timing, the hypervisor will have a long delay.
(At about uptime at 106 ms, 632 ms, we will get many cycles late when we call k_cycle_get_32/64(). Actually, we found this via  the tests/kernel/timer/timer_behavior test, add some debug info we see the clock cycle we got has large drift at this two timing. 
These points affects some timing sensitive testcases (such as tests/timer/timer_behavior, tests/kernel/smp etc) or even some scheduling tests those measure the time.We are still not totally clear why the hypervisor will have such a delay at these point. It is most likely not an issue of Zephyr, because the clock cycle read from rdtsc provided by hypervisor.

Although the failed test are test_workq_on_smp() and test_smp_release_global_lock(), but the failure only have something to do with the timing of the tests executed.   

A suggested workaround solution is to add a 700ms boot delay to avoid these point. We can add CONFIG_BOOT_DELAY=700 for it.

After applying the 700ms DELAY for acrn_ehl_crb, test the SMP test pass in 20 times.  Hi, @mengxianglinx , you can verify it, thanks.

mengxianglinx, 2022-10-17:
Yes @enjiamai , I run twister with CONFIG_BOOT_DELAY=700 15 times,  the failed case passed.

closed 2022-10-17 by mengxianglinx