zephyrproject-rtos/zephyr #49659: logging: LOG_* appends 0x0D to 0x0A

koffes, 2022-08-30:
**Describe the bug**
When printing raw hex values a `CR (0x0A)` is automatically appended if you print a `LF (0x0A)`.

**Impact**
This causes parsers which depend on raw hex data to fail.
Take for instance `LOG_PRINTK("%c%c", 0x0A, 0x33)` will print as` 0x0A 0x0D 0x33`.

**Expected behavior**
A global KConfig to turn off this function would work, but is not ideal as it can break other parts of an application.
Preferred solution would be to have some LOG_*** which uses a parallell `printk` which does not append `CR`.

koffes, 2022-09-02:
Relates to https://github.com/zephyrproject-rtos/zephyr/issues/23019

dstudejio, 2022-09-04:
@koffes I was about to open an issue and found this one. 

The order in which `cr_out_func` appends the \r may also be at issue.  We noticed because of this function, LOG_PRINT("hello_nasty\n"); gets output as "hello_nasty\n\r" where we'd expect to see \r\n.

PR #49680 (2022-09-05): logging: Add LOG_RAW macro
Add macro for logging raw formatted string. It is similar to
LOG_PRINTK macro but contrary to LOG_PRINTK it should not append
carriage return character to any new line character found in the
string. LOG_PRINTK processed by log_output module has that to
mimic printk behavior.

Fixes #49659.
closed 2022-09-05 by cfriedt