zephyrproject-rtos/zephyr #49677: STM32U5 consumes more current using power management

vibhor-meshram, 2022-08-30:
**Describe the bug**
I am using nucleo_u575zi_q board. I want to implement shutdown mode using power management scheme and wake up the MCU using an external interrupt. For starters I am testing  zephyr\samples\boards\stm32\power_mgmt\blinky example to test the basic power management. I am using zephyr 3.10 version.

I created power states nodes in nucleo_u575zi_q.dts file (since nodes were not present there).
power-states {
		stop0: state0 {
			compatible = "zephyr,power-state";
			power-state-name = "suspend-to-idle";
			substate-id = <1>;
			min-residency-us = <100>;
		};
		stop1: state1 {
			compatible = "zephyr,power-state";
			power-state-name = "suspend-to-idle";
			substate-id = <2>;
			min-residency-us = <500>;
		};
		stop2: state2 {
			compatible = "zephyr,power-state";
			power-state-name = "suspend-to-idle";
			substate-id = <3>;
			min-residency-us = <900>;
		};

&cpu0 {
	cpu-power-states = <&stop0 &stop1 &stop2>;
};

In the blinky proj.conf file following is the configuration I have:

CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
CONFIG_CONSOLE=n
CONFIG_SERIAL=n

**Expected behavior**
When CONFIG_PM is set to y, the MCU should consume low power than what it would consume in active state (CONFIG_PM=n).

**Observation**
If 
CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
CONFIG_CONSOLE=n
CONFIG_SERIAL=n

Current Consumption = 12.8 mA

If 
CONFIG_PM=n
CONFIG_PM_DEVICE=n
CONFIG_PM_DEVICE_RUNTIME=n
CONFIG_CONSOLE=n
CONFIG_SERIAL=n

Current Consumption = 7.3 mA

**

**Impact**
High

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain Zephyr SDK V 3.10

erwango, 2022-08-30:
@vibhor-meshram Did you enable lptim node in order to use it as a LP ticker ?
```
&clk_lse {
	status = "okay";
}

&lptim1 {
	clocks = <&rcc STM32_CLOCK_BUS_APB3 0x00000800>,
		 <&rcc STM32_SRC_LSE LPTIM1_SEL(3)>;
	status = "okay";
};
```

vibhor-meshram, 2022-08-31:
Hi @erwango I added the above nodes. No change in in output. Adding my dts file for your reference.
[nucleo_u575zi_q-common.txt](https://github.com/zephyrproject-rtos/zephyr/files/9458433/nucleo_u575zi_q-common.txt)

erwango, 2022-09-07:
@vibhor-meshram I'm closing the point as can't be reproduced. Please re-open if you have more inputs.

closed 2022-09-07 by erwango