zephyrproject-rtos/zephyr #49678: Zephyr 3.2 module updates overview

fabiobaltieri, 2022-08-30:
Umbrella issue for tracking requests for updating various modules before Zephyr 3.2, listing all modules and an overview of the status compared to the correseponding upstream:

- [ ] canopennode
  - last updated somewhere in 2020
  - upstream at https://github.com/CANopenNode/CANopenNode
- [x] chre
- [ ] cmsis
  - at 5.8.0, 5.9.0 avilable
  - https://github.com/ARM-software/CMSIS_5/releases
- [x] edtt
- [x] fatfs
- [ ] fff
  - checked out at 2019, upstream has recent commits
  - https://github.com/meekrosoft/fff
- [ ] hal_altera
  - link in the README not working, references quartus v17.0, 21.1.1 is available
  - https://www.intel.com/content/www/us/en/collections/products/fpga/software/downloads.html?s=Newest
- [ ] hal_atmel
  - current version unclear, latest v3.52.0
  - https://www.microchip.com/en-us/tools-resources/develop/libraries/advanced-software-framework
- [ ] hal_espressif
  - on 4.3, 4.4 available
  - https://github.com/espressif/esp-idf
- [x] hal_gigadevice
- [x] hal_infineon
- [x] hal_microchip
   - upstream seems stale https://github.com/MicrochipTech/Peripheral-mec1701
- [x] hal_nordic
- [ ] hal_nuvoton
  - never been updated, upstream has recent commits
  - https://github.com/OpenNuvoton/M480BSP
- [x] hal_nxp
- [x] hal_openisa
  - upstream is stale https://github.com/open-isa-rv32m1/rv32m1_sdk_riscv
- [ ] hal_quicklogic
  - upstream has recent commits
  - https://github.com/QuickLogic-Corp/qorc-sdk/commits/master/HAL/src
- [ ] hal_rpi_pico
  - at 1.3.0, 1.4.0 just released
  - https://github.com/raspberrypi/pico-sdk/commits/master
- [ ] hal_silabs
  - at 3.1.0 (?), 4.1.0 available
  - https://github.com/SiliconLabs/gecko_sdk
- [x] hal_st
- [x] hal_stm32
- [x] hal_telink
- [ ] hal_ti
  - at simplelink 4.40.04.04, # out of date, simplelink 6.20.00.29 available
  - https://www.ti.com/tool/SIMPLELINK-CC13XX-CC26XX-SDK#downloads
- [ ] hal_xtensa
  - can't quite tell where upstream is
- [x] libmetal
- [x] liblc3
- [ ] littlefs
  - at v2.4, v2.5 available
  - https://github.com/littlefs-project/littlefs
- [x] loramac-node
- [ ] lvgl
  - at v8.2.0, v8.3.1 is available
  - https://github.com/lvgl/lvgl
- [ ] lz4
  - local repo is somewhere around 1.9.3, v1.9.4 available
  - https://github.com/lz4/lz4
- [ ] mbedtls
  - at v3.1.0, v3.2.1 available
  - https://github.com/Mbed-TLS/mbedtls
- [x] mcuboot
- [x] mipi-sys-t
- [x] nanopb
- [x] net-tools
  - no upstream
- [x] nrf_hw_models
- [x] open-amp
- [x] openthread
- [x] picolibc
- [ ] segger
  - at v3.30, v3.32 available
  - https://www.segger.com/downloads/systemview/
- [x] sof
- [ ] tflite-micro
  - out of date, stuck somewhere in 2021
  - https://github.com/tensorflow/tflite-micro
- [x] tinycbor
  - deprecated
- [x] tinycrypt
  - upstream last commit in 2019
  - https://github.com/intel/tinycrypt
- [ ] TraceRecorderSource
  - at 4.6.3, 4.6.6 available
  - https://github.com/percepio/TraceRecorderSource
- [x] trusted-firmware-m
- [x] tf-m-tests
- [ ] psa-arch-tests
  - mostly updated, few new upstream commits
  - https://github.com/ARM-software/psa-arch-tests/commits/main
- [ ] uoscore-uedhoc
  - out of date, more commits upstream
  - https://github.com/eriptic/uoscore-uedhoc/commits/main
- [x] zcbor
- [x] zscilib
  - no upstream

henrikbrixandersen, 2022-08-30:
> * [ ]  canopennode
>   * last updated somewhere in 2020
>   * upstream at https://github.com/CANopenNode/CANopenNode

We are tracking upstream v1.3-master (https://github.com/CANopenNode/CANopenNode/commits/v1.3-master) and as such we are up-to-date (except a couple of commits which breaks the upstream API).

stephanosio, 2022-08-30:
> [ ] cmsis
> * at 5.8.0, 5.9.0 avilable
> * https://github.com/ARM-software/CMSIS_5/releases

There is already an issue open for this:
https://github.com/zephyrproject-rtos/zephyr/issues/45279

Unfortunately, it does not look like it will be done for 3.2.

fabiobaltieri, 2022-08-31:
@yperess @nashif @nandojve @sylvioalves @glaubermaroto @uLipe @MulinChao @ChiHuaL @WealianLiao @kowalewskijan @yonsch @yonsch @mnkp @chrta @cfriedt @gmarull @dcpleung @brgl @Navin-Sankar @d3zd3z @ceolin @nordic-krch @laurenmurphyx64 @tleksell-pe @microbuilder @rlubos

Hi! If you are listed here you are listed as maintainer for a Zephyr module that has updates available upstream. The merge window closes on September 9th, does any of these needs an update? Feel free to tick the corresponding item, add a comment, open an issue, a PR or whatever suits.

Thanks!

yperess, 2022-08-31:
@fabiobaltieri fff's commit is just a restructure of the directory and CI, there's another on the way which I hope to merge shortly. That last one will actually change the header and I'll pull both at the same time.

chrta, 2022-08-31:
>  - [ ] hal_silabs (@yonsch @mnkp @chrta)
at 3.1.0 (?), 4.1.0 available
https://github.com/SiliconLabs/gecko_sdk

I do not plan to update the sdk. The version information (that the currently used version is 3.1.0) is correct. See also https://github.com/zephyrproject-rtos/zephyr/issues/43139

nandojve, 2022-08-31:
Hi @fabiobaltieri ,

> hal_atmel (@nandojve)
> current version unclear, latest v3.52.0

hal_atmel did not track ASF versions. The hal_atmel only have registers files and are added/updated based on http://packs.download.atmel.com. I think there won`t be any new activity for v3.2.

hal_gigadevice is usually tracked by @gmarull and me and I believe there is no pending things till moment.

dcpleung, 2022-08-31:
upstream of hal_xtensa is https://github.com/foss-xtensa/xtensa-hal, which has no update.

sylvioalves, 2022-08-31:
@fabiobaltieri, `hal_espressif` will be update to v4.4.1 with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/49682

ChiHuaL, 2022-09-01:
>- [ ] hal_nuvoton (@MulinChao @ChiHuaL @WealianLiao)
never been updated, upstream has recent commits
https://github.com/OpenNuvoton/M480BSP

Actually, We (@MulinChao and @ChiHuaL) are not maintainers for the HAL.
We maintain Nuvoton EC series chip (NPCX series) and put everything in the Zephyr repository based on the disscussion [here](https://github.com/zephyrproject-rtos/hal_nuvoton/pull/2#issuecomment-671139938). We didn't use the hal for EC chip.

The Hal part is used for general MCU chips (Numicro) in Nuvoton.
The hal maintainer should be @ssekar15.

ssekar15, 2022-09-01:
>hal_nuvoton (@MulinChao @ChiHuaL @WealianLiao)
>never been updated, upstream has recent commits
>https://github.com/OpenNuvoton/M480BSP

@fabiobaltieri 
I will do it

cfriedt, 2022-09-05:
There is no rush to update hal_ti right now. I am in discussion with 2 candidates (TI employees) who may be better as maintainers.

yonsch, 2022-09-05:
No need to update hal_rpi_pico for now