zephyrproject-rtos/zephyr #49687: Allow having .west folder and west.yml in the same folder

SkryptX, 2022-08-30:
We would like to setup our project with the `.west` folder and the `west.yml` in the same folder. That is currently not possible, because `west init` is inflexible and really wants to create its `.west` folder **next to** the specified directory (even if it is a `.`). I currently see no reason why this should be enforced since changing the `.west/config` file to the content below works just fine (with `west update` at least). Maybe there is a deeper reason why it is the way it is, which I haven't found out yet?

To support it like the `west init` `directory` argument would have to be uncoupled from the `--mf` argument, but I lack the foresight for a good suggestion on the correct and backwards-compatible command line interface. 

Directory structure
```
projectname
-- .west
---- config
-- west.yml
```

.west/config
```
[manifest]
path = .
file = west.yml

[zephyr]
base = zephyr
```

gregshue, 2022-08-30:
Won't your proposal also end up with a manifest repository only having `west.yml`?  If so, it seems you have gained nothing.

Note that a workspace-level repository would be a prime location for workspace-scope content (e.g., document style templates, workspace-level documentation).  This very quickly grows into T2 where your repository is the center of the hub rather than `zephyr/`.

SkryptX, 2022-09-01:
My proposed T4 is would allow something like this:

`*` means it is part of the workspace git repository
```
projectname (aka workspace; generated by git clone)
-- *.git
-- .west (generated by west init)
---- config
-- *application1
---- ...
---- *west.yml
-- *application2
---- ...
---- *west.yml
-- modules (handled by west.yml)
-- zephyr (handled by west.yml)
-- *west.yml
-- *Jenkinsfile
-- *README.md
```
Like this it is easy to navigate and open in an IDE like VSCode and prevents the spam of too many Git repositories, which usually incurs a higher administrative overhead. It also enables top-level files like a README for the whole project and how to set it up. West currently only supports a workflow where it replaces Git on the workspace level and this might just not be desired in every case.

SkryptX, 2022-09-01:
I just realized that a change like this would need a significant rethinking of the way Zephyr and West do things. Currently the whole Zephyr/West world is based on the workspace and the manifests which define projects which define the modules which define all the rest of the chain. 

In my opinion it is a shame west is designed the way it is, since it makes so many assumptions and leaves so little wiggle room for different ideas of versioning and Git usage. I might have to consider doing away with west since I will probably not get enough support for my idea to change this whole thing, open it up and make it more flexible.

gregshue, 2022-09-01:
West is supposed to be optional in the Zephyr environment.  It is a very thin layer on top of other scripts and services. IIRC, Google (platinum member) does not use west. It does not need to solve other ideas of versioning and Git usage.

It sounds like you would be better served by having zephyr and other modules be tied into your repository as Git submodules.

johnhatten2, 2024-05-29:
I think I came across the same need and I found a solution. We basically wanted to support 2 applications in a single git repo, and have the .git at the same level as the .west. This way, someone can clone the repo, run west init/west update, the zephyr/ and modules/ folder would spawn inside the git repo instead of polluting the top folder. 

Taking the [zephyr example application](https://github.com/zephyrproject-rtos/example-application), we create a example-application/west-manifest/ folder and we move west.yml in it. We added an example-application/app2 folder for our 2nd app for our needs. This is similar to a [T3 topology](https://github.com/zephyrproject-rtos/example-application) with the exception there is less git repos

example-application/           # Both a .west/ workspace and a .git/ repo
├── app1/               
│   ├── CMakeLists.txt
│   ├── prj.conf
│   └── src/
│       └── main.c
├── app2/                        
│   ├── CMakeLists.txt
│   ├── prj.conf
│   └── src/
│       └── main.c
├── boards/
├── doc/
├── manifest-repo/      
│   └── west.yml        # main manifest, import zephyr, stm32, cmsis
├── modules/
│   └── hal/
│       └── stm32/      # .git/ submodule 
│       └── cmsis/      # .git/ submodule 
└── zephyr/             # .git/ submodule

kaspt, 2025-01-08:
https://github.com/zephyrproject-rtos/zephyr/issues/49687#issuecomment-2137290986

Lock nice this solution. But for me is not clear do you manage your dependencies with `west update` and use the revisions in the west.yml or do you use it without west and do a `git submodule update`

johnhatten2, 2025-01-08:
> [#49687 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/49687#issuecomment-2137290986)
> 
> Lock nice this solution. But for me is not clear do you manage your dependencies with `west update` and use the revisions in the west.yml or do you use it without west and do a `git submodule update`

We use git clone, then west init and west update (whenever an update is needed).
Since there's a single west.yaml, both applications share the same dependencies (never a problem with our developpement process)