zephyrproject-rtos/zephyr #49703: eSPI: Add platform specific Slave to Master Virtual Wires

vphirema, 2022-08-30:
**Is your feature request related to a problem? Please describe.**
As per eSPI spec, there are about 64 platform specific Slave to Master Virtual Wires that can be used for internal usages.
Current code doesn't support it hence need to add the support.

Table 2-14. PCH/SoC eSPI Virtual Wires Mapping (Address 50h to 5Fh)
https://cdrdv2.intel.com/v1/dl/getContent/562633?explicitVersion=true

**Describe the solution you'd like**
Add about 8 platform platform specific Slave to Master Virtual Wires in  ./include/zephyr/drivers/espi.h and extend the list when needed. 
Also SOC specific code needs to be implemented in respective EC vendors code

**Describe alternatives you've considered**
None

**Additional context**
In order to reduce the physical GPIOS going from EC/PD, Virtual Wire GPIOs (0 to 5) are used to indicate Type-C overcurrent in Intel Meterolake SOC hence need to add this support in the Zephyr EC code

vphirema, 2022-08-30:
cc: @nashif  @MulinChao @keith-zephyr
I have started implementing the code in Chromium repo. Need to publish it in Zephyr code base
https://chromium-review.googlesource.com/c/chromiumos/third_party/zephyr/+/3849741

keith-zephyr, 2022-08-31:
Not directly related to your change, but has Intel defined replacement terms for master/slave with the eSPI protocol?  The regular SPI protocol has adopted controller/peripheral. https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names/

vphirema, 2022-08-31:
Hi Keith, I will file an internal bug and see if the eSPI owners can change the naming convention

vphirema, 2022-09-02:
https://github.com/zephyrproject-rtos/zephyr/pull/49710
is merged hence closing

closed 2022-09-02 by vphirema