zephyrproject-rtos/zephyr #49711: tests/arch/common/timing/arch.common.timing.smp fails for CAVS15, 18

laurenmurphyx64, 2022-08-31:
**Describe the bug**
tests/arch/common/timing/arch.common.timing.smp fails for CAVS 15 and CAVS 18.

**To Reproduce**
```
twister --disable-suite-name-check -p intel_adsp_cavs18 --device-testing --device-serial-pty=/home/herbert/zephyr-intel/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh --west-flash=/home/herbert/zephyr-intel/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh --no-skipped-report -x=CONFIG_BOOT_DELAY=500 -v -s tests/arch/common/timing/arch.common.timing.smp
```

**Logs and console output**
Same `handler.log` for both:

```
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.1.0-4409-ge6d7118f0ee7  (delayed boot 500ms) ***
Running TESTSUITE arch_timing
===================================================================
START - test_arch_timing
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/common/timing/src/main.c:66: perform_tests: diff1_ns not within WAIT_NS +/- WAIT_NS * TOLERANCE
FAIL - test_arch_timing in 0.003 seconds
===================================================================
START - test_arch_timing_smp
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/common/timing/src/main.c:66: perform_tests: diff1_ns not within WAIT_NS +/- WAIT_NS * TOLERANCE
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/common/timing/src/main.c:66: perform_tests: diff1_ns not within WAIT_NS +/- WAIT_NS * TOLERANCE
FAIL - test_arch_timing_smp in 0.003 seconds
===================================================================
TESTSUITE arch_timing failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [arch_timing]: pass = 0, fail = 2, skip = 0, total = 2 duration = 0.006 seconds
- FAIL - [arch_timing.test_arch_timing] duration = 0.003 seconds
- FAIL - [arch_timing.test_arch_timing_smp] duration = 0.003 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: e48a778ee91d4220b57b33fd693b8065
PROJECT EXECUTION FAILED
```

**Expected behavior**
Test pass

**Impact**
Unsure, will leave it to @edersondisouza

**Environment**
 - Ubuntu 20.04
 - Zephyr SDK (also confirmed for XCC)
 - Latest main at time of writing (e6d7118f0ee7fbf542124ab6845ba6879de8aaf7)

laurenmurphyx64, 2022-08-31:
What else is weird is that I can't build the same test with XCC for CAVS 15, 18 or 25 (I know I can build with XCC, just built `hello_world`)

XCC environment:
```
XTENSA_BUILDS_DIR=/home/herbert/xcc/install/builds/RG-2017.8-linux
XTENSA_CORE=X6H3CNL_2017_8
XTENSA_SYSTEM=/home/herbert/xcc/install/builds/RG-2017.8-linux/X6H3CNL_2017_8/config
XTENSA_TOOLCHAIN_PATH=/home/herbert/xcc/install/tools
XTENSA_TOOLS_ROOT=/home/herbert/xcc
XTENSA_TOOLS_VERSION=RG-2017.8-linux
```

```
$  twister --build-only -p intel_adsp_cavs15 -p intel_adsp_cavs18 -p intel_adsp_cavs25 -s tests/arch/common/timing/arch.common.timing.smp -v
```

`build.log`:

```
...
external/zephyr/tests/arch/common/timing/src/main.c
cc1: warnings being treated as errors
/home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:83: warning: ‘arch_timing_test_arch_timing’ declared inline after being called
/home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:83: warning: previous declaration of ‘arch_timing_test_arch_timing’ was here
/home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:102: warning: ‘arch_timing_test_arch_timing_smp’ declared inline after being called
/home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:102: warning: previous declaration of ‘arch_timing_test_arch_timing_smp’ was here
...
```

edersondisouza, 2022-08-31:
> What else is weird is that I can't build the same test with XCC for CAVS 15, 18 or 25 (I know I can build with XCC, just built `hello_world`)
> 
> XCC environment (dunno if you need alla this):
> 
> ```
> XTENSA_BUILDS_DIR=/home/herbert/xcc/install/builds/RG-2017.8-linux
> XTENSA_CORE=X6H3CNL_2017_8
> XTENSA_SYSTEM=/home/herbert/xcc/install/builds/RG-2017.8-linux/X6H3CNL_2017_8/config
> XTENSA_TOOLCHAIN_PATH=/home/herbert/xcc/install/tools
> XTENSA_TOOLS_ROOT=/home/herbert/xcc
> XTENSA_TOOLS_VERSION=RG-2017.8-linux
> ```
> 
> ```
> $  twister --build-only -p intel_adsp_cavs15 -p intel_adsp_cavs18 -p intel_adsp_cavs25 -s tests/arch/common/timing/arch.common.timing.smp -v
> ```
> 
> `build.log`:
> 
> ```
> ...
> external/zephyr/tests/arch/common/timing/src/main.c
> cc1: warnings being treated as errors
> /home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:83: warning: ‘arch_timing_test_arch_timing’ declared inline after being called
> /home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:83: warning: previous declaration of ‘arch_timing_test_arch_timing’ was here
> /home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:102: warning: ‘arch_timing_test_arch_timing_smp’ declared inline after being called
> /home/herbert/zephyr-external/zephyr/tests/arch/common/timing/src/main.c:102: warning: previous declaration of ‘arch_timing_test_arch_timing_smp’ was here
> ...
> ```

Does any `ZTEST` build with xcc?

PR #49771 (2022-09-01): soc/xtensa/intel_adsp/cavs: Fix XTENSA_CCOUNT_HZ value
The default value is actually 400MHz on cavs15 and cavs18.

Fixes: #49711

--

Note: In a ideal world, this info should come from clock control if available.
commit a7e91665 (2022-09-01): soc/xtensa/intel_adsp/cavs: Fix XTENSA_CCOUNT_HZ value

The default value is actually 400MHz on cavs15 and cavs18.

Fixes: #49711