zephyrproject-rtos/zephyr #49717: mcumgr: Bluetooth transport fix prevents passing GATT notify status back to SMP

nordicjm, 2022-08-31:
**Describe the bug**
This issue: https://github.com/zephyrproject-rtos/zephyr/issues/49661 and bug fix: https://github.com/zephyrproject-rtos/zephyr/pull/49691 resolves an issue which caused a bluetooth deadlock, however as a result, the result code of the send operation can no longer be passed back to SMP from the bluetooth code as doing so would involve use of the system work queue which would bring the issue back again. This can be resolved by either migrating all of SMP to a new work queue or thread so that it has no reliance on the system work queue, this may also be needed for additional transports e.g. UDP or CAN.

**Impact**
If there is an error in sending the data for whatever reason, SMP is unaware and thinks everything was sent successfully

nordicjm, 2022-09-01:
https://github.com/zephyrproject-rtos/zephyr/pull/49691 has been changed to resolve this issue

closed 2022-09-07 by nordicjm

zainakbari, 2026-05-25:
Adding a data point from NCS v3.3.0 in case it helps anyone debugging similar symptoms.

**Environment**: nRF52840 (XIAO BLE Sense), NCS v3.3.0, MCUmgr SMP over BLE transport. Client: Nordic iOSMcuManagerLibrary 1.14.3.

**Symptom**: after a successful image-mgmt upload (~30s, ~320 KB) + `ImageStatesWriteRequest(hash, confirm=False)` round-trip, the next `ResetWriteRequest` never returns. iOS client times out after 40s. Sensor stays connected throughout (LL link alive, GATT session held). No chip reboot. No HW WDT fire. SMP response just never arrives.

The fix from #49691 is present in NCS v3.3.0's `smp_bt.c::smp_bt_tx_pkt` (`bt_gatt_notify_cb` + completion semaphore + disconnect-aware early-exit). So this isn't exactly #49717, but the symptom shape (SMP response stranded; framework unaware) is similar.

**Mitigation** (per Nordic DevZone [#124402](https://devzone.nordicsemi.com/)): bumped BT TX buffer counts in `prj.conf`:

```
CONFIG_BT_BUF_ACL_TX_COUNT=7        # NCS default 3
CONFIG_BT_L2CAP_TX_BUF_COUNT=8      # NCS default 4
CONFIG_BT_ATT_TX_COUNT=7            # NCS default 3
```

**Measured before/after** on the same chip + same test sequence (mimicking iOSMcuManagerLibrary's `testAndConfirm` flow):
- Before bump: reset response **40-second timeout** (never arrives)
- After bump: reset response **0.028–0.029s** (BLE round-trip time)

~1400× difference on a binary otherwise byte-identical at slot0. The bump fixes the symptom, but I can't claim to have characterized the root cause. Plausible hypotheses:

1. `k_sem_take(&cpd->smp_notify_sem, K_FOREVER)` at lines 552-555 blocks waiting on a completion callback that doesn't fire under some particular controller-side TX-queue state.
2. The MTU-halve cascade on `-ENOMEM` (lines 524-540) returns "success" without all bytes sent if MTU drops below `SMP_BT_MINIMUM_MTU_SEND_FAILURE`.
3. iOS connection-event timing under sustained back-pressure starves the SMP response slot.

Happy to share my reproducer (NCS v3.3.0 strict-prod firmware + Python test script) if anyone wants to dig in.