zephyrproject-rtos/zephyr #49738: Bluetooth: Controller: Restarting periodic advertising causes crash when ADV_SYNC_PDU_BACK2BACK=y

jakkra, 2022-08-31:
**Describe the bug**
Stopping and then starting periodic advertising causes an assert here: https://github.com/zephyrproject-rtos/zephyr/blob/0612dd433d77bf2aee899170ea1efd491f325540/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_sync.c#L345

`ASSERTION FAIL [pdu] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_sync.c:345
`

Crash happens when using chained periodic advertising:
```
CONFIG_BT_CTLR_ADV_SYNC_PDU_BACK2BACK=y
CONFIG_BT_CTLR_DF_PER_ADV_CTE_NUM_MAX=5
```

 - What target platform are you using? nRF52833, Zephyr 
 - Is this a regression? Same issue on Zephyr version used in nRF Connect 2.0

**To Reproduce**
My code does this and you should be able to reproduce by adding this to the end of main in https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_tx:
```
    __ASSERT_NO_MSG(0 == bt_le_per_adv_stop(adv_set));
    __ASSERT_NO_MSG(0 == bt_le_ext_adv_stop(adv_set));
    struct bt_le_per_adv_param per_adv_param = {
        .interval_min = minAdvInterval,
        .interval_max = maxAdvInterval,
        .options = BT_LE_ADV_OPT_USE_TX_POWER,
    };
    int err = bt_le_per_adv_set_param(adv_set, &per_adv_param);
    if (err) {
        LOG_ERR("failed (err %d)\n", err);
        return;
    }
    LOG_INF("Periodic advertising enable...");
    int err = bt_le_per_adv_start(adv_set);
    if (err) {
        LOG_ERR("failed (err %d)\n", err);
        return;
    }
    LOG_INF("success\n");

    LOG_INF("Extended advertising enable...");
    err = bt_le_ext_adv_start(adv_set, &ext_adv_start_param);
    if (err) {
        LOG_ERR("failed (err %d)\n", err);
        return;
    }
    LOG_INF("success\n");
```
**Expected behavior**
No crash.

**Impact**
Cannot stop/start or change periodic adv. interval. when chained periodic adv. is used.

**Logs and console output**
```
<inf> app: Bluetooth initialized
<inf> app: Setting TxPower: 4
<inf> app: Set Tx Power: 4
<inf> bt_adv_aoa: success

<inf> bt_adv_aoa: Set ext adv data...
<inf> bt_adv_aoa: success

<inf> bt_adv_aoa: Update CTE params...
<inf> bt_adv_aoa: success

<inf> bt_adv_aoa: Periodic advertising params set...
<inf> app: Started blink thread
<inf> bt_adv_aoa: success

<inf> bt_adv_aoa: Enable CTE...
<inf> bt_adv_aoa: success

<inf> bt_adv_aoa: success

<inf> bt_adv_aoa: Periodic advertising enable...
<inf> bt_adv_aoa: success

<inf> bt_adv_aoa: Extended advertising enable...
<inf> bt_adv_aoa: success

<inf> app: Pressed button => Change adv. interval
<inf> app: New interval: 1 => 100
<inf> bt_adv_aoa: Adv stopped
<inf> bt_adv_aoa: success
<inf> bt_adv_aoa: Periodic advertising enable...
<inf> bt_adv_aoa: success
<inf> bt_adv_aoa: Extended advertising enable...
<inf> bt_adv_aoa: success
ASSERTION FAIL [pdu] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_sync.c:34
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr 
 - Commit SHA or Version used: 894423e9aed34a394b97c995c33549513b0ac4a5

cvinayak, 2022-09-02:
@jakkra Thank once again with the easy to reproduce steps. Please verify the fix in #49856 and revert back with the results.

jakkra, 2022-09-02:
@cvinayak Thanks for the quick fix, not crashing anymore with https://github.com/zephyrproject-rtos/zephyr/pull/49856

PR #49856 (2022-09-09): Bluetooth: Controller: Fix periodic advertising re-enable assertion
When periodic advertising with chaining is re-enabled after
changing interval or just re-enabled then the chain PDUs
where not duplicated causing the LLL to assert detecting the
presence of aux pointer but no linked chain PDUs.

Fixes #49738.
closed 2022-09-09 by nashif