zephyrproject-rtos/zephyr #49746: twister: extra test results

asemjonovs, 2022-08-31:
**Describe the bug**
Extra test results are generated in json report when `--build-only` and `--test-only` twister arguments are used.  Seems to occur when `src` folder is not present in same folder as `yaml` file.

```
        {
            "name":"tests/ztest/base/testing.ztest.base.verbose_0",
            "arch":"posix",
            "platform":"native_posix",
            "run_id":"56d4763e67f18b7d71cb3616509de23f",
            "runnable":true,
            "status":"passed",
            "execution_time":"2.04",
            "testcases":[
                {
                    "identifier":"testing.ztest.base.verbose_0",  # This is not a defined test case
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":null
                },
                {
                    "identifier":"testing.ztest.base.verbose_0.fixture_pointer",
                    "execution_time":"0.00",
                    "status":"passed"
                },
```

**To Reproduce**
Steps to reproduce the behavior:
1. mv tests/ztest/base/src/ tests/ztest/base/src_bug
2. modify tests/ztest/base/CMakeLists.txt (rename `src` to `src_bug`)
3. ./scripts/twister -c -T tests/ztest/base --build-only -p native_posix
4. ./scripts/twister -c -T tests/ztest/base --test-only -p native_posix
5. Observe json report contains test config names as test cases.

**Expected behavior**
Test reports to contain the defined test cases in the binary.

**Impact**
annoyance

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...) 
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nashif, 2022-09-07:
This is a known problem and will go away once we completely move to new ztest API and get list of test cases from the binary.

nashif, 2022-10-11:
This should be fixed by now with how we parse testcases using symbols.