zephyrproject-rtos/zephyr #49760: frdm_kl25z: sample.usb.dfu Kconfig issue causing build failure

stephanosio, 2022-08-31:
**Describe the bug**

Build fails for the `sample.usb.dfu` sample on `frdm_kl25z` due to Kconfig dependency issues.

**To Reproduce**

```
twister -v -N -M -p frdm_kl25z -s sample.usb.dfu
```

**Expected behavior**

Build succeeds.

**Logs and console output**

```
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/zephyr/zephyr/samples/subsys/usb/dfu
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /__w/zephyr/zephyr/.cache
-- Zephyr version: 3.1.99 (/__w/zephyr/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: frdm_kl25z
-- Found host-tools: zephyr 0.15.0 (/opt/toolchains/zephyr-sdk-0.15.0)
-- Found dtc: /opt/toolchains/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.15.0 (/opt/toolchains/zephyr-sdk-0.15.0)
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/arm/frdm_kl25z/frdm_kl25z.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/frdm_kl25z/samples/subsys/usb/dfu/sample.usb.dfu/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /__w/zephyr/zephyr/twister-out/frdm_kl25z/samples/subsys/usb/dfu/sample.usb.dfu/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/frdm_kl25z/samples/subsys/usb/dfu/sample.usb.dfu/zephyr/dts.cmake

warning: FLASH_PAGE_LAYOUT (defined at drivers/flash/Kconfig:54) has direct dependencies FLASH_HAS_PAGE_LAYOUT && FLASH with value n, but is currently being y-selected by the following symbols:
 - STREAM_FLASH (defined at subsys/storage/stream/Kconfig:7), with value y, direct dependencies y (value: y)

warning: FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) has direct dependencies FLASH_HAS_DRIVER_ENABLED with value n, but is currently being y-selected by the following symbols:
 - MCUBOOT_IMG_MANAGER (defined at subsys/dfu/Kconfig:25), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)
Parsing /__w/zephyr/zephyr/samples/subsys/usb/dfu/Kconfig
Loaded configuration '/__w/zephyr/zephyr/boards/arm/frdm_kl25z/frdm_kl25z_defconfig'
Merged configuration '/__w/zephyr/zephyr/samples/subsys/usb/dfu/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /__w/zephyr/zephyr/cmake/modules/kconfig.cmake:307 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /__w/zephyr/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /__w/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /__w/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: e9b29a6f830e6663495960078de8bd06c66726b9

stephanosio, 2022-08-31:
cc @dleach02 @mmahadevan108 @danieldegrasse

mmahadevan108, 2022-08-31:
This test needs to be excluded on `frdm_kl25z` as flash is disabled. I see the below error when running the test command, am I missing something in my environment?
```
~/zephyrproject/zephyr$ scripts/twister -v -N -M -p frdm_kl25z -s sample.usb.dfu
Renaming output directory to /home/mahesh/zephyrproject/zephyr/twister-out.1
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.1.0-4462-g838f5813003d
INFO    - Using 'zephyr' toolchain.
Traceback (most recent call last):
  File "scripts/twister", line 405, in <module>
    ret = main()
  File "scripts/twister", line 294, in main
    tplan.discover()
  File "/home/mahesh/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/testplan.py", line 118, in discover
    raise TwisterRuntimeError("No test cases found at the specified location...")
twisterlib.error.TwisterRuntimeError: No test cases found at the specified location...
```

closed 2022-09-01 by mmahadevan108