zephyrproject-rtos/zephyr #49764: Hash based dictionary logging

rbbrns, 2022-08-31:
Problem:
Zephyr dictionary logging relies on a json dictionary that must be regenerated for every image. This is because the dictionary maps the string to addresses. This creates friction for developers using dictionary based logging because they need to ensure they have the exact dictionary generated for their specific mage before they can decode any strings. 

Proposed Solution:
Instead of mapping strings to addresses, map strings to a hash. This allows developers to maintain a common dictionary for their project and append new strings as needed. This reduces the friction on developers because they can decode the vast majority of strings in a log without needing to fetch a new dictionary.

This is the approach taken by Pigweed in the [pw_tokenizer module](https://pigweed.dev/pw_tokenizer/)

Alternatives:
Instead of natively supporting hash based dictionary logging, Zephyr can provide an abstraction around the message encoding and decoding so developers can use a custom encoding scheme. The [custom logging frontend abstraction](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/logging/log_frontend.h) currently provided is insufficient since the message encoding has already occurred. 

Related issues: 
* https://github.com/zephyrproject-rtos/zephyr/issues/44942
* https://github.com/zephyrproject-rtos/zephyr/issues/42840

nashif, 2022-08-31:
@dcpleung  FYI

rbbrns, 2022-08-31:
@yperess

quic-egmc, 2022-10-17:
how much will token sizes grow?  an address is relatively small footprint whereas hashes can be substantially larger.  If we end up converting a 10 byte string into a 20 byte hash we've made things worse.

Also we don't want to use processor cycles to hash, especially with alternate frontends that try to minimize processor overhead.

keith-zephyr, 2022-10-17:
> how much will token sizes grow? an address is relatively small footprint whereas hashes can be substantially larger. If we end up converting a 10 byte string into a 20 byte hash we've made things [...]

The pw_tokenizer module uses a 32-bit hash for the tokens.  The [example](https://pigweed.dev/pw_tokenizer/#example-tokenized-logging) from the documentation demonstrates how a 41 byte length format string requires only 4 bytes in the binary at runtime.  With only a 32-bit hash, there is a chance for hash collisions, but Pigweed tooling does attempt to automatically resolve collisions.  The documentation also provides tips for manually resolving collisions.

The hash to string database is created at build time, and does not require any hashing code running on the board's CPU.

quic-egmc, 2022-10-17:
Thanks for clarifying.  This sounds really interesting.  Curious how preprocessor-based hashing works - if you happen have a pointer to the guts of the code that does this I'd be most appreciative.

rbbrns, 2022-10-18:
Take a look at the headers in  https://pigweed.googlesource.com/pigweed/pigweed/+/refs/heads/main/pw_tokenizer/public/pw_tokenizer/internal

hongshui3000, 2023-01-17:
similar reference:https://github.com/rokath/trice
https://interrupt.memfault.com/blog/trice#just-like-printf-but-better

rbbrns, 2023-01-17:
This change should make it possible for pigweed tokenizer and trice to work in zephyr: https://github.com/zephyrproject-rtos/zephyr/pull/53069#event-8094355056

closed 2026-10-23 by rbbrns