zephyrproject-rtos/zephyr #49786: nsim_em: tests: fail to run tests/kernel/timer/timer_behavior

shaoanxli, 2022-09-01:
**Describe the bug**
Numerical conditions are not met: max_us <= max_us_bound. And the value of max_us=20000.000000, max_us_bound=1100.000000.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p nsim_em -T tests/kernel/timer/timer_behavior/ -v
3. See error

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-4500-g202b2ff91fbc  ***^M
Running TESTSUITE timer_behavior^M
===================================================================^M
START - test_periodic_behavior^M
periodic timer behavior test^M
periodic timer samples gathered, calculating statistics^M
timer clock rate 5000000, kernel tick rate 100, expected period 1000 us, samples 10000, period 5000.000000 cycles^M
statistics samples 10000, rollovers 0, mean 19998.604380 us, variance 19475.604089 us, stddev 139.555023 us, min 6043.800000 us, max 20000.000000 us^M
statistics samples 10000, rollovers 0, mean 99993.021900 cycles, variance 486890.102221 cycles, stddev 697.775146 cycles, min 30219 cycles, max 100000 cycles^M
timer start cycle 20320, end cycle 999950539, total time 199986043.800000, expected time 10000000.000000, difference 189986043.800000^M
test log 20000.000000 1100.000000^M
^M
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/main.c:167: timer_behavior_test_periodic_behavior: (max_us <= max_us_bound is false)^M
Longest timer period too long (off by more than 10%)^M
 FAIL - test_periodic_behavior in 202.491 seconds^M
===================================================================^M
TESTSUITE timer_behavior failed.^M
^M
------ TESTSUITE SUMMARY START ------^M
^M
SUITE FAIL -   0.00% [timer_behavior]: pass = 0, fail = 1, skip = 0, total = 1 duration = 202.491 seconds^M
 - FAIL - [timer_behavior.test_periodic_behavior] duration = 202.491 seconds^M
^M
------ TESTSUITE SUMMARY END ------^M
^M
===================================================================^M
RunID: 8900688af049d75900a428f73091e678^M
PROJECT EXECUTION FAILED^M

```

**Environment :**
OS: Linux
Toolchain Zephyr SDK 0.15.0
Commit SHA or Version: 202b2ff91fbced27a325258fb49d1f27e2ff7566

nashif, 2022-09-01:
@teburd  should this be disabled for this simulator?

teburd, 2022-09-01:
If this is an emulated environment then yes, disable the test. The test does not work in simulated/emulated environments.

yinfangw, 2022-09-02:
on acrn_ehl_crb, the test case also failed:

START - test_periodic_behavior
periodic timer behavior test
periodic timer samples gathered, calculating statistics
timer clock rate 1900000000, kernel tick rate 10000, expected period 1000 us, samples 10000, period 1900000.000000 cycles
statistics samples 10000, rollovers 0, mean 999.994210 us, variance 52798.839798 us, stddev 229.779984 us, min 3.275263 us, max 23184.668947 us
statistics samples 10000, rollovers 0, mean 1899988.998100 cycles, variance 190603811669.693100 cycles, stddev 436581.937500 cycles, min 6223 cycles, max 44050871 cycles
timer start cycle 50252674852, end cycle 69252564833, total time 9999942.095263, expected time 10000000.000000, difference -57.904737

Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/main.c:164: timer_behavior_test_periodic_behavior: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than 10%)
FAIL - test_periodic_behavior in 0.963 seconds

yinfangw, 2022-09-02:
For this test case, it failed when using HPET timer by default. But it failed when using APIC TSC timer: detailed in https://github.com/zephyrproject-rtos/zephyr/issues/49611

PR #49824 (2022-09-02): test: timer: Disable nsim_em
The test will always fail on emulated/simulated environments. Exclude
this one which was failing.

Fixes #49786
closed 2022-09-02 by fabiobaltieri