zephyrproject-rtos/zephyr #49789: it8xxx2_evb: tests/crypto/tinycrypt/ test takes longer on sdk 0.15.0

hunterhu3000, 2022-09-01:
**Describe the bug**
tests/crypto/tinycrypt/ fails on sdk 0.15.0, because the timerout for this test is 500s, but it takes longer. If we set the timeout to 600s, the test can pass. 

Calculate the actual time of the test:
On SDK 0.15.0
Total 569s 
test_ecc_dh in 171.386 seconds
PASS - test_ecc_dsa in 181.298 seconds

On SDK 0.14.2
Total 363s
test_ecc_dh in 272.805 seconds
PASS - test_ecc_dsa in 287.976 seconds

Why these cases takes 100s more on 0.15.0?

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? it8xxx2_evb
 - Is this a regression? No, it is related to the SDK

**To Reproduce**
Steps to reproduce the behavior:
west build -p always -b it8xxx2_evb tests/crypto/tinycrypt/
OR
twister -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests/crypto/tinycrypt/ -vv

**Expected behavior**
0.15.0 should take equal or less time than 0.14.2

**Impact**
some applications will run slower on 0.15.0

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...) 0.15.0
 - Commit SHA or Version used: e6d7118f0ee7

hunterhu3000, 2022-09-01:
It looks like both functions are related to the montecarlo algorithm which is calculation intensive.
@GTLin08, Is it possible to locate which instruction takes longer in 0.15.0, and is it related to the compiler? Thanks

GTLin08, 2022-09-02:
I test with commit: [66da4c2](https://github.com/zephyrproject-rtos/zephyr/commit/66da4c285412fa60e3c5cdeac53dac476f0a8060) and on SDK 0.15.0, the result is pass. 
test_ecc_dh in 168.43 seconds
test_ecc_dsa in 177.955 seconds
Total = 38 duration = 349.826s
Before this commit: [66da4c2](https://github.com/zephyrproject-rtos/zephyr/commit/66da4c285412fa60e3c5cdeac53dac476f0a8060), there are many changes with test: crypto, you can test again after adding these commits.

hunterhu3000, 2022-09-03:
Yes, the issue can be closed, as it is fixed by the update.

closed 2022-09-03 by hunterhu3000