zephyrproject-rtos/zephyr #49790: Intel CAVS25: Failure in tests/boards/intel_adsp/smoke

shuai1x, 2022-09-01:
**Describe the bug**
On intel_adsp_cavs25 platform, tests/boards/intel_adsp/smoke failed.
This case has a one in ten chance of failure.
Bisect result: Can’t find the exact commit.

Please also mention any information which could help others to understand
the problem you're facing:
 - intel_adsp_cavs25
 - tests/boards/intel_adsp/smoke failed.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/boards/intel_adsp/smoke/ --no-skipped-report -vv
2. see error

**Logs and console output**
```
START - test_2nd_cpu_halt
halt/restart core 1...
halt/restart core 2...
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/cpus.c:167: halt_and_restart: (alive_flag is true)
cpu didn't halt
FAIL - test_2nd_cpu_halt in 0.251 seconds
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 15.0

edersondisouza, 2022-09-02:
This is an interesting one. Firstly, I could reproduce it, but the ratio wasn't anything like 1 in 10. From around 500 runs, I got the error 2 times. That in itself is curious. Secondly, I couldn't reproduce it using previous SDK (0.14.2), which is a hint for why this hadn't appeared before. Finally, I think I have fix, but I'm not sure I can explain it completely. Will send a PR soon.

edersondisouza, 2022-09-02:
Hi @shuai1x, could you please check if my draft PR https://github.com/zephyrproject-rtos/zephyr/pull/49877 fixes this issue for you? If it does, I'll make it a proper PR.

shuai1x, 2022-09-05:
> Hi @shuai1x, could you please check if my draft PR #49877 fixes this issue for you? If it does, I'll make it a proper PR.

Ok, will reply to you soon.

shuai1x, 2022-09-05:
> Hi @shuai1x, could you please check if my draft PR #49877 fixes this issue for you? If it does, I'll make it a proper PR.

Hi, I have checked your draft PR, and it can't fix this issue. And I have also tested 500 times, it would fail 6 times.

edersondisouza, 2022-09-06:
Oh, that's a bummer. Just to clarify, did it at least make it less likely? On the original report, it is mentioned 1 in 10, but more likely 1 in 100 with my draft PR. So, did it become less likely, or no influence at all?

shuai1x, 2022-09-07:
> Oh, that's a bummer. Just to clarify, did it at least make it less likely? On the original report, it is mentioned 1 in 10, but more likely 1 in 100 with my draft PR. So, did it become less likely, or no influence at all?

I think I need to test it again to compare the results of these two tests. And will reply to you soon.

shuai1x, 2022-09-08:
> Oh, that's a bummer. Just to clarify, did it at least make it less likely? On the original report, it is mentioned 1 in 10, but more likely 1 in 100 with my draft PR. So, did it become less likely, or no influence at all?

I did the test again. And I wonder whether the failure is random. Because before I tested it, it would fail 1 in 10, but this test I ran 100 times, and failed 27, which seems 3 in 10. And after using your draft PR, it would pass 5 in 100. So it seems your PR make it become less.

edersondisouza, 2022-09-08:
Thanks for the input, I'm finalizing a patch here that should truly solve the issue, stay tuned =D

edersondisouza, 2022-09-08:
Ok, so I updated my draft PR - I think it solves this issue. Even if it doesn't, it still fix some code issues. @shuai1x could you check, please?

shuai1x, 2022-09-08:
> Ok, so I updated my draft PR - I think it solves this issue. Even if it doesn't, it still fix some code issues. @shuai1x could you check it, please?

Ok, will reply to you soon.

commit 578e8ee9 (2022-09-08): soc/xtensa/intel_adsp/cavs: Fix CPU sporadically not idling

Sometimes, on CAVS 25, the CPU may fail to idle. This was due a failure
during test initialisation - the test is supposed to run from CPU0, and
halt and restart the other ones. However, code to pin thread to CPU was
incorrect - a running thread can't be pinned to a CPU (the
k_thread_cpu_mask* calls were returning error).

This patch solves this by creating a thread that is pinned to CPU0
before it is run.

While at it, add some more checks for returned values.

Fixes: #49790
closed 2022-09-08 by nashif