zephyrproject-rtos/zephyr #49811: DHCP cannot obtain IP, when CONFIG_NET_VLAN is enabled

maweicoltsmart, 2022-09-01:
branch：main

DHCP cannot obtain IP, when CONFIG_NET_VLAN is enabled

The log output is as follows:
```
[00:00:35.201,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   ciaddr=0.0.0.0
[00:00:35.201,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   yiaddr=192.168.31.198
[00:00:35.201,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   siaddr=192.168.31.1
[00:00:35.201,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   giaddr=0.0.0.0]
[00:00:35.201,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]): Unexpected op (2), xid (0 vs 46078717) or chaddr
```

maweicoltsmart, 2022-09-01:
If I modify the program to ignore this error, I still cannot get the IP, and the log output is as follows:
```
[00:00:34.133,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]): Received dhcp msg [op=0x2 htype=0x1 hlen=6 xid=0x45b48a44 secs=0 flags=0x80 chaddr=00:04:A3:00:00:01
[00:00:34.133,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   ciaddr=0.0.0.0
[00:00:34.133,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   yiaddr=192.168.31.198
[00:00:34.133,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   siaddr=192.168.31.1
[00:00:34.133,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   giaddr=0.0.0.0]
[00:00:34.133,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_server_id: 192.168.31.1
[00:00:34.133,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_lease_time: 43200
[00:00:34.133,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_renewal: 21600
[00:00:34.133,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_rebinding: 37800
[00:00:34.133,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_subnet_mask 255.255.255.0
[00:00:34.133,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 28
[00:00:34.134,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_router: 192.168.31.1
[00:00:34.134,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 43
[00:00:34.134,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 12
[00:00:34.134,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_end
[00:00:34.134,000] <dbg> net_dhcpv4: dhcpv4_handle_reply: (rx_q[0]): state=disabled msg=offer
```

jukkar, 2022-09-06:
At least the IP addresses and other values seem to have reasonable and non-random values, so I would assume that VLAN header is properly discarded by the Ethernet L2. Your PR title indicates that there is no error if VLAN is disabled, is that so?
Have you checked whether the DHCP debug output matches the actual data being received, i.e., are you able to capture the data in Wireshark for example?

rlubos, 2022-09-07:
@maweicoltsmart Can you provide more details, what platform do you use, what's your application, what network setup do you use? I've tried to use reproduce with `samples/net/vlan` running on `native_posix` (with DHCPv4 enabled) and I've seen no issues at all, my device obtained IPv4 address w/o issues.

maweicoltsmart, 2022-09-12:
@rlubos @jukkar Sorry for not replying in time due to busy work. The above problem occurs because I enabled `CONFIG_NET_VLAN` but did not call the function` int net_eth_vlan_enable(struct net_if *iface, uint16_t tag)`. In order to solve this problem, I added a patch to the Ethernet driver: when receiving an Ethernet packet, take out the vlan tag in the header and call the function` int net_eth_vlan_enable(struct net_if *iface, uint16_t tag)`, so that IP can be obtained through DHCP.
I guess I don't need to do the above patch in my program if the existing Ethernet program does the following special processing: if `CONFIG_NET_VLAN` is enabled but the function` int net_eth_vlan_enable(struct net_if *iface, uint16_t tag) `is not called, ignore the vlan tag in the received packet, and the outgoing Ethernet packet header does not contain the vlan tag.

rlubos, 2022-09-12:
@maweicoltsmart Which driver exactly are you referring to? There is a `net_eth_is_vlan_enabled()` function that verifies if VLAN has been enabled on an interface, and the function is used in the Ethernet LL to determine whether to fill/strip the VLAN header.

> if CONFIG_NET_VLAN is enabled but the functionint net_eth_vlan_enable(struct net_if *iface, uint16_t tag)is not called, ignore the vlan tag in the received packet

Shouldn't the packet be dropped though in such case? If the interface is not a part of a particular VLAN "subnet", it shouldn't be processing packets distributed within the VLAN I think?

> and the outgoing Ethernet packet header does not contain the vlan tag.

That should be the case already, the Ethernet LL will not inlcude the VLAN field in the header if the VLAN is not enabled on an interface.

maweicoltsmart, 2022-09-13:
@rlubos In most cases, the device does not set the vlan tag in advance, and does not provide the interface for setting the vlan tag, but the switch will send the data packet with the vlan tag to the device. In this case, the device will not be able to access the set vlan tag switch.

carlescufi, 2022-09-21:
@maweicoltsmart please provide the information requested by @rlubos, including which platform and driver you are using

maweicoltsmart, 2022-09-25:
I found a new problem. When `CONFIG_NET_UDP_CHECKSUM=n`, the IP can be obtained on a Huawei switch, but when `CONFIG_NET_UDP_CHECKSUM=y`, the IP cannot be obtained. From the output log, it can be known that the DHCP offer packet sent by the switch is discarded by the protocol stack. Forgive me I didn't save the raw data but just took pictures, one picture is the wires hark and the other is the log output by the device. The original packet and checksum can be seen from the photo. The photos are as follows:

[image]
[image]

maweicoltsmart, 2022-09-26:
@rlubos 
Should be the same issue as issue #50614

rlubos, 2022-09-27:
@maweicoltsmart Please post wireshark logs next time, it's not really helpful when you need to parse the packet manually based on the blurry photo...

Anyway, it seems that the UDP checksum field is not set in the received packet (checksum missing), such packets are by default dropped by Zephyr. In order to allow for such packet processing, you need to enable `CONFIG_NET_UDP_MISSING_CHECKSUM=y` explicitly.

carlescufi, 2022-10-05:
Closing since there is no clarity over whether there is an actual issue here at all. Please reopen if you have logs and a reproducible example that we can use to debug it.

closed 2022-10-05 by carlescufi