zephyrproject-rtos/zephyr #49816: ISOTP receive fails for multiple binds with same CAN ID but different extended ID

benediktibk, 2022-09-01:
**Describe the bug**
If you call isotp_bind twice, both with the same receiving CAN ID (isotp_msg_id.std_id) but different ISOTP extended IDs (isotp_msg_id.ext_addr), it is impossible to receive something on the second ISOTP extended ID.

This is caused by the filter mechanism of CAN. can_attach_isr is getting called for the specific CAN ID, with ctx (struct isotp_recv_ctx) as argument. This then later leads to the receive_can_rx_isr getting called always with the ctx of the first filter. Somewhere later than the extended ID of ISOTP is getting checked (process_ff_sf), which will always fail for a packet of the second call to isotp_bind.

I found this issue in commit 3f826560aaf81a444018293bd6acce3c339fe150 (release Zephyr 2.7.0), but couldn't find a fix in the history until now.

**To Reproduce**
Steps to reproduce the behavior:
1. call isotp_bind twice with same CAN ID but different extended ISOTP ID
2. receive packets on the second extended ISOTP ID -> will get dropped in process_ff_sf

**Expected behavior**
It should be possible to use the same CAN ID for different extended ISOTP IDs for receiving packets.

**Impact**
It is impossible to use ISOTP extended IDs on the same CAN ID for receiving.

martinjaeger, 2022-09-05:
@benediktibk I got a bit confused by the term "ISOTP ID". But I guess you meant the ISO-TP target address (called `N_TA` in the standard) which can be either part of the CAN ID or the payload. I edited your comment accordingly. Let me know if I got anything wrong.

Regarding the issue itself: At first sight it should be possible to assign two different callbacks for the same CAN ID in the CAN drivers, but I have never tested it. Do you think this is a bug in the ISO-TP implementation or in the underlying CAN driver?

Which hardware (i.e. CAN driver) did you use?

benediktibk, 2022-09-05:
Sorry for the confusion, but I think N_TA is what I am talking about. Basically byte 2 in:
[image]

I found this issue on a SAMC21. As it looks like it is possible to register to different callbacks for the same CAN ID, but only the first one is actually being used later on.

henrikbrixandersen, 2022-09-05:
> I found this issue on a SAMC21. As it looks like it is possible to register to different callbacks for the same CAN ID, but only the first one is actually being used later on.

This is a well-known hardware limitation for many CAN controllers. It is somewhat documented in the CAN API documentation, but the documentation could be improved a bit: https://github.com/zephyrproject-rtos/zephyr/blob/64ab22bf43f937443121b414ddd7997072fe555a/include/zephyr/drivers/can.h#L1046: "If a frame matches more than one attached filter, the priority of the match is hardware dependent."

martinjaeger, 2022-09-05:
> Less work and also more clear than the current state would it be to let it fail for the second call of iso_tp_bind?

That would be better than silently accepting a non-working configuration. But is it sufficient for your use case? If yes, do you think you could submit a PR?

benediktibk, 2022-09-05:
Unfortunately my setup is bound to use zephyr 2.7.0, I cannot upgrade to a more recent version. Also, I do not have hardware like a development kit for this MCU available to test my changes.

For us such a solution would actually be sufficient, as we just stopped using the extended IDs altogether due to this issue. Luckily, not a big deal in our situation.

closed 2023-11-25 by github-actions[bot]