zephyrproject-rtos/zephyr #49821: USB DFU with MCUBOOT fails on Windows

Light411, 2022-09-01:
**Describe the bug**
When uploading a signed image using dfu-util on windows, the transfers fails and output "Lost device after RESET ?". 

From what I understood I'm not the only one facing this issue, It has been discussed on Discord both on the general channel but also on the MCUBOOT one. 

The SoC I'm using is stm32g474re, but it seems to also affect other vendors. 
USB DFU works flawlessly on Linux I'm only experiencing the issue on Windows.

What I tried :

- I made sure that I'm not enabling the USB_COMPOSITE that was known to cause trouble on Windows. #23337
- I made sure that POLLTIMEOUT is at it's default value as it seems to be linked to similar issues #15497
- I tried creating a USB device on Zadig to install WinUSB not only on the runtime, but also on the transfer HWID.

**To Reproduce**

Prj.conf relevant configuration is 

``` 
# USB
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="MCUBOOT"
CONFIG_USB_COMPOSITE_DEVICE=n
CONFIG_USB_CDC_ACM=n
CONFIG_USB_DEVICE_PID=0x0101
CONFIG_USB_DEVICE_DFU_PID=0x0102

#Upgrade with swapping the memory partition without the need of scratch partition
CONFIG_BOOT_SWAP_USING_MOVE=y

# Enable USB Download Firmware Upgrade (DFU)
CONFIG_BOOT_USB_DFU_GPIO=y
```

I used Zadig v2.7 to load WinUSB v6.1 driver on MCUBOOT run-time dfu (2fe3:0101). 

**Expected behavior**
Dfu-util is supposed to be able to reset the USB from run-time to transfer mode. 

**Impact**
The project we are working on aims at having end users able to download binaries to the board using dfu-util. Most users are using Windows and this issue is a showstopper for us.

**Logs and console output**

When listing the DFU capable USB device, we already see that it fails at enumerating correctly.

```
C:\Users\jal\Downloads\dfu-util-0.9-win64\dfu-util-0.9-win64>dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Cannot open DFU device 04f2:b6c6
Found Runtime: [2fe3:0101] ver=0301, devnum=33, cfg=1, intf=0, path="1-5", alt=0, name="UNKNOWN", serial="5056500D00310056"x
```

When flashing it gives 
```
C:\Users\jal\Downloads\dfu-util-0.9-win64\dfu-util-0.9-win64>dfu-util -v -a 1 -D firmware.signed.bin 
dfu-util 0.9                                                                                                                                                                                                                                                                                                                                            Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
Invalid DFU suffix signature 
A valid DFU suffix will be required in a future dfu-util release!!!
Cannot open DFU device 04f2:b6c6
Opening DFU capable USB device...
[... 84 lines cut ...]
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_get_config_descriptor] index 0
libusb: debug [parse_configuration] skipping descriptor 0xb
libusb: debug [parse_endpoint] skipping descriptor 25
libusb: debug [parse_endpoint] skipping descriptor b
libusb: debug [parse_endpoint] skipping descriptor 25
libusb: debug [parse_endpoint] skipping descriptor b
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_get_config_descriptor] index 0
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_get_config_descriptor] index 0
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_get_config_descriptor] index 0
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_get_config_descriptor] index 0
Lost device after RESET?   
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain Zephyr SDK
 - Commit SHA or Version used Zephyr 3.1

**Additional context**
Not much more to add AFAIK

nordicjm, 2022-09-01:
Issue also observed on nrf52833-based module and stm32f103-based module. Some old discussions on github mentioning this issue from 2+ years ago

jfischer-no, 2022-09-07:
> Issue also observed on nrf52833-based module and stm32f103-based module.

You can reproduce it on nRF52833DK?

jfischer-no, 2022-09-07:
@Light411 Since you opened it here, can you reproduce the issue using Zephyr's sample `samples/subsys/usb/dfu`?

nordicjm, 2022-09-08:
> > Issue also observed on nrf52833-based module and stm32f103-based module.
> 
> You can reproduce it on nRF52833DK?

BL653 DVK which is the same hardware/interface wise as nRF52833dk. After having more of a play with it, it seems to be the winusb driver, I tried installing the libusbK drivers using zadig instead, and after installing the libusbK package with system drivers, suddenly dfu-util can communicate with the device on windows. So it's a far cry faff from the ability to do a mcuboot serial recovery update with just a single executable and no drivers to needing the libusbK driver, libusbK system drivers and an executable for this. Perhaps this should be added to the documentation.

jfischer-no, 2022-09-08:
> After having more of a play with it, it seems to be the winusb driver, I tried installing the libusbK drivers using zadig instead, and after installing the libusbK package with system drivers, [...]

I agree that "mcuboot serial recovery update" is much easier to use for Windows OS users. I have no measurements regarding throughput but remember that USB DFU was faster.
I do not get to test it until week 38.

Light411, 2022-09-08:
I tried using libusbK this afternoon, and I managed to download the image sucessfully on Windows ! 

IMO the best fix here would be to include the windows descriptor in ucb_dfu.c file so that Windows natively load libusbK driver for the given run-time and dfu mode VID:PID. 

I've seen that windows descriptors are already in tree. I tried implementing them to try my luck implementing a WCID feature for my device but I was not able to make it work. 
More specifically, I was able to populate my descriptors, but I wasn't able to understand how to handle these descriptors to the usb enabling function.
Do you think it could work ?

jfischer-no, 2022-09-08:
> I tried using libusbK this afternoon, and I managed to download the image sucessfully on Windows !
>
> IMO the best fix here would be to include the windows descriptor in ucb_dfu.c file so that Windows [...]

If it is about Microsoft OS * Descriptor, please take a look at samples/subsys/usb/webusb.

Light411, 2022-09-09:
I found two implementation of MS OS descriptors available in Zephyr
`CONFIG_USB_DEVICE_OS_DESC` which from my understanding is only capable of MS OS descriptor V1 and is focused on WCID feature. 
`CONFIG_USB_DEVICE_BOS` which from my understanding is capable of the extended features offered by MS OS descriptor V2 that I indeed found in the very useful webusb sample.

I was thinking that the first one was more suitable for the task as usb_dfu.c only need to provide the descriptors needed for the WCID feature. Also from what I read, WCID has been deployed in 2012 while BOS v2 is more recent and would let older versions of windows aside of the support. 

At the moment `CONFIG_USB_DEVICE_OS_DESC` seems to be used only by `subsys/usb/device/class/netusb` and by a test `tests/subsys/usb/os_desc`. 
I tried to implement it to check the feasibility but I didn't manage, I was not able to pass the descriptor to my usb_enable() function.

My main question here is : Would this fix the problem for windows users ? 
Second one : Is LibusbK supported by WCID ?
Last one : How do vendor bootloaders and dfu compatible boards / device supports Windows in a plug and play fashion ? 
Using WCID + WinUsb or any other driver ? Or implementing custom drivers and linking them to their VID:PID ? 

PS I found this great documentation on WCID: https://github.com/pbatard/libwdi/wiki/WCID-Devices

desowin, 2022-09-14:
> I tried using libusbK this afternoon, and I managed to download the image sucessfully on Windows !

This is expected due to the actual root cause of the issue. WinUSB does not support host initiated USB bus resets, while libusbK.

> IMO the best fix here would be to include the windows descriptor in ucb_dfu.c file so that Windows natively load libusbK driver for the given run-time and dfu mode VID:PID.

I think adjusting Zephyr for WinUSB limitations is better. USB Device Firmware Upgrade Specification, Revision 1.1 specifies:
`Bit 3: device will perform a bus detach-attach sequence when it receives a DFU_DETACH request. The host must not issue a USB Reset. (bitWillDetach) 0 = no 1 = yes`

Zephyr sets `bitWillDetach` to 0 and hence DFU does not work with WinUSB, but does with libusbK. To have WinUSB compatible DFU implementation, device have to set `bitWillDetach` to 1 and perform the bus detach-attach sequence.

> Second one : Is LibusbK supported by WCID ?

It can be supported if user (application explicitly run by user) installs the libusbK drivers before the device is plugged in.

> Last one : How do vendor bootloaders and dfu compatible boards / device supports Windows in a plug and play fashion ?
> Using WCID + WinUsb or any other driver ? Or implementing custom drivers and linking them to their VID:PID ?

A lot of different approaches. Some use custom drivers, some use even custom vendor defined protocols. In my opinion using WCID + WinUSB together with USB DFU class is the most cost effective solution (i.e. least amount of effort required both in development and maintenance).

jfischer-no, 2022-09-20:
> Regarding the WinUSB limitation it is pretty hard to come up with a link to something that does not exist. The [winusb.h header documentation](https://learn.microsoft.com/en-us/windows/win32/api/winusb/) lists the API provided by WinUSB.

Well it would be easier if they just write that there is no API for it instead of placing dead links in the code. Anyway, limitations of WinUSB do not seem to be a bug in Zephyr to me.

desowin, 2022-09-21:
> Well it would be easier if they just write that there is no API for it instead of placing dead links in the code.

Nobody placed dead links in the code. It is just how internet works, the links do become dead over time. So will the links placed in Zephyr code/documentation, especially the ones that point to latest version and not some specific version. This is why besides just a link, there should be enough context for the future reader.

If the comment you are referring to is what you [linked earlier](https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L3184-L3191) then the author did all they could do to provide enough context, that is:

```
/*
 * from the "How to Use WinUSB to Communicate with a USB Device" Microsoft white paper
 * (http://www.microsoft.com/whdc/connect/usb/winusb_howto.mspx):
 * "WinUSB does not support host-initiated reset port and cycle port operations" and
 * IOCTL_INTERNAL_USB_CYCLE_PORT is only available in kernel mode and the
 * IOCTL_USB_HUB_CYCLE_PORT ioctl was removed from Vista => the best we can do is
 * cycle the pipes (and even then, the control pipe can not be reset using WinUSB)
 */
```

The "WinUSB does not support host-initiated reset port and cycle port operations" is basically saying there is no API for it. It also provides really good context for people familiar with Windows development that there is basically no hope to implement host initiated reset in user-space.

> Anyway, limitations of WinUSB do not seem to be a bug in Zephyr to me.

Then consider WinUSB compatibility as a feature. It is just a taxonomy. The thing is, that the end-user quite often does not make the distinction and instead reports that something does not work and considers this to be a bug. It is easy to notice that something does not work as expected, but it is hard to tell what is the root cause (because otherwise the reporter would just fix the root cause in the first place instead of going through the trouble of reporting).

The average user that buys any peripheral that runs Zephyr and uses USB DFU is likely to use Windows to update the firmware, as Windows currently has about 74% desktop market share. The end user expects the manufacturer to provide working software package that can update the firmware. The lowest maintenance solution is USB DFU with WinUSB WCID descriptors, as it eliminates the driver development, signing and installation problem completely.

jfischer-no, 2022-09-21:
> WCID and this fix will provide the ergonomics expected and needed for most.

Again, it is not a bug in Zephyr and therefore not a fix and has to wait until release is finished.

commit c27d48c8 (2022-10-06): usb: dfu: Support DFU with WinUSB on Windows

DFU on Windows with WinUSB driver failed with "Lost device after RESET?"
error because WinUSB does not support host initiated resets. USB Device
Firmware Upgrade Specification Revision 1.1 attribute bitWillDetach can
be used to overcome WinUSB reset limitation. When bitWillDetach is set,
it is the device responsibility to detach and reattach itself to the bus
after receiving DFU_DETACH request.

Add and enable by default USB_DFU_WILL_DETACH configuration option,
because it is the only way to support WinUSB driver. WinUSB driver is
preferable becaus [... cut]
closed 2022-10-06 by fabiobaltieri