zephyrproject-rtos/zephyr #49823: Example Application: Use of undocumented zephyr/module.yml in application folder

SkryptX, 2022-09-01:
In the current example application (https://github.com/zephyrproject-rtos/example-application) there is a `zephyr` folder with a `module.yml` file, which enables the use of the device trees, Kconfigs etc. just like in a module.

I cannot find any mention of this mechanism anywhere in the documentation. It might very well be that I haven't fully understood west and its magic, but I really can't find where it says that this is allowed and working inside an application folder. Or if not specific to the application folder, how it is setup in general.

carlescufi, 2022-09-01:
Here you go:
https://docs.zephyrproject.org/latest/develop/modules.html#integrate-modules-in-zephyr-build-system

closed 2022-09-01 by carlescufi

SkryptX, 2022-09-01:
I've read this section multiple times and it does not tell me how the module discovery process works. There is this section https://docs.zephyrproject.org/latest/develop/modules.html#module-inclusion which explains it somewhat, but not completely.

My questions is: How do modules get found? Is it with the `west.yml` and if so, how does the manifest repository of the example-application get added to the list? Is it per directory and if so, why doesn't it recognize my dummy module which satisfies the conditions in the link above and how does it find the modules in the `modules` folder? I'm really at a loss to how it exactly works. I'm very certain I would find _a_ way to do it, but I want to know all the ways and then select the best one depending on my use case.

In essence I'm asking how `ZEPHYR_MODULES` gets constructed and where in the documentation I can find this.

carlescufi, 2022-09-01:
@SkryptX you seem to have come to the right conclusions yourself. The module discovery logic, including the manifest repository as a module (eg. example-application) is located here:

https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/zephyr_module.cmake

which in turn invokes this script here:

https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/zephyr_module.py

and that script uses the west APIs (eg. `Manifest.get_projects()` and others) to list all west projects and look for a `zephyr/module.yml` in them.

carlescufi, 2022-09-01:
@SkryptX we'd be very grateful if you could send a Pull Request with the missing documentation bits, we always welcome contributions! cc @tejlmand.