zephyrproject-rtos/zephyr #49845: net: Disable data path prints

krish2718, 2022-09-01:
Currently, networking layer has loads of per-packet prints, e.g., see below, having prints in data path is not a good idea, we already support statistics, so, that can be used for any errors/drops. These should be removed or converted to debug.

```
 <err> net_arp: Gateway not set for iface 0x20001a60
  ....
 <wrn> pkt cloning failed, pkt %p dropped
   ....
```

@rlubos

rlubos, 2022-09-02:
I'm not really convinced about lowering packet allocation errors log priority. Packet allocation failures usually indicate problems either with system configuration or some net stack component (not freeing the packets in time). Note that those reports are usually preceded with a considerable timeout during the packet allocation, where the corresponding thread is suspended, so the performance impact of the logger should be negligible in such cases (especially with deferred logging).

@jukkar What's your view on the topic?

krish2718, 2022-09-02:
Not only performance impact, but also the usability of the prints, wouldn't it be better if we report stats? The print in itself doesn't help debug, just shows that there is some problem right, so, stats might be good enough.

krish2718, 2022-09-02:
ok, agreed on packet allocation failures, thought we could separate debug stats from mandatory stats (like `/proc/net/snmp` in Linux). 

Also, what about other prints? I only mentioned a few I saw, but looking at source there are many. e.g., `gateway not set`, frag related prints etc.

rlubos, 2022-09-02:
Well, I can only refer here to `<err> net_arp: Gateway not set for iface 0x20001a60`, as along with the packet allocation failure is the only one mentioned here, but I would consider this as useful information. It informs that ARP procedure for a packet outside of the local network failed because there is no gateway information. 

As for others, I guess we need to analyze each case individually, I'm not saying that it's not possible that there is some log that pollutes the output in certain conditions. I just don't see the two mentioned here as irrelevant.

krish2718, 2022-09-02:
Understood, I was just giving some examples. 

I am still not convinced using per-packet prints (thought depending on log buffer size they might suppressed) as a way to get users attention, esp. as they can't do anything about it (end-customers). And for developers, if they see low performance, then using stats to see drops, alloc failures is preferred, esp. as it gives the counters, so, we can see % of failures. But I do see your point, so, will leave it to you.

jukkar, 2022-09-02:
Setting allocation errors to <err> level sounds like a good plan, after all the system is possibly in bad shape after those, especially if the recovery from out-of-mem is not possible or is error prone.

krish2718, 2024-02-08:
> @krish2718 you are also encouraged to help moving this issue forward by providing additional information and confirming this request/issue is still relevant to you.

I still feel my comment https://github.com/zephyrproject-rtos/zephyr/issues/49845#issuecomment-1235457441 is relevant, but I will leave it to the maintainers to take a call on this.

rlubos, 2024-02-09:
@jukkar I think we need to decide whether we want to keep it open (and comply with the request to drop the logs), or close the issue.

As for me, I vote for keeping the logs as they are (i.e. wrn/err level logs on actual errors, even if there considered at per-packet basis).

jukkar, 2024-02-09:
Yes, I agree, at least those two messages described in the beginning of the issue are useful.

krish2718, 2025-07-12:
Re-opening the issue, as with this commit https://github.com/zephyrproject-rtos/zephyr/commit/4e0386a44578f08f5ea3895c62187ecfd5ce645b we now see a flood of prints when running UDP TX Zperf `zperf udp upload -a 192.168.1.253 5001 3600 1K 50M`, these failures are expected as driver is unable to handle the incoming traffic and is dropping, but user doesn't even have control over shell till Zperf completes.

```
art:~$ [00:03:22.000,274] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.000,427] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.000,610] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ --- 26 messages dropped ---
[00:03:22.001,007] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ --- 10 messages dropped ---
[00:03:22.001,159] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ --- 6 messages dropped ---
[00:03:22.001,434] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ --- 5 messages dropped ---
[00:03:22.001,586] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.001,739] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.002,227] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.009,857] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.010,223] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.010,375] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.010,650] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.010,803] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.010,955] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.030,426] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.035,675] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.037,078] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.038,513] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.038,696] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.038,970] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ --- 10 messages dropped ---
[00:03:22.039,154] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.039,337] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ --- 4 messages dropped ---
[00:03:22.039,611] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ --- 1 messages dropped ---
[00:03:22.042,846] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ --- 5 messages dropped ---
[00:03:22.043,029] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.043,182] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.043,365] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.066,619] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.066,772] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.067,047] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.067,199] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.067,382] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.210,021] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.210,174] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.211,395] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.211,853] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.236,267] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.236,419] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.236,694] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.236,846] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.237,121] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.237,396] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.237,579] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.237,731] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.237,915] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.238,403] <wrn> net_if: iface 1 pkt 0x200673c0 send failure status -1
uart:~$ [00:03:22.261,932] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.262,207] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.262,390] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ [00:03:22.262,542] <wrn> net_if: iface 1 pkt 0x20067378 send failure status -1
uart:~$ --- 18 messages dropped ---
```
Re-iterating my take above and also in https://github.com/zephyrproject-rtos/zephyr/issues/80167#issuecomment-2549505968 we shouldn't have per-packet print in data-path which is of no use to end-user, statistics are the way to debug packet drops. But if something has to be brought to attention, then we can add something similar to `printk_once` in Linux kernel to notify of the packet drops to trigger the investigation.

krish2718, 2025-09-29:
Yes, we can close this, thanks.

closed 2025-09-29 by krish2718