zephyrproject-rtos/zephyr #49851: Bluetooth Controller with Extended Advertising

LingaoM, 2022-09-02:
Hi,

i'm use Zephyr Bluetooth Controller to develop bluetooth extended advertsing application, but found the second pdu offset too big than  
ncs sdk.

For example, ncs sdk aux offset almost 400us, but the current zephyr sdk bigger than 10ms

[image]

LingaoM, 2022-09-02:
[ncs_ext_adv_demo.zip](https://github.com/zephyrproject-rtos/zephyr/files/9476330/ncs_ext_adv_demo.zip)

cvinayak, 2022-09-02:
In the Zephyr Controller design, the scheduling of AUX_ADV_IND PDUs are different to ensure that they can be placed non-overlapping with other states and roles, and the ADV_EXT_IND PDUs would be scheduled with the required random_delay.