zephyrproject-rtos/zephyr #49852: uart: extra XOFF byte in the read buffer

ric96, 2022-09-02:
**Describe the bug**
I am reading 256 byte long uart rx into a uint8_t array of size 256 and have flow control set to none. I can see on my logic analyzer that the tx device is not sending XOFF.
However, my first read element as position 0 in the array is always XOFF (decimal 19) and every other element is shifted by one, so i end up having 257 elements instead.
This is on an stm32F401cc.
as a workaround I can ignore the first received byte and start saving from the second byte and it works fine but its a weird workaround. 

I'm setting all the config through `uart_configure()` at runtime. And my dt is bare minimum:
```
&usart2 {
	pinctrl-0 = <&usart2_tx_pa2 &usart2_rx_pa3 &usart2_cts_pa0 &usart2_rts_pa1>;
	pinctrl-names = "default";
	status = "okay";
	current-speed = <115200>;
};
```

This is what my config read looks like, so i can confirm flow control is set to off:
```
ret = uart_config_get(console, &console_config);
        printk("CONSOLE CONFIG:\nBAUD: %d\nDATA BITS: %u\nFLOW CTRL %u\nPARITY %u\nSTOP BITS %u\n", 
console_config.baudrate, console_config.data_bits, console_config.flow_ctrl, console_config.parity, console_config.stop_bits);

## OUTPUT

CONSOLE CONFIG:
BAUD: 115200
DATA BITS: 3
FLOW CTRL 0
PARITY 0
STOP BITS 1
```

I am reading uart using interrupt, very similar to the echo sample, here you can see the workaround by starting the counter at -1.
```
void serial_cb(const struct device *dev, void *user_data)
{
    uint8_t c;

    if (!uart_irq_update(dutconsole)) {
        return;
    }

    while (uart_irq_rx_ready(dutconsole)) {

        uart_fifo_read(dutconsole, &c, 1);
        if(rx_buf_pos > -1)
        {
            recv_dat_8bit[rx_buf_pos] = c;
        }
        rx_buf_pos++;
        
    }
}
```

i also tried a sending just single byte at one time and even then the rx buffer has 2 bytes, first one being XOFF.
Now I'm wondering if this is an expected behavior?

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - stm32F401cc

**Expected behavior**
The rx buffer should only contain revived bytes in a NO Flow control config.

**Environment (please complete the following information):**
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used
     -  Zephyr: f1bafda5e594cc13a4efa1fbc44057104f89f8b6
     - STM32 HAL: 5b1610c3cf46589d50e2634ba240b7b38bc59c53

ric96, 2022-09-02:
Hi @erwango 
I had some time to do more debugging, TL:DR its not a Zephyr or STM32 HAL Bug. So i'm closing this now with this note.

I took a longer capture of on the logic analyser and saw that the linux board is the one sending the phantom XOFF. I am using libmraa, and its sending this byte right after I set the flow-control to OFF which is very early in the code.
Apologies for the noise.

closed 2022-09-02 by ric96

erwango, 2022-09-05:
> Apologies for the noise.

Np, Thanks for the heads up