zephyrproject-rtos/zephyr #49874: STM32G HW_STACK_PROTECTION Warning

photonthunder, 2022-09-02:
When compiling STM32G code I see the following warning:

warning: HW_STACK_PROTECTION (defined at arch/Kconfig:249) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: ARCH_HAS_STACK_PROTECTION (=n).

Asked on Discord and got the following:

"I remember seeing the same thing on stm32f1, I think it's just that the stm32 hal is not configured right
Since it's kind of in-built on Cortex M..."

Any insight?

erwango, 2022-09-14:
@photonthunder Can you at least provide an example command (build which board on which sample ?) that generates the following warning ?
I won't try to guess

photonthunder, 2022-09-14:
I saw it when I used board stm32g071b_disco and then add CONFIG_HW_STACK_PROTECTION=y to the hello world sample, then you get the warning...

closed 2022-09-14 by erwango