zephyrproject-rtos/zephyr #49889: ctf trace: unknown event id when parsing samples/tracing result on reel board

smrtos, 2022-09-04:
**Describe the bug**
Try to capture ctf trace on reel_board with below steps:

Preparation:

0. As mentioned [here](https://github.com/zephyrproject-rtos/zephyr/issues/24457#issuecomment-617486501), to make sure only trace data go through the UART, I comment out all the `printk()` in [main.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/tracing/src/main.c).

Build the sample app:

```
1. cmake -GNinja -DBOARD=reel_board -DCONF_FILE=prj_uart_ctf.conf -B bld_ctf_reel -S samples/subsys/tracing/
2. cd bld_ctf_reel
3. ninja flash
```

Then capture the trace from UART with script from [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/tracing/trace_capture_uart.py):

`4.  python ../scripts/tracing/trace_capture_uart.py -d /dev/ttyACM0 -b 115200 -o ctf_data/cap1.ctf`

To avoid the boot banner message, do another capture 

`5.  python ../scripts/tracing/trace_capture_uart.py -d /dev/ttyACM0 -b 115200 -o ctf_data/cap2.ctf`

Put the capture file and CTF metadata in one folder and parse it with `babeltrace`:

```
6. mkdir ctf_data
7. cp ../subsys/tracing/ctf/tsdl/metadata ./ctf_data/
8. mv cap2.ctf ./ctf_data/
9. babeltrace ctf_data
```

Below error shows up:

```
[error] Event id 32 is unknown.
[error] Reading event failed.
Error printing trace.

```

I also tried use `minicom` to capture the UART output. And saw similar errors.

I also found issue https://github.com/zephyrproject-rtos/zephyr/issues/24457, which seems to be the same as what I did.

smrtos, 2022-09-04:
Attached is the trace data.
[ctf_data_reel.zip](https://github.com/zephyrproject-rtos/zephyr/files/9484430/ctf_data_reel.zip)

nashif, 2022-09-06:
are you sure you have the latest `../subsys/tracing/ctf/tsdl/metadata` file? ID 32 is defined in the latest file. Please double check.

nashif, 2022-09-08:
ping

nashif, 2022-09-15:
My suspecion here is that the data capture is not complete, they way you are making the changes is not ideal, tracing needs to have a dedicated backend. See how this work with qemu which emits the trace to a file under build/, those events you are missing show up correctly.

smrtos, 2022-09-23:
No time to try it recently. Will take a look.

smrtos, 2022-09-23:
I did checked the qemu command line in the build folder before.
But no time to further investigate it recently. Will continue the work.
Sorry for the late response.

nashif, 2022-10-13:
there was a similar report, so there might be an issue here, will continue looking...

diocorreia, 2022-12-05:
Having similar issue (but with nrf52840), @smrtos did you figure out the problem? I've already tried two releases of Zephyr  (v2.7.0 and 3.2.0) and the issue occurs in both. I've also tried to use TraceCompass, but the outcome is the same (fails when parsing the file).

closed 2023-02-19 by github-actions[bot]

ball-hayden, 2023-10-17:
Also seeing issues with invalid CTF traces while trying to open in TraceCompass.

My setup is using the nRF52840DK with the tracing sample and `prj_usb_ctf.conf`.

Please find attached an example (invalid) trace:
[trace.zip](https://github.com/zephyrproject-rtos/zephyr/files/12925768/trace.zip)

ball-hayden, 2023-10-23:
In my case, the issue was that I hadn't copied the `metadata` file to sit next to the trace.
RTFM. Sorry.