zephyrproject-rtos/zephyr #49915: Syncing with devices with per. adv. int. < ~10ms eventually causes events from BT controller stop arriving

jakkra, 2022-09-05:
**Describe the bug**
When using periodic adv. intervals <~10ms then sometimes BT Controller stops reporting any events, including sync termination, CTEs, scan events etc.
 
 - What target platform are you using? nRF52833, Zephyr 
 - Is this a regression? Not checked.

**To Reproduce**
Have 10 tags sending per. adv. with 7.5ms interval with adv. train of 5 and CTE enabled. 
I have not found replicatable way to reproduce, sometimes it's enough to just let all 10 tags sync and then wait a minute and it happens. Sometimes it's easier to reproduce by randomly resetting some tags. I have reproduced without CTE sampling enabled, however may or may not be easier to reproduce with it on. I have also reproduced with only 5-6 tags but I think it's easier with more.

**Expected behavior**
BT controller keeps reporting events.

**Impact**
BT Controller stops generating events, application will be in strange state.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-4244-g1995d349db37  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
Scan callbacks register...success.
Periodic Advertising callbacks register...success.
Start scanning...success
[00:00:00.004,638] <inf> main: Creating Sync for tags in list
Waiting for periodic advertising...
Scan is running...
[00:00:00.328,277] <wrn> main: Interval: 6
[00:00:00.328,521] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.470,764] <wrn> main: Interval: 6
[00:00:00.470,977] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.498,992] <wrn> main: Interval: 6
[00:00:00.499,206] <dbg> main: scan_recv: Added tag from per adv list
Scan is running...
[00:00:00.709,167] <wrn> main: Interval: 6
(709) PER_ADV_SYNC[0]: [DEVICE]: 36:DC:42:90:4D:0E (random) synced, Interval 0x0006 (7 ms), PHY LE 2M
[00:00:00.710,174] <dbg> main: sync_cb: Removed tag from per adv list
[00:00:00.710,174] <inf> main: Creating Sync for tags in list
Enable receiving of CTE...
(710) success. CTE receive enabled.
[00:00:00.741,027] <wrn> main: Interval: 6
[00:00:00.741,241] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.762,207] <wrn> main: Interval: 6
[00:00:00.762,420] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.883,422] <wrn> main: Interval: 6
[00:00:00.883,666] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.972,381] <wrn> main: Interval: 6
(988) PER_ADV_SYNC[1]: [DEVICE]: 3B:66:BD:7F:64:67 (random) synced, Interval 0x0006 (7 ms), PHY LE 2M
[00:00:00.988,983] <dbg> main: sync_cb: Removed tag from per adv list
[00:00:00.989,013] <inf> main: Creating Sync for tags in list
Enable receiving of CTE...
(989) success. CTE receive enabled.
Running...
TAGS:sync...,sync...,33,sync...,2,sync...,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 4 Synced: 2 Num err: 0/35
[00:00:01.096,862] <wrn> main: Interval: 6
[00:00:01.097,106] <dbg> main: scan_recv: Added tag from per adv list
Scan is running...
[00:00:01.310,974] <wrn> main: Interval: 6
(1312) PER_ADV_SYNC[2]: [DEVICE]: 08:F7:34:24:2B:28 (random) synced, Interval 0x0006 (7 ms), PHY LE 2M
[00:00:01.312,622] <dbg> main: sync_cb: Removed tag from per adv list
[00:00:01.312,622] <inf> main: Creating Sync for tags in list
Enable receiving of CTE...
(1313) success. CTE receive enabled.
[00:00:01.730,895] <wrn> main: Interval: 6
Scan is running...
[00:00:01.944,915] <wrn> main: Interval: 6
[00:00:01.945,129] <dbg> main: scan_recv: Added tag from per adv list
Running...
TAGS:sync...,55,56,sync...,66,sync...,sync...,sync...,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
[... 147 lines cut ...]
Running...
----->After here no events arrive from controller, even powering of tags etc. won't generate sync term<-----
TAGS:0,0,0,0,0,0,0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 10 Num err: 0/0
Running...
TAGS:0,0,0,0,0,0,0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 10 Num err: 0/0
Running...
TAGS:0,0,0,0,0,0,0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 10 Num err: 0/0
Running...
TAGS:0,0,0,0,0,0,0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 10 Num err: 0/0
Running...
TAGS:0,0,0,0,0,0,0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 10 Num err: 0/0
Running...
TAGS:0,0,0,0,0,0,0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 10 Num err: 0/0
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr 
 - Commit SHA or Version used: 894423e9aed34a394b97c995c33549513b0ac4a5

cvinayak, 2022-09-08:
@jakkra Please check if #50022 the event stall problem?

jakkra, 2022-09-08:
@cvinayak unfortunately still samm issue.

cvinayak, 2022-09-08:
> @cvinayak unfortunately still same issue.

Ok, my fault that I used simulation and an own sample (https://github.com/zephyrproject-rtos/zephyr/pull/42721) to check whether my changes were covered by trying to sync to 5 other periodic advertisers at 7.5ms interval. I will use your sample instead and try reproducing the issue. Sorry for taking your time on the verification.

cvinayak, 2022-09-09:
@jakkra I was able to reproduce the stall using my own sample today when I had 7 advertisers synchronized. What I observed is, the system recovers after 8 minutes which corresponds to the 24-bit RTC rollover. 

I was not able to build your attached sample, it overflowed RAM.

Could you please confirm that in your application, after the stall, the system resume after ~8 mins?

jakkra, 2022-09-09:
@cvinayak I'm not in office today but will test on monday. Regarding overflow strange, but just decrease the `CONFIG_BT_PER_SYNC_MAX `to like 10 instead of 25 and `CONFIG_BT_CTRL_DF_PER_SCAN_CTE_NUM=1`

jakkra, 2022-09-12:
@cvinayak yes after about 8 minutes I get some events, see log:
```
Queued: 2 Synced: 8 Num err: 0/0
Running...
TAGS:0,0,0,0,0,sync...,0,sync...,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 2 Synced: 8 Num err: 0/0
Running...
TAGS:0,0,0,0,0,sync...,0,sync...,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 2 Synced: 8 Num err: 0/0
Running...
TAGS:0,0,0,0,0,sync...,0,sync...,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 2 Synced: 8 Num err: 0/0
----------------> 8 minutes passed around here <-------------------
--- 4 messages dropped ---
[00:08:17.301,879] <dbg> main: term_cb: Device was synced:
--- 3 messages dropped ---
(497305) PER_ADV_SYNC[5]: [DEVICE]: 0F:D9:D7:94:29:59 (random) sync terminated reason: 31
[00:08:17.306,030] <dbg> main: term_cb: Device was synced:
[00:08:17.306,243] <dbg> main: term_cb: Removed tag from per adv list
(497309) PER_ADV_SYNC[0]: [DEVICE]: 27:FD:B7:9F:15:AC (random) sync terminated reason: 31
[00:08:17.309,356] <dbg> main: term_cb: Device was synced:
[00:08:17.309,570] <dbg> main: term_cb: Removed tag from per adv list
(497311) PER_ADV_SYNC[2]: [DEVICE]: 3C:D3:E5:89:52:D1 (random) sync terminated reason: 31
[00:08:17.311,981] <dbg> main: term_cb: Device was synced:
[00:08:17.312,225] <dbg> main: term_cb: Removed tag from per adv list
(497319) PER_ADV_SYNC[7]: [DEVICE]: 18:DC:19:27:F2:37 (random) sync terminated reason: 31
[00:08:17.319,488] <dbg> main: term_cb: Device was synced:
[00:08:17.319,854] <dbg> main: term_cb: Removed tag from per adv list
(497326) PER_ADV_SYNC[8]: [DEVICE]: 11:B4:6D:C3:AA:D8 (random) sync terminated reason: 62
[00:08:17.326,995] <wrn> main: (497326)___Device not synced cancel sync tmo

[00:08:17.326,995] <wrn> main:
                               info->reason: 62

[00:08:17.327,026] <inf> main: Creating Sync for tags in list
Scan is running...
Running...
```

cvinayak, 2022-09-12:
@jakkra Thank you. From GPIO pin debugging of the scheduling, the timeouts of each periodic sync and the scanning seem to be running in the ULL context but I do not see any LLL code be invoked. I am continuing to inspect the implementation, so far it has been difficult in not getting to a fix quickly.

cvinayak, 2022-09-14:
@jakkra I believe to have nailed down the cause of the repeated skip of colliding events in #50231, please verify and revert back so that the fix can make it to the latest release versions.

cvinayak, 2022-09-15:
@jakkra @mtpr-ot 

I have concluded that an issue in the implementation of the `ticker_resolve_collision()` function is causing continuous skipping of all active radio events due to missing information to calculate the age of the radio events since their last successful use of the radio.

https://github.com/zephyrproject-rtos/zephyr/blob/b27c5d73ef09278186dc8858dfbed2a33dff6031/subsys/bluetooth/controller/ticker/ticker.c#L719-L743

@jakkra I will ask you to use `CONFIG_BT_TICKER_LOW_LAT=y` alongwith a minor fix to enable its use. You will need this PR #50252

jakkra, 2022-09-15:
@cvinayak thanks, this seems to resolve the issue, will do a bit more testing but so far so good!

cvinayak, 2022-09-15:
In `ticker_resolve_collision()`, the problem I am trying to understand is (2 tickers of 7.5ms period and 7.5 ms reservation, ticker1 to ticker2 has 5ms to start with):

1. ticker1 current age = 7.5 ms + (1 lazy x 7.5 ms) = 15 ms, force = 0; ticker2 next age = (7.5 - 5) + (1 lazy x 7.5 ms ) = 10 ms, force = 1, next_force = 1. ticker1 looses.
2. ticker2 current age = 7.5 + (1 x 7.5) = 15 ms, force = 1; ticker1 next age = (7.5 - 2.5) + (2 x 7.5 ms) = 20 ms, force = 0, next_force = 0, ticker 2 looses.
3. ticker1 current age = 7.5 ms + (2  x 7.5 ms) = 22.5 ms, force = 0; ticker2 next age = (7.5 - 5) + (2 x 7.5 ms ) = 17.5 ms, force = 1, next_force = 1.  ticker1 looses.
4. ...

am I, right? that ticker1 and ticker 2 never get to run...

cvinayak, 2022-09-16:
@jakkra Do let me know if you were able to do more testing and if any issues faced while using `CONFIG_BT_TICKER_LOW_LAT=y`?

Meanwhile, I have this PR for `CONFIG_BT_TICKER_LOW_LAT=n` #50311, do switch back and test with this PR (i.e. upstream default new implementation should give higher radio utilization, is the future implementation for new features in Controller).

jakkra, 2022-09-16:
1. `CONFIG_BT_TICKER_LOW_LAT=y` works fine.
2. With: [#50311 ](https://github.com/zephyrproject-rtos/zephyr/pull/50311) 
    -  `CONFIG_BT_TICKER_LOW_LAT=n` and `CONFIG_BT_TICKER_PRIORITY_SET=y` => Does not work same as before ("hangs")
    -  `CONFIG_BT_TICKER_LOW_LAT=n` and `CONFIG_BT_TICKER_PRIORITY_SET=n` => Works.

There is a difference between 1. and 2.
**With 1.**
25 tags synced I get an average of ~8 CTE's per second (uneven distributed over the tags, some have 5 some 20)
```
11  13  15  11  5  16  11  16  6  5  11  9  13  6  12  12  10  10  11  11  12  15  15  12  6
------------------------------------------------------------------------------------
10  11  16  12  10  14  12  15  5  4  11  12  12  5  14  12  9  10  12  13  10  13  9  13  4
------------------------------------------------------------------------------------
11  10  15  13  13  11  8  13  5  4  11  11  14  4  12  9  9  11  10  10  9  15  9  14  6
------------------------------------------------------------------------------------
10  11  12  11  10  11  8  14  6  7  8  17  9  4  12  13  10  10  11  13  11  11  12  14  4
------------------------------------------------------------------------------------
10  11  12  12  10  20  10  14  5  6  9  9  11  5  15  13  12  11  11  14  12  12  11  15  4
```
**With 2.** 
25 tags synced I get an average of ~4 CTE's per second (however most tags are at ~4 CTE's per second so even dsitribution of CTE's)

Even if in 2. the distribution is more even. I'd say 1. is still better as the minimum CTEs/second from tags are still not less than in 2.

1.

2.

cvinayak, 2022-09-16:
@jakkra Thank you and appreciate your thoroughness in verifying the fix.

Regarding the rate of CTE, I would think of it to depend on both, how much of the advertisers are overlapping over the 7.5ms interval, and how the observer listen window is scheduled. I am not  very familiar with the direction finding CTE feature, and would leave it you to decide which 1 or 2 is best for your application.

A suggestion, but you do not have to spend your immediate time to try it, to use continuous scanning say 10 ms interval 10ms window and have `CONFIG_BT_CTLR_SCAN_UNRESERVED=y`. This will provide most radio time to periodic sync roles and scanning get to use spare time between the periodic sync event.

Thank you again for contributing towards a stable Controller.

jakkra, 2022-09-19:
> @jakkra Thank you and appreciate your thoroughness in verifying the fix.
>
> Regarding the rate of CTE, I would think of it to depend on both, how much of the advertisers are overlapping over the 7.5ms [...]

Less CTE's is pretty much same thing as missed periodic adv. packets (AUX_SYNC_IND), if the radio finds the per. adv. packet, it will also sample the CTE (assuming there is RAM). For us it's a tradeoff around as many CTE's as possible, but also being able to find new tags and re-sync to tags we lost sync to.

The more AUX_SYNC_IND/CTE's the better for us. So the behaviour with `CONFIG_BT_TICKER_LOW_LAT=y` (more CTEs) is preferred (**so with 1.**)

Using CONFIG_BT_TICKER_LOW_LAT=y together with https://github.com/zephyrproject-rtos/zephyr/pull/50311 I get this crash pretty much immediately. 
`ASSERTION FAIL [(ticker_status == 0) || (ticker_status == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan_aux.c:699`

> A suggestion, but you do not have to spend your immediate time to try it, to use continuous scanning say 10 ms interval 10ms window and have `CONFIG_BT_CTLR_SCAN_UNRESERVED=y`. This will provide [...]

`CONFIG_BT_CTLR_SCAN_UNRESERVED=y` This sounds interesting and I'll try it out. I guess this make it harder to sync to new devices also if already many are synced?

cvinayak, 2022-09-19:
> Using CONFIG_BT_TICKER_LOW_LAT=y together with #50311 I get this crash pretty much immediately.
> `ASSERTION FAIL [(ticker_status == 0) || (ticker_status == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan_aux.c:699`

You this #50252 in addition.

PR #50252 (2022-09-20): Bluetooth: Controller: Fix TICKER_LOW_LAT Ext. Scan assert check
Fix Extended Scanning assertion check when using Ticker Low Latency Implementation where the failure to schedule a ticker timeout is returned inline compared to deferred failure in the operation callback.

Auxiliary PDU is received using a single-shot ticker timeout with ticks_slot reservation. This single-shot ticker is allowed to fail if it is overlapping another event.

Fixes #49915 when `CONFIG_BT_TICKER_LOW_LAT=y` used.
PR #50311 (2022-09-21): Bluetooth: Controller: Fix repeated skip in ticker resolve collision
Enforce next_is_older only when current has equal force as with the next ticker.

Added Kconfig to disable priority feature in ticker that is currently not used by Zephyr Bluetooth Low Energy
Controller.

The priority feature if enabled then a custom ULL is needed by vendors to avoid repeated skipping of overlapping events as next_has_priority check uses lazy value that would be always lazy_next > lazy_current as currently skipped event becomes the next event with lazy value incremented by 1.

Regression in commit 3a9173afe15 [... cut]
closed 2022-09-20 by fabiobaltieri