zephyrproject-rtos/zephyr #49917: http_client_req() sometimes hangs when peer disconnects

zagor, 2022-09-05:
**Describe the bug**
We have a multi-node system with some inter-node communication performed over https on ethernet.
In our system tests, we restart nodes and/or node software to test resilience and robustness.

Once in a blue moon when the http server node (or server software) restarts, the http client call to `http_client_req()` never returns. It hangs forever and blocks the thread.

**To Reproduce**
Reproduction is difficult. We run 10-hour system tests over night with thousands of http requests and hundreds of server restarts and the problem typically occurs only once or twice for each nightly run.

I have done some attemps at isolating the scenario and force it to occur more frequently but have so far not had great success.

**Expected behavior**
I expect http_client_req() to always return, at most `timeout` ms after being called.

**Impact**
Impact is pretty severe since all execution in the calling thread is stalled permanently.

**Environment (please complete the following information):**
- The http client os is Zephyr 3.1 on STM32. 
- The http server os is Linux 5.4 on Freescale i.MX6.

We first hit this issue on Zephyr 2.7, updated to 3.1 to see if it was fixed, but alas, no.

**Context**
I know this can look like a hopeless ticket. I don't expect anyone to be able to read this and say "eureka, here's the fix". What I'm hoping for is input from people who know the networking code better than me and can discuss possible causes and point at where to dig deeper.

jukkar, 2022-09-05:
It sounds like a locking problem i.e., some part of the code is waiting a lock to release but it cannot be released as the thread cannot continue because of lock so a deadlock is seen. There is not much locking or anything else related to that in the `http_client_req()` API. I suspect that there might be still some tcp issue present.
Is your application doing anything related to locking when the http callback is called?
Are you able to attach a debugger to the stalled thread?

carlescufi, 2022-09-07:
> Once in a blue moon when the http server node (or server software) restarts, the http client call to `http_client_req()` never returns. It hangs forever and blocks the thread.

Can you try to get a stack trace when that happens?

zagor, 2022-09-07:
@jukkar We have a mutex protecting data used in the call, but I fail to see how that could affect http_client_req(). We use the .response callback but it doesn't touch that mutex. I'll add logging to the response callback to see if that reveals any clues.

I am not currently able to attach a debugger to these systems.

zagor, 2022-09-07:
> Can you try to get a stack trace when that happens?

If we can think of a way to detect the hang and log the stack trace from another thread, then yes.
I am unfortunately currently unable to reproduce the issue under conditions where I have a debugger connected.

zagor, 2022-09-12:
I added logging to the response callback, and it shows that the hang occurs prior to it.

Do we have any code for printing an arm stack trace in zephyr? I searched but didn't find anything other than the https://github.com/zephyrproject-rtos/zephyr/issues/4416  feature request. And CONFIG_EXCEPTION_STACK_TRACE seems to be x86 only.

carlescufi, 2022-09-21:
> Do we have any code for printing an arm stack trace in zephyr?

No, we don't. But if you are running with a debugger connected or attach a debugger once the hang happens you should be able to get the stack trace from any of the threads as long as your debugger has support for Zephyr (most do)

Can you try to produce a minimal reproducible sample and environment so that @rlubos can try and reproduce it locally?

carlescufi, 2022-09-28:
Downgrading to low priority since we are currently unable to reproduce.

carlescufi, 2022-10-19:
Closing due to inactivity. Please reopen if you can provide a simple reproducible setup.

closed 2022-10-19 by carlescufi

zagor, 2023-03-15:
I now have a scenario where I can repeat this problem and have been able to pinpoint the issue.

I connect a persistent TCP socket to a server on another physical machine and request http url /ping over it every 5 seconds. I then cut power to the server machine, thus leaving the socket up but in a peer-disconnected state. The next attempt to request /ping blocks longer than the timeout.

Specifically, at the end of `http_client_req()` it calls `http_wait_data()` which in turn calls `zsock_recv()`. This call blocks for a very long time if the socket peer is dead.

`http_client_req()` has a delayed worker that calls `zsock_shutdown()` on the socket when the timeout expires, but it doesn't work. `zsock_recvfrom()` already holds the socket mutex shutdown needs and so `zsock_shutdown()` can do nothing but block forever on `k_mutex_lock`.

Here is a log excerpt showing the sequence of mutex interactions after a call to http_client_req() with a 3000 ms timeout:

```
[00:00:27.694,000] <inf> http_request: > http_client_req /ping
[00:00:27.694,000] <inf> net_sock: lock 0x20007010 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: lock 0x20007030 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: unlock 0x20007030 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: unlock 0x20007010 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: lock 0x20007010 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: lock 0x20007030 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: unlock 0x20007030 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: unlock 0x20007010 z_impl_zsock_sendto
[00:00:27.695,000] <inf> net_sock: lock 0x20007010 z_impl_zsock_recvfrom
[00:00:27.695,000] <inf> net_sock: lock 0x20007030 z_impl_zsock_recvfrom
[00:00:30.695,000] <inf> net_sock: lock 0x20007010 z_impl_zsock_shutdown
```

I have in my application implemented a workaround to this problem by setting sockopt SO_RCVTIMEO on the socket to the same as the timeout I pass to `http_client_req()`.

zagor, 2023-03-15:
I don't have permission to reopen the issue so I'll leave that to someone who does.

zagor, 2023-03-15:
Ah, I forgot an important detail: The socket is a TLS socket. 

This means each call goes through the vtable twice, and locks two mutexes. But the `k_condvar_wait()` in `zsock_wait_for_data` only releases the innermost mutex:

```
[00:03:12.438,000] <inf> http_request: > http_client_req /ping
[00:03:12.438,000] <inf> net_sock: mutex_lock 0x20007010 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_lock 0x20007030 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_unlock 0x20007030 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_unlock 0x20007010 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_lock 0x20007010 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_lock 0x20007030 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_unlock 0x20007030 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_unlock 0x20007010 z_impl_zsock_sendto
[00:03:12.439,000] <inf> net_sock: mutex_lock 0x20007010 z_impl_zsock_recvfrom
[00:03:12.439,000] <inf> net_sock: mutex_lock 0x20007030 z_impl_zsock_recvfrom
[00:03:12.439,000] <inf> net_sock: condvar_wait 0x20007030 zsock_wait_data
[00:03:15.439,000] <inf> net_sock: mutex_lock 0x20007010 z_impl_zsock_shutdown
```

I'm glad to hear this is fixed in 3.3.

rlubos, 2023-03-15:
> Ah, I forgot an important detail: The socket is a TLS socket.

Indeed, that would explain it then:
https://github.com/zephyrproject-rtos/zephyr/issues/54275

This one is not fixed yet, but I have it on my radar.

zagor, 2024-05-22:
I'm not sure if anyone will see this on a closed ticked, but we just updated to Zephyr 3.3 and unfortunately this bug is still present.

zagor, 2024-05-31:
> I'm not sure if anyone will see this on a closed ticked, but we just updated to Zephyr 3.3 and unfortunately this bug is still present.

I was mistaken. My problem was #73580.