zephyrproject-rtos/zephyr #49922: Make picolibc the default C library for Zephyr

stephanosio, 2022-09-05:
#44096 introduced the picolibc, a full-featured yet light-weight C library and fork of newlib, as a module for Zephyr that can be configured and fine-tuned at the Zephyr build time to minimise its footprint.

This issue describes the tasks required for making the picolibc the default C library support in the Zephyr RTOS.

## Tasks

- [ ] Ensure that picolibc is compatible with and tested on all architectures supported by Zephyr
    - [ ] aarch64-zephyr-elf
    - [ ] arc-zephyr-elf
    - [ ] arc64-zephyr-elf
    - [ ] arm-zephyr-eabi
    - [ ] mips-zephyr-elf
    - [ ] nios2-zephyr-elf
    - [ ] riscv64-zephyr-elf
    - [ ] sparc-zephyr-elf
    - [ ] x86_64-zephyr-elf
    - [ ] xtensa-*zephyr-elf
- [ ] Fix any incompatibilities with the picolibc in the Zephyr codebase
- [ ] [Make `PICOLIBC` the default `LIBC_IMPLEMENTATION`](https://github.com/zephyrproject-rtos/zephyr/blob/5c27067dcd9bc03d911aaf5b1628ab56e1c5f4ca/lib/libc/Kconfig#L26-L30)

## Related discussions

* [picolibc footprint comparison](https://github.com/zephyrproject-rtos/zephyr/pull/26545#issue-648441961)
* [Rationale and overall requirements](https://github.com/picolibc/picolibc/issues/177#issuecomment-1073815655)
* [Configuration requirements](https://github.com/picolibc/picolibc/issues/177#issuecomment-1076039486)
* [Adding picolibc as a module](https://github.com/zephyrproject-rtos/zephyr/issues/44143)

keith-packard, 2022-09-05:
I think we should use an SDK version that includes picolibc -- that way we can run both C and C++ tests and use the picolibc module and the SDK version. I think that means getting picolibc into a version of the SDK that we can use from the github CI system?

stephanosio, 2022-09-06:
> I think we should use an SDK version that includes picolibc -- that way we can run both C and C++ tests and use the picolibc module and the SDK version. I think that means getting picolibc into a version of the SDK that we can use from the github CI system?

Yes, https://github.com/zephyrproject-rtos/sdk-ng/pull/287 will need to be completed as part of this.

stephanosio, 2022-09-19:
cc @nashif @carlescufi @tejlmand

stephanosio, 2022-09-19:
**Toolchain WG on 2022-09-20**

* @nashif: This will need to be discussed and reviewed at the project level (TSC).
* @nashif: In general, no strong objection to the general direction of making picolibc the default C library; however, there are some concerns regarding the subsequent effects that this can have:
    * @nashif: How do we prevent the core parts of Zephyr (e.g. kernel) from abusing all the available C library functions that may lead to overall footprint increase?
        * @stephanosio: We can define a minimal set of C standard library functions that are sanctioned to be used in the core parts of Zephyr and guard any extraneous functions by a Kconfig such that they are not available for use by default.
    * @nashif: Using the C standard library functions that are supported by the picolibc but not by the minimal libc will render Zephyr unusable with the minimal libc. How do we prevent that?
        * @stephanosio: We can take a similar approach as the above. But, the minimal libc is TOO minimal and it should not be the "golden standard" to which all Zephyr subsystems are doomed.
        * @stephanosio: For kernel, limiting the availability of the C standard library functions to what is currently provided in the minimal libc seems reasonable. For other subsystems, more flexibility should be allowed even at the cost of breaking the compatibility with the minimal libc.

keith-packard, 2022-09-19:
Is the eventual plan to get rid of the minimal libc?

stephanosio, 2022-09-19:
> Is the eventual plan to get rid of the minimal libc?

Not in the foreseeable future (it will be too controversial). The minimal libc will stay to provide somewhat self-contained? way of testing the kernel.

To quote what @nashif posted in the Teams chat:

> [1:10 AM] Nashif, Anas
in zephyr 1.0.0....

> [1:10 AM] Nashif, Anas
config MINIMAL_LIBC        bool        prompt "Build minimal c library"        help        Build integrated minimal c library. This integrated library is available        to support kernel functionality and test cases. It is not designed to be        used with applications. For applications, please use an external C        library such as newlib.

nashif, 2022-09-19:
> > Is the eventual plan to get rid of the minimal libc?
> 
> Not in the foreseeable future (it will be too controversial). The minimal libc will stay to provide somewhat self-contained? way of testing the kernel.

My point above was that we need to just be cautious with such a move (picolibc as the default libc, dropping minimal, etc.) and think about all possible consequences that might affect users. Last time we tried to make such a move (make newlib default), there were countless issues and that did not happen.

keith-packard, 2022-09-20:
> Last time we tried to make such a move (make newlib default), there were countless issues and that did not happen. (#3102)

Thanks for that link, really useful to be able to review what happened in the past. It looks like picolibc has a few benefits that should help resolve some of the issues from the past:

 * picolibc's printf is smaller than printk, so we can use printf everywhere and save space.
 * picolibc uses TLS for all thread-specific data, avoiding reentrancy issues and the large struct _reent
 * picolibc can be built as a module, instead of being tied to the toolchain.

We also need to ensure that picolibc sees more testing and that bugs related to using picolibc are resolved. I had been working on this slowly during the early picolibc integration work, and many of the patches required to work with picolibc have been merged.

We also need to get Zephyr-specific changes integrated into picolibc -- to make sure the right POSIX APIs are exposed in Picolibc headers when compiling Zephyr code. There are already bits for a few operating systems present, so it's easy and natural to add another one.

Finally, we need to get the Zephyr SDK built to include picolibc along with libstdc++ support. When building with C++, it would be quite a bit of work to get libstdc++ built as a module, so instead we should use the existing crosstool-ng support to get libstdc++ built for picolibc and then use picolibc from the SDK instead of as a module. That's blocked on deciding how to package the SDK to include picolibc. Simply adding picolibc will increase the SDK size quite a bit, potentially going over 2GB. Add to that the requests for including debug symbols in the SDK to ease development.

nashif, 2022-09-21:
@stephanosio  @keith-packard  how far can we get by enabling picolibc as the default in Zephyr as a module? Would this work with all supported toolchains? Adding picolibc to the SDK is fine, however we need to make sure other non-SDK toolchains still work and are not left behind and there need to be parity. We should not get into the situation where using zephyr with some features enabled forces the use the Zephyr SDK.

stephanosio, 2022-09-21:
> Would this work with all supported toolchains

Noting that even the minimal libc seems to be having issues with some proprietary toolchains (e.g. with armclang as mentioned by @tejlmand during the meeting), it will likely need some tweaking.

But, in principle, it is no different to the minimal libc in that it can be adapted to work with any toolchains we support now and in the future because it is integrated to the Zephyr build process and we build it from source.

> however we need to make sure other non-SDK toolchains still work and are not left behind and there need to be parity. We should not get into the situation where using zephyr with some features enabled forces the use the Zephyr SDK.

As long as the toolchain itself allows using a custom C library (i.e. not linking to the default C library embedded in the toolchain), there is nothing fundamentally preventing the picolibc module from working with them (basically, the same requirement as the minimal libc).

keith-packard, 2023-01-18:
What kind of per-architecture testing are you looking for? Do you want external picolibc tests? Or internal Zephyr tests? External picolibc tests rely on architecture-specific support for stand-alone applications to be included in picolibc, which would be nice, but is a pretty big lift for some architectures. Picolibc currently has this for arm, aarch64, power9, risc-v and x86. Internal zephyr tests don't use much of the library, so coverage is weak. I'd be happy to include additional architectures in the picolibc test suite, if someone wants to provide the code.

evgeniy-paltsev, 2023-01-19:
Just a note for the case of ARC (both arc and arc64) & ARC MWDT toolchain we want to keep Zephyr minimal libc as a default - as we doesn't support piclolib with MWDT toolchain.

Not fully sure if we want to switch ARC & GNU toolchain to have picolibc as default (as we don't support it for ARC officially)

keith-packard, 2023-01-19:
> Just a note for the case of ARC (both arc and arc64) & ARC MWDT toolchain we want to keep Zephyr minimal libc as a default - as we doesn't support piclolib with MWDT toolchain.

Is that 'don't currently support' or 'aren't interested in ever supporting'? Iow, would picolibc support be possible in the future? Picolibc supports a fairly wide range of compilers, so adding arc mwdt compiler support should be pretty simple.

> Not fully sure if we want to switch ARC & GNU toolchain to have picolibc as default (as we don't support it for ARC officially)

Makes sense; providing support for multiple toolchains isn't free.

keith-packard, 2023-01-20:
I'm gonna mark arm32, arm64, riscv and x86_64 as 'tested' -- they're all tested with the picolibc (and glibc) test suites, and they all pass the twister tests (with the pending PRs, ofc).

stephanosio, 2023-03-15:
**Re: Document module and toolchain-bundled Picolibc synchronisation policy**

Picolibc integration is currently possible in two modes: using [Zephyr picolibc module](https://github.com/zephyrproject-rtos/picolibc), using toolchain-bundled pre-compiled picolibc (e.g. in Zephyr SDK) -- it is necessary to define a synchronisation policy so that the picolibc available in these two modes are identical at every Zephyr release point.

The policy is as follows:

* Always keep Zephyr and SDK picolibc module references in sync
    * When the picolibc module reference is updated in the Zephyr [`west.yml`](https://github.com/zephyrproject-rtos/zephyr/blob/main/west.yml), update the picolibc submodule reference in the [Zephyr SDK repository](https://github.com/zephyrproject-rtos/sdk-ng).
    * When the picolibc submodule reference is updated in the [Zephyr SDK repository](https://github.com/zephyrproject-rtos/sdk-ng), update the picolibc module reference in the Zephyr [`west.yml`](https://github.com/zephyrproject-rtos/zephyr/blob/main/west.yml).
* For every Zephyr release, ensure that a corresponding Zephyr SDK release is available with the same version of picolibc available in the Zephyr release.
    * One week before the first Zephyr release candidate is tagged, tag a Zephyr SDK release candidate.
    * Tag additional Zephyr SDK release candidates as the picolibc module reference is updated after the first release candidate.

It is also necessary to clearly document the difference between the Zephyr picolibc module and the toolchain-bundled picolibc, and note that these two may not be in sync when using the development branch (i.e. `main`) or using a Zephyr SDK version that is not recommended for the particular Zephyr release.

yperess, 2023-03-15:
> **Re: Document module and toolchain-bundled Picolibc synchronisation policy**
> 
> Picolibc integration is currently possible in two modes: using [Zephyr picolibc module](https://github.com/zephyrproject-rtos/picolibc), using toolchain-bundled pre-compiled picolibc (e.g. in Zephyr SDK) -- it is necessary to define a synchronisation policy so that the picolibc available in these two modes are identical at every Zephyr release point.
> 
> The proposed policy is as follows:
> 
> * Always keep Zephyr and SDK picolibc module references in sync
>   
>   * When the picolibc module reference is updated in the Zephyr [`west.yml`](https://github.com/zephyrproject-rtos/zephyr/blob/main/west.yml), update the picolibc submodule reference in the [Zephyr SDK repository](https://github.com/zephyrproject-rtos/sdk-ng).
>   * When the picolibc submodule reference is updated in the [Zephyr SDK repository](https://github.com/zephyrproject-rtos/sdk-ng), update the picolibc module reference in the Zephyr [`west.yml`](https://github.com/zephyrproject-rtos/zephyr/blob/main/west.yml).
> * For every Zephyr release, ensure that a corresponding Zephyr SDK release is available with the same version of picolibc available in the Zephyr release.
>   
>   * One week before the first Zephyr release candidate is tagged, tag a Zephyr SDK release candidate.
>   * Tag additional Zephyr SDK release candidates as the picolibc module reference is updated after the first release candidate.
> 
> It is also necessary to clearly document the difference between the Zephyr picolibc module and the toolchain-bundled picolibc, and note that these two may not be in sync when using the development branch (i.e. `main`) or using a Zephyr SDK version that is not recommended for the particular Zephyr release.
> 
> cc @nashif @galak @carlescufi @yperess

This seems fine to me. I think that when I last looked at the SDK release cadence it roughly matched to the Zephyr one (being every couple of months). Personally I'm more than happy with keeping the separate releases for the SDK and Zephyr while just tagging the picolibc module when we cut a new SDK release. This way it's easy enough for me to see what commits went in after my version of the SDK and I can choose if I want to stick with the precompiled SDK or want to live on the module cadence. I think this approach would be easier for the SDK maintainers but again, I'm happy with either approach.

keith-packard, 2023-03-15:
This seems reasonable to me as well; how would you like to implement it though? Do you want synchronized PRs to both Zephyr and SDK?

stephanosio, 2023-03-15:
> This seems reasonable to me as well; how would you like to implement it though? Do you want synchronized PRs to both Zephyr and SDK?

Yes, that is the idea.

keith-packard, 2023-03-15:
Ok, I'll make that part of my process then.

jgl-meta, 2023-04-26:
@stephanosio @keith-packard  With regards to the 3.4 release will this be done before?

keith-packard, 2023-04-26:
> @stephanosio @keith-packard With regards to the 3.4 release will this be done before?

Technically, I think we're almost there (blocked on #54637). Given the scope of the change here, should this be brought to the TSC for a ruling?

jgl-meta, 2023-04-26:
> Technically, I think we're almost there (blocked on #54637, which is waiting for @galak to review). Given the scope of the change here, should this be brought to the TSC for a ruling?

@keith-packard I believe it should be brought up and at least mentioned to give people a chance to weigh in.

evgeniy-paltsev, 2023-04-26:
BTW, in case of ARC we are keeping minimal libc as default, however we want to do testing with picolibc on all platforms (in case of GNU toolchain).

I'm wondering if there is a way to do that easily without duplicate each platform to have picolibc platform version.

@keith-packard any thoughts about how this can be implemented?

stephanosio, 2023-04-27:
> > Technically, I think we're almost there (blocked on #54637, which is waiting for @galak to review). Given the scope of the change here, should this be brought to the TSC for a ruling?
> 
> @keith-packard I believe it should be brought up and at least mentioned to give people a chance to weigh in.

@jgl-meta @keith-packard This was brought up in the TSC meeting a few weeks ago during the Toolchain WG report. The only concern was ensuring that the core parts of Zephyr (mainly kernel) can still build using the minimal libc (i.e. coming up with a test strategy to ensure that the kernel does not make use of any C standard functions that are not part of the minimal libc).

stephanosio, 2023-04-27:
> BTW, in case of ARC we are keeping minimal libc as default, however we want to do testing with picolibc on all platforms (in case of GNU toolchain).
> 
> I'm wondering if there is a way to do that easily without duplicate each platform to have picolibc platform version?

@evgeniy-paltsev We will need to keep Picolibc as the default libc for the Zephyr SDK ARC GNU toolchain (i.e. what is used in the CI) so that we can build the tests and samples that make use of the C standard functions that are not part of the minimal libc.

As for the MetaWare toolchain, you can keep `PICOLIBC_SUPPORTED` de-selected to indicate that Picolibc is not supported on it and therefore cannot be the default libc when using the MetaWare toolchain.

stephanosio, 2023-05-03:
During the discussion of https://github.com/zephyrproject-rtos/zephyr/pull/57340 in the TSC meeting on 2023-05-03, the following concerns were raised:

1. What is the scope of the "Zephyr kernel" that is supposed to be compatible with the minimal libc?
1. How do we prevent people from committing code that uses non-standard functions available in Picolibc.

I will look into addressing these concerns by:

1. Documenting the scope of the "Zephyr kernel."
1. Implementing a test to ensure that "Zephyr kernel" does not make use of any functions that are not available in the minimal libc.
1. Documenting the list of _allowed_ standard C library functions that can be used in the broader Zephyr codebase.
1. Implementing a CI check to ensure that _disallowed_ standard C library functions are not used.

keith-packard, 2023-05-03:
> During the discussion of #57340 in the TSC meeting on 2023-05-03, the following concerns were raised:
> 
>     1. What is the scope of the "Zephyr kernel" that is supposed to be compatible with the minimal libc?
> 
>     2. How do we prevent people from committing code that uses non-standard functions available in Picolibc.
> 
> 
>     2. Implementing a test to ensure that "Zephyr kernel" does not make use of any functions that are not available in the minimal libc.

I think this duplicates 4. below?

>     3. Documenting the list of _allowed_ standard C library functions that can be used in the broader Zephyr codebase.

I suggest that this should be explicit about whether Zephyr expects these functions to follow the POSIX or C standard, as well as a general policy about whether functions that exist only in the POSIX standard could be considered for inclusion in this list.

>     4. Implementing a CI check to ensure that _disallowed_ standard C library functions are not used.

Should this instead say "that only _allowed_ standard C library functions are used"? Seems like an allow list will be easier to manage than a disallow list.

andyross, 2023-05-03:
> Should this instead say "that only allowed standard C library functions are used"? Seems like an allow list will be easier to manage than a disallow list.

Or just "the allowed list is whatever is in minimal libc", and the needed CI check is "build everything in tests/kernel with minimal libc".

keith-packard, 2023-05-03:
> Or just "the allowed list is whatever is in minimal libc", and the needed CI check is "build everything in tests/kernel with minimal libc".

That's certainly easier than scanning code for use of symbols present in picolibc but not present in the minimal C library.

keith-packard, 2023-05-04:
I hacked up a script that generates a list of symbols exported by picolibc and not exported by the minimal C library by scanning the public header files from both. It's a terrible combination of coccinelle and classic Unix tools; there's probably a much cleaner way of doing this ...

https://github.com/keith-packard/zephyr/tree/libc-exclude

It reads the picolibc headers from the SDK and the minimal C library headers from Zephyr sources.

stephanosio, 2023-05-04:
> > Should this instead say "that only allowed standard C library functions are used"? Seems like an allow list will be easier to manage than a disallow list.
> 
> Or just "the allowed list is whatever is in minimal libc", and the needed CI check is "build everything in tests/kernel with minimal libc".

That was the original plan and the TSC was not satisfied with that.

As mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/49922#issuecomment-1533407667, first of all, we need to clearly define the scope of "Zephyr kernel" because limiting its scope to `kernel/` is not enough. After that, we can document the policy and CI check for that -- now that is the first part.

The second part is to come up with an allow list of the allowed standard C library functions (and naturally a disallow list of the non-standard and deprecated standard C library functions available in Picolibc and other common "full" LIBCs) for the broader Zephyr codebase (i.e. not just the kernel), because we do not want people abusing the non-standard functions and the deprecated standard functions provided in Picolibc.

The first part is quite straight forward once we clearly define the scope of "Zephyr kernel" (though, that it is by no means an easy task because everyone will have a slightly different opinion on what "Zephyr kernel" or "core parts of Zephyr" really is).

The second part can be implemented in a few different ways; for example:

1. Modifying Picolibc to self-limit the scope of functions available (something like `__ZEPHYR_VISIBLE`).
1. Code scan (CI check) using the disallow list of non-standard and deprecated standard C library functions

I am leaning towards the second option because it is a more general solution (as in not libc-specific) and can be used to enforce the our "libc function usage policy" across multiple LIBCs, which is important because we may have CI workflows for third-party toolchains that do not use Picolibc in the future.

stephanosio, 2023-05-05:
> The second part can be implemented in a few different ways; for example:
> 
> 1. Modifying Picolibc to self-limit the scope of functions available (something like `__ZEPHYR_VISIBLE`).
> 2. Code scan (CI check) using the disallow list of non-standard and deprecated standard C library functions
> 
> I am leaning towards the second option because it is a more general solution (as in not libc-specific) and can be used to enforce the our "libc function usage policy" across multiple LIBCs, which is important because we may have CI workflows for third-party toolchains that do not use Picolibc in the future.

Another simpler option could be to:

1. Use Picolibc with `_ANSI_SOURCE` so that only the _standard_ functions are available.
2. Define the prototypes of the allowed extension functions in the Zephyr-side local Picolibc headers to only permit the allowed extension functions to be used (https://github.com/zephyrproject-rtos/zephyr/pull/57598).
3. Add a CI check to prevent any upstream source files from defining the feature test macros (e.g. `_GNU_SOURCE`).

This will effectively allow us to restrict the Picolibc function usage to what is proposed in #57598, without coming up with an explicit disallow list for the CI check.

Given that any other C standard libraries implementing extensions will have some form of feature test macros, this should also be adaptable across multiple LIBCs.

stephanosio, 2023-05-05:
> Alternatively, adding support for `_ZEPHYR_SOURCE` to picolibc so that it would expose the Zephyr API set directly, allowing code integrated into Zephyr to define `_POSIX_SOURCE`, `_BSD_SOURCE`, `_DEFAULT_SOURCE` or `_GNU_SOURCE` as desired. As we want to move to that solution in the future anyways, how about we just get there now instead? It's all the same work -- auditing the set of functions exposed by the library, but placing the results in the picolibc source code will ensure that it's handled uniformly wrt the other `_SOURCE` macros.

No problem with this approach whatsoever; in fact, I prefer this to #57598. I am kind of rushing because I want to get this done in one way or another before the 3.4 feature freeze.

> There are existing uses of `_DEFAULT_SOURCE`, `_GNU_SOURCE`, `_BSD_SOURCE` and `_POSIX_C_SOURCE` in the code, which would need to get fixed. Fortunately not many though.

Yes, fixing those should be quite trivial (in theory).

> This would also allow code which happens to use these names not get caught by the disallow list on accident.

Correct.

> It wouldn't catch code which includes explicit prototypes for these functions from gaining access to the underlying C library function. Which error would you prefer?

Sure; but, brute-force attempts like that will be easily noticed and rejected during a human code review.

> What other C libraries are relevant here? Aside from newlib, I thought most of the others were basic C libraries without POSIX support?

This was more or less hypothetical -- if we ever want to add support for a libc that includes extensions in the future.

> And I would not be surprised if we could get newlib to adopt the `_ZEPHYR_SOURCE` changes.

https://github.com/zephyrproject-rtos/zephyr/pull/57619#issuecomment-1536783732

keith-packard, 2023-05-21:
Which kernel tests do you want built with the minimal C library? I'm working on a PR to add kernel tests that use `CONFIG_MINIMAL_LIBC=y` to select the minimal C library.

JordanYates, 2023-06-09:
Potential issue: printing 64bit integers (long long support) using the pre-built libraries requires using the full floating point library variant. Currently `PICOLIBC_IO_LONG_LONG` doesn't do anything for pre-built libraries, and values are silently truncated to 32bits.
https://github.com/zephyrproject-rtos/sdk-ng/issues/684

tejlmand, 2023-08-21:
Toolchain wg: @keith-packard anything left before this is ready for merge ?
All issues in https://github.com/zephyrproject-rtos/zephyr/pull/57340 are resolved.

keith-packard, 2023-08-22:
The regression isn't related to picolibc at all; it's caused by a change in how the microbit emulator is run. #61719 has the details.

closed 2023-09-04 by nashif