zephyrproject-rtos/zephyr #49923: ASSERTION FAIL [!arch_is_in_isr()] @ WEST_TOPDIR/zephyr/kernel/sched.c:1449

caco3, 2022-09-05:
I am seeing `ASSERTION FAIL [!arch_is_in_isr()] @ WEST_TOPDIR/zephyr/kernel/sched.c:1449` when stress testing my device.

I tried to trace it further down.
That mentioned line is in the function [z_impl_k_sleep](https://github.com/zephyrproject-rtos/zephyr/search?q=z_impl_k_sleep).
That function only seems to get called in [z_vrfy_k_sleep](https://github.com/zephyrproject-rtos/zephyr/search?q=z_vrfy_k_sleep).
But that function doesn't get called by any other function! How is this possible it still then?

stephanosio, 2022-09-05:
This is lacking too much information. Please file the bug report using the provided bug template.

caco3, 2022-09-05:
@stephanosio I don't see why you need more information to explain why a function can assert althrough it never gets called!

I will add more details once I understand that and can trace it further!

stephanosio, 2022-09-05:
> I don't see why you need more information to explain why a function can assert althrough it never gets called!

which architecture/platform? steps to reproduce, ...

If you think there is something wrong with the upstream code, please file a proper bug report using the template.

closed 2022-09-05 by stephanosio

caco3, 2022-09-05:
@stephanosio So to avoid explaining yourself you simply close the issue?

Well done!