zephyrproject-rtos/zephyr #49924: tests: drivers: pwm_api and pwm_loopback tests failed on frdm_k64f boards

enjiamai, 2022-09-05:
**Describe the bug**
The tests/driver/pwm/pwm_api and tests/driver/pwm/pwm_loopback failed, I found this during the migrating it to the new ztest API.
The test failed due to device is not ready. Seem like it is not configurated correctly on frdm_k64f.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b frdm_k64f tests/drivers/pwm/pwm_api  (also the tests/drivers/pwm/pwm_loopback)
2. west flash
3. sudo picocom -b 115200 /dev/ttyACM0
4. See error

**Expected behavior**
The test should pass.

**Impact**
The test is broken on frdm_k64f

**Logs and console output**
Failed log of pwm_api test:
```
E: Unsupported clock name
E: Could not get clock frequency
E: Unsupported clock name
E: Could not get clock frequency
*** Booting Zephyr OS build zephyr-v3.1.0-4745-g5c27067dcd9b  ***

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/pwm/pwm_api/src/main.c:20: test_main: (device_is_ready(dev) is false)
PWM device is not ready
E: ***** BUS FAULT *****
E:   Imprecise data bus error
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x20000060 r12/ip:  0x00000043 r14/lr:  0x00008d5b
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x000089bc
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200001d0 (main)
E: Halting system

```
Failed log of pwm_loopback test:
```
E: Unsupported clock name
E: Could not get clock frequency
E: Unsupported clock name
E: Could not get clock frequency
*** Booting Zephyr OS build zephyr-v3.1.0-4745-g5c27067dcd9b  ***

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/pwm/pwm_loopback/src/test_pwm_loopback.c:29: get_test_pwms: (device_is_ready(out->dev) is false)
pwm loopback output device is not ready
E: ***** BUS FAULT *****
E:   Imprecise data bus error
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x200000c0 r12/ip:  0x00000043 r14/lr:  0x00009b3f
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x000097a0
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000230 (main)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain:  Zephyr SDK 0.15.0 and 0.14.2
 - Commit SHA:  In any recent commits. bisecting it.

**Additional context**

hakehuang, 2022-09-06:
debug found the error is at ftm initial
```
mcux_ftm_init (dev=<optimized out>) at /home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/pwm/pwm_mcux_ftm.c:398
398             if (err != 0) {
(gdb)
402             if (config->channel_count > ARRAY_SIZE(data->channel)) {
(gdb)
407             if (!device_is_ready(config->clock_dev)) {
(gdb)
412             if (clock_control_get_rate(config->clock_dev, config->clock_subsys,
(gdb)
414                     LOG_ERR("Could not get clock frequency");
(gdb)
415                     return -EINVAL;

```

hakehuang, 2022-09-06:
@galak , @dleach02 , @mmahadevan108

closed 2022-09-16 by mmahadevan108